meatpiHQ/wican-fw #277: Request Support for Mitsubishi iMiEV/Citroen C-Zero/Peugeot iOn (and Mitsubishi Outlander PHEV)

Heavendenied81, 2024-12-07:
I started using MeatPi to send the SOC of my Hunydai Ioniq to my openWB Wallbox and i habe to say it works great. Before I used EvNotiPi but that needed extra Raspberry Pi and was sometimes unstable.

So I already tried to get this to work with my Peugeot iOn and used the information I already have from EVNotiPi but unfortunately it does not work yet...

Can maybe someone how understands all this "CAN" help me?

Here is the information I have from the time I started working with EVNotiPi:

ATFCSH761, ATFCSD300000, ATFCSM1, ATSH761, 2101, carriage return

These commands were supplied by anko on [https://www.myoutlanderphev.com/forum/v ... =10&t=1796](https://www.myoutlanderphev.com/forum/viewtopic.php?f=10&t=1796)

I get the following back:

762;16;46;97;1;71;71;1;87
762;33;1;1;84;69;13;26;62
762;34;0;61;12;1;87;1;87
762;35;0;232;0;246;0;204;8
762;36;15;15;1;133;0;118;183
762;37;254;0;0;1;84;139;144

As far as I understood the SOC is the part B05 (in this case the value is 71)

I tried to use this in Customm PIDs but what I can see than in my MQTT Explorer is:
rx={"bus":"0","type":"rx","ts":3787,"frame":[{"id":883,"dlc":8,"rtr":false,"extd":false,"data":[196,195,130,238,12,170,0,7]},{"id":805,"dlc":2,"rtr":false,"extd":false,"data":[1,0,0,0,0,0,0,0]}]}
error={"error": "Timeout, pid: 2101"}

As so far I know) the protocol is the same for all mentioned cars maybe there is someone who can support?

Thanks.

njh, 2024-12-07:
I have a Peugeot iOn and got it working:
https://github.com/meatpiHQ/wican-fw/discussions/172

Someone else did too:
https://github.com/meatpiHQ/wican-fw/discussions/65

This just done by listening to the CAN messages being sent repeatedly to the bus.
Most other cars are supported by request/response.
I am not sure if anyone has managed to do this for i-Miev or not - I suspect it should be possible.
Then it would be possible to add a Vehicle profile to WiCAN, the way that other vehicles are supported.

I am personally using it for:
- logging millage and battery level to a Google Sheet
- telling my OpenEVSE charger how full the battery is

Heavendenied81, 2024-12-07:
Thanks for your hint.

This is working in general. But unfortunately it is sending always soc = {"soc": xx.xxxxxx}
For my openWB I need only the payload part and only xx.xx is allowed (not 6 digits behind the dot.

Do you know if it is possible to change that?

With the custom PIDs its possible to select MQTT Wallbox as Type and than only the payload is sent.

njh, 2024-12-08:
@meatpiHQ have said that he will add support for raw CAN bus packets to the new vehicle configuration system.
It would be a really good enhancement.

For now I personally use Node-RED to translate the MQTT packets from one format to another.

[image]

This is the code for the 'Set topic' Node-RED Function node that I use:
```javascript
msg.topic = Object.keys(msg.payload)[0]
msg.payload = Object.values(msg.payload)[0]
return msg;
```
(it converts messages from `{"soc": xx.xxxxxx}` to `{"topic": "soc", "payload": xx.xxxxxx}`)

Heavendenied81, 2024-12-13:
@meatpiHQ 
Thank you so much. 
It works great!

Now I finally have a easy and reliable solution for my Ioniq as well as my iOn to transmit the SOC to my openWB.

Thanks!

P.S.:
I tried to use the same parameters that I now use from your json but it did not work before. Is it really that there need to be a vehicle profile with this and I additionaly configure it as custom PID or should it also work only with the custom PID?

meatpiHQ, 2024-12-13:
@Heavendenied81 Thanks for confirming!

The custom PID should work same as the profile, but with the current firmware version custom PIDs wont show up in the HA integration.

Heavendenied81, 2024-12-13:
OK, than maybe I did something else wrong.
But as now its working I'm super happy!

I already started little advertising for WiCan in the openWb forum because I think its really the easiest solution out there to bring the SOC of many differnet cars to openWB.

closed 2024-12-13 by Heavendenied81