meatpiHQ/wican-fw #279: Stellantis Peugeot e-208 / Opel Corsa

deadrabbit87, 2024-12-08:
### Discussed in https://github.com/meatpiHQ/wican-fw/discussions/278

<div type='discussions-op-text'>

<sup>Originally posted by **deadrabbit87** December  7, 2024</sup>
Hi, 

i wan't to get rid of the original Stellantis / My-Peugeot App and use WiCan instead of that. 

As i see, theres no template for this vehicle. What i found is [this](https://www.goingelectric.de/forum/viewtopic.php?p=1514148&sid=038f7b3d9279c69e714d3d0d47a95c73#p1514148) thread in which some IDs are mentioned: 
```
#######
####### Setup commands
#######
ATWS ---> Warm reset set back to defaults
ATI ---> Show interface ID
ATE0 ---> Turn off terminal echo
AT@1 --->
ATSP6 ---> Set to protocol 6: ISO 15765-4 CAN (11 bit ID 500 kbaud)
ATAT1 --->
ATL0 ---> Disable line feed
ATS0 ---> Disable sending spaces between hex characters
ATH1 ---> Turn on odb2 header
ATCAF1 ---> Turn on CAN auto formatting
ATSH79B ---> Set request message header 79B
ATFCSH79B ---> Set flow control header (CAN ID)
ATFCSD300000 ---> Set flow control data
ATFCSM1 ---> Set flow control mode; 1 = User defined CAN ID and data

#######
####### Data commands
#######
ATSH6A2 ---> Set the header of transmitted OBD messages to header 6A2
ATFCSH6A2 ---> Set flow control header (CAN ID)
ATCRA682 ---> Set the CAN hardware filter to 682
22D4341 ---> Ambient temperature
22D8EF1 ---> Battery temperature
22D4021 ---> Speed
ATSH6B4 ---> Set the header of transmitted OBD messages to header 6B4
ATFCSH6B4 ---> Set flow control header (CAN ID)
ATCRA694 ---> Set the CAN hardware filter to 694
22D8161 ---> Battery current
22D8151 ---> Battery voltage
22D86F1 ---> Battery minimum voltage
22D8701 ---> Battery maximum voltage
22D4101 ---> SOC calibrated
22D8651 ---> kWh available
22D8601 ---> SOH
22D8101 ---> SOC
```

Is there a way to create a template for this vehicle with this data? 

</div>

atze09, 2024-12-09:
I already put this into this topic https://github.com/meatpiHQ/wican-fw/issues/119
Haven't had the time to create a template for my corsa (currently fine with soc only).

What need's to be done is read out each value and define the calculation.
So if you read out the CAN and compare with carscanner value the calculation method can be defined.
@meatpiHQ will also support.

deadrabbit87, 2024-12-09:
I've already done this, the log from CarScanner is attached. I will check, if i can figure out the conversion factors.
[log_soc_5479.txt](https://github.com/user-attachments/files/18058317/log_soc_5479.txt)

atze09, 2024-12-09:
> I've already done this, the log from CarScanner is attached. I will check, if i can figure out the conversion factors. [log_soc_5479.txt](https://github.com/user-attachments/files/18058317/log_soc_5479.txt)

Great, yes now the conversion factors need to be figured out and put into the expression of the car template

atze09, 2024-12-09:
`{
	"car_model": "Opel: Corsa",
	"init": "ATE0;ATH1;ATSPA6;ATS0;ATM0;ATAT1;ATSH6B4;ATFCSD300000;ATFCSM1;",
	"pids": [
		{
			"pid": "22D8101",
			"parameters": [
				{
					"name": "SoC",
					"pid_init": "ATCRA694;",
					"expression": "(B3*20)/51",   ??
					"unit": "%",
					"class": "battery"
				}
			]
		},
		{
			"pid": "??",
			"parameters": [
				{
					"name": "Odometer",
					"pid_init": "??;",
					"expression": "([B3:B6])/10",   ??
					"unit": "km",
					"class": "distance"
				}
			]
		},
		{
			"pid":"??",
			"parameters": [
				{
					"name": "??",
					"pid_init": "??;",
					"expression": "(256*B3)+B4",   ??
					"unit": "km",
					"class": "distance"
				}
			]
		}
	]
}`
@meatpiHQ Could you please double check the init part?

deadrabbit87, 2024-12-09:
The factors are not the main problem. I'm not very deep into CAN Bus, so i could not figure out, which init parameters are necessary. 

This is what i endet up with: 
```json
{
    "car_model": "Peugeot: E-208",
    "init": "ATSP6;ATE0;ATCAF;ATL0",
    "pids": [
        {
            "pid": "22D4101",
            "pid_init":"ATSH6B4",
            "parameters": [
                {
                    "name": "SOC",
                    "expression": "(B6 + B7)/512",
                    "unit": "%",
                    "class": "battery"
                }
            ]
        },
    ]
}
```

As i suspect `B*` stands for the Byte? The answer from the car is for a soc of `54,79%`:
```
>22D4101
6940562D4106D9D
```
`6D9D` in decimal is `28061`. `28061/512=54,81`

atze09, 2024-12-10:
> The factors are not the main problem. I'm not very deep into CAN Bus, so i could not figure out, which init parameters are necessary.
> 
> This is what i endet up with:
> 
> ```json
> {
>     "car_model": "Peugeot: E-208",
>     "init": "ATSP6;ATE0;ATCAF;ATL0",
>     "pids": [
>         {
>             "pid": "22D4101",
>             "pid_init":"ATSH6B4",
>             "parameters": [
>                 {
>                     "name": "SOC",
>                     "expression": "(B6 + B7)/512",
>                     "unit": "%",
>                     "class": "battery"
>                 }
>             ]
>         },
>     ]
> }
> ```
> 
> As i suspect `B*` stands for the Byte? The answer from the car is for a soc of `54,79%`:
> 
> ```
> >22D4101
> 6940562D4106D9D
> ```
> 
> `6D9D` in decimal is `28061`. `28061/512=54,81%`.
> 
> I'm not sure if this is correct.
> 
> Is it possible in carscanner, to trigger only one Paremeter? Therefore the reverse enginnering would be much easier.

Yes the reverse engineering is possible, if you click only on sensors and the special sensor in the carscanner app you get the response of this single sensor.
If you are in the progress of reverse engineering, battery information (voltage soh and energy) would be very interesting.

meatpiHQ, 2024-12-13:
answered in https://github.com/meatpiHQ/wican-fw/discussions/278

Kblund, 2026-11-05:
Findings from a Peugeot e-208 (50 kWh, 2020) on a **WiCAN PRO, firmware 6.0.0-beta.7**, OBD port = 11-bit CAN at 500 kbit/s. The UDS values below were matched against Car Scanner on the same drive (its requests and the car's replies captured on the bus at the same time). Unless noted, each one matched Car Scanner exactly.

**Byte numbering.** `d0` = the first data byte *after* the `62 DID DID` echo. The WiCAN v6 expression is given too: on v6 betas, B0 is the ISO-TP length byte, so for a single-frame reply `d0` = `B4`.

### UDS (service 22), requests via `ATSH` + `ATCRA`

**BMS: `ATSH6B4;ATFCSH6B4;ATCRA694;`**
| DID | Value | Formula | WiCAN v6 expression |
|---|---|---|---|
| D815 | Pack voltage, V | `[d0:d1]/16` | `[B4:B5]/16` |
| D816 | Pack current, A (+ = discharge) | `([d2:d3]-11264)/64` | `([B6:B7]-11264)/64` |
| D410 | SoC, % | `[d0:d1]/512` | `[B4:B5]/512` |
| D865 | Energy available, kWh | `[d0:d1]/64` | `[B4:B5]/64` |
| D860 | SOH, % | `[d1:d2]/16` | `[B5:B6]/16` |
| D86F / D870 | Min / max cell voltage, V | `[d0:d1]/1000` | `[B4:B5]/1000` |
| D43B / D43C | Highest / lowest cell number | `d0+1` | |
| D47A | Insulation resistance, kΩ | `[d2:d3]` | |
| D440 | All cell voltages: 240 bytes = 120 × 2-byte slots, 108 used (multi-frame) | scale not verified, presumably mV | |

**VCU: `ATSH6A2;ATFCSH6A2;ATCRA682;`**
| DID | Value | Formula |
|---|---|---|
| D802 | Vehicle speed, km/h | `[d0:d1]/128` |
| D8D0 | Motor torque, Nm (signed) | `s16[d0:d1]/10` |
| D8D9 | Motor power, W (signed) | `s16[d0:d1]*100` |
| D8FB | Coolant temperature, °C | `[d0:d1]/10` |
| D8F9 | Traction circuit coolant, °C | `d1/10` |
| D8FF | A/C high-side pressure, bar | `[d0:d1]/10` |
| D8CD | Motor stator temperature, °C | `d1` |
| D8CE | DC-DC converter temperature, °C | `d1` (r = 0.997, not exact) |
| D8E1 | On-board charger temperature, °C | `d1` |
| D8D6 | Motor coolant inlet temperature, °C | `d1` |
| D8EF | Traction battery temperature, °C | `d0` (WiCAN: `B4`) |
| D434 | Outside temperature, °C | `d0` |

**MCU: `ATSH6A6;ATFCSH6A6;ATCRA686;`**
| DID | Value | Formula |
|---|---|---|
| D49C | **Odometer, km (4 bytes)** | `[d0:d3]`, WiCAN v6 `[B4:B7]` |
| DC00 | Motor speed, rpm | `[d0:d1]` |
| DCB6 | Motor torque, Nm (signed) | `s16[d0:d1]/32` |

@pimvullers @deadrabbit87: D49C on the MCU is the full 4-byte odometer, so no 65,535 km limit. Example reply: `07 62 D4 9C 00 01 88 B5` = 100533 km.

### Broadcast frames (no polling needed)
Byte 0 = first data byte; bit 0 = LSB. These were found by correlating with the polled values above and with manual tests (each door, gear and climate action repeated with ~10 s spacing).

| ID | Signal | Decoding | Confidence |
|---|---|---|---|
| 0x38D | Vehicle speed | `[B0:B1] × 0.01` km/h | r = 1.000 against D802 |
| 0x30D | 4 wheel speeds | 4 × 16-bit `× 0.01` km/h; 1–2 are probably the driven front wheels, left/right unconfirmed | good |
| 0x3A8 | SoC | `[B4:B5] / 512` %, the same value as D410 | exact |
| 0x208 | Motor torque | `signed [B5:B6] / 128` Nm | r = 0.9996 against D8D0 |
| 0x44D | Motor speed | `[B0:B1] × ~0.85` rpm | r = 0.9999 against DC00; the scale is unexplained |
| 0x552 | Odometer | `[B4:B6]` km, 1 Hz. Bytes 1–3 are a clock in 0.1 s steps | exact against D49C |
| 0x5B2 | Outside temperature | `B2 / 2 − 40` °C | matched D434 |
| 0x489 | Gear selector | low nibble of B0: **0 P, 1 R, 2 N, 3 D** (0x349 B3 low nibble repeats it) | confirmed with P-N-R-N-D-N-R-N-D |
| 0x3F2 | B mode (strong regen) | B0 bit 6 (byte 0 = `0x41` in B, `0x01` in D) | confirmed |
| 0x3AD | Electric parking brake | low nibble of B3: 1 applied, 3 releasing, 0 released, 2 applying | confirmed |
| 0x412 | Doors / boot, 1 = open | B6: bit 3 front left, bit 4 front right, bit 5 rear left, bit 6 rear right, bit 7 boot | confirmed per door |
| 0x512 | Central locking | B1 bit 4, 1 = locked | confirmed |
| 0x612 | High beam | B1 bit 2 | from the earlier community DBC; not retested here |
| 0x3CE | Climate running | B4 bit 4 (B4 = `0x31`/`0x32` on, `0x01` off) | confirmed with on/off/on/off |

- **Setpoint:** the A/C temperature setpoint itself does **not** seem to be on the OBD-port bus. Only its effects show, in 0x3CE and 0x3F2.
- **WiCAN v6 filter bug:** filters on frames whose first byte is `0x01`–`0x07` get cut short (#1041), so 0x552 can't be read through an autopid filter on beta.7 yet. Everything else in the table works as a filter, and 0x3F2 does too, because only byte 0 is needed.

I have all of this as a DBC file and can share it if anyone wants it.