meatpiHQ/wican-fw #280: KIA Sorento Support

s0ftice, 2024-12-10:
Model: KIA Sorento 2.2(8AT) CRDI 2WD S/R

Hi all,
I am using the Home Assistant integration with firmware 3.48 on my WiCAN-OBD.
The integration works, as I can see the Diagnostic entities as well as Status Voltage with the correct values.
All other entities show as "Unavailable".

I have tried to set the following vehicle models "Kia/Hyundai: Niro/Soul/Kona", "AAA: Generic", and "Kia: Niro PHEV".
But all only give me the Status Voltage, the other entities remain unavailable.

As mentioned on https://meatpihq.github.io/wican-fw/home-assistant/integration_setup
"Select your vehicle model from the dropdown list. If your vehicle model is not yet supported, please _open an issue on GitHub_, and we will walk you through adding support."

Here I am :)

Thanks.

s0ftice, 2024-12-21:
@meatpiHQ appreciate if someone can give me pointers with regards to "If your vehicle model is not yet supported, please open an issue on GitHub, and we will walk you through adding support."

Thanks :)

meatpiHQ, 2025-01-02:
@s0ftice sorry for late response.

I'll help you create a profile for the Sorento. Is this an EV?

Which parameters would you like to add?

s0ftice, 2025-01-03:
Thanks for your response.
It's a Diesel.

I would love to have mileage, door & window locked/open status, fuel level, tire pressure, light status, maintenance reminders such as service due date, oil pressure, engine temperature.

Cheers

meatpiHQ, 2025-01-05:
@s0ftice I think Sorento should support the standard PIDs. I suggest that you update to the latest firmware version v4.00. You can scan and add the supported parameters.

1- Set the protocol to Autpid, submit changes
2- After reboot, go to Automate page and select the ECU protocol, try 6 first.
3- Press scan, selects the parameters and then submit 

https://github.com/meatpiHQ/wican-fw/releases/tag/v4.00

s0ftice, 2025-01-23:
Thanks so much, that worked. I got the scanned PIDs now in Home Assistant.
I am just wondering if there are any vendor-specific PIDs for my car model that go beyond the SAE standard and cover the use cases I am looking for, such as
- 	Odometer readings.
- 	Fuel level percentage.
- 	Door and window statuses.
- 	Light statuses.
- 	Tire pressure from TPMS.
- 	Maintenance-related data like oil life or service reminders.

Is there a way to scan for these? The vendor-specific section only offers Kia/Hyundai (Niro/Soul/Kona) and Kia (Niro PHEV). How does one find the init strings and expressions to create a custom JSON?

Thanks! :)

dpkano, 2026-09-09:
Any evolution on this front? I'm also interested in adding new kia cars...

s0ftice, 2026-09-09:
not from my side unfortunately, I gave up. Could not get the information I was looking for.
I now use the doorbell camera and a local LLM to alert me whenever the car is not locked (regularly checking the side mirrors position) :)

dpkano, 2026-09-09:
ah. thanks for the reply. I'll try collecting the logs and reverse engineer the PIDs sometime later...