meatpiHQ/wican-fw #284: Full configuration backup/restore

MarcosAPMT, 2024-12-16:
Hi. I would like you to add an option to export and restore all WiCAN configuration from the web interface, preferably in some format that we can modify, such as XML or JSON. We are considering using the devices massively and being able to generate the configuration and share it with end users would be ideal for us. Thanks.

meatpiHQ, 2024-12-16:
@MarcosAPMT Configuration as in everything in the settings page?

If you'd like, we can customize the logo on the enclosure? You can contact me on the website if you're interested https://www.meatpi.com/contact

MarcosAPMT, 2024-12-16:
> @MarcosAPMT Configuration as in everything in the settings page?
> 
> If you'd like, we can customize the logo on the enclosure? You can contact me on the website if you're interested https://www.meatpi.com/contact

Everything from the Settings and the Automate tabs would be ideal. It really sounds like a basic feature for a device of this kind. We don't need any other fancy stuff 😄

meatpiHQ, 2024-12-22:
@MarcosAPMT You can try the new beta, the upload/download is available in the system tab

[wican-fw_obd_v400Beta4.zip](https://github.com/user-attachments/files/18221204/wican-fw_obd_v400Beta4.zip)

MarcosAPMT, 2024-12-23:
Hey! Just tested it and seems to be working great. This is just what we needed. Great UI changes too. Looking forward to the stable release. Thanks a lot!

closed 2024-12-23 by MarcosAPMT