meatpiHQ/wican-fw #288: Nissan Leaf ze0 support

timnijland3, 2024-12-18:
I just bought the WiCan adapter.

It works fine, but the Nissan Leaf Ze0 isn't supported yet.
I'm okay with fiddling around myself so get the SoC of the Ze0. But I can't find a way to import a custom profile (docs not finished yet)

With a bit of help from the internet I managed to put together this, I dont know if it works because i can't test it:

`{
  "car_model": "Nissan : leaf-ze0",
  "init": "ATSP6;ATFCSH79B;ATST96;ATFCSD300000;ATFCSM1;",
  "pids": [
    {
      "pid": "2101",
      "pid_init": "ATSH79B;ATFCSH79B;ATCRA7BB;",
      "parameters": [
        {
          "name": "Hx",
          "expression": "[B3]*0.5",
          "unit": "%",
          "class": "battery"
        },
        {
          "name": "SoC",
          "expression": "[B5]*0.5",
          "unit": "%",
          "class": "battery"
        }
      ]
    }
  ]
}
`

meatpiHQ, 2025-01-01:
@timnijland3 The Leaf ZE1 Did not work on the AZE0 ?

> While in the "Monitor" page when using a 500k Bitrate i saw a lot of data come by. Can you please help me?
 
Can you please post a screenshot of the monitor tab with the frames?

timnijland3, 2025-01-02:
@meatpiHQ No, the ZE1 is different compared to the AZE0. Here i have a .dbc file from the ZE1 and the AZE0 for you. I tried to fiddle around with it a bit, got it to work somehow, but changed something and now its not working anymore. But i don't know what i did change to break it! But with the info from the AZE0 DBC file i got the SoC correct.
[Nissan Leaf CAN DBC.zip](https://github.com/user-attachments/files/18292050/Nissan.Leaf.CAN.DBC.zip)

Also, is it possible to read the "CablePluggedIn" as a boolean and can i somehow trigger the car to "Start Charging" / "Stop Charging"? This would be so awesome if thats doable.

meatpiHQ, 2025-01-05:
@timnijland3 Did you try car scanner?

Seems like the ZE1 broadcasts the data on the CAN bus. Currently this is not supported by the vehicle profiles, but I plan to add support for it soon.

If you can read the data in car scanner, this means it supports getting the data through PID requests then it would be easy to create the profile.

timnijland3, 2025-01-06:
@meatpiHQ Yes, with the mode set to ELM327 i was able to use the CarScanner app to read out the Nissan Leaf AZE0.

meatpiHQ, 2025-01-06:
@timnijland3 That's great! Please export the logs from car scanner. 

[image]

timnijland3, 2025-01-06:
@meatpiHQ ￾
[InstantFuelConsumption VSS=Id=13:Vehicle speed:False]

===== DISCONNECTED AT: 24.12.2024 09:53:38
===== LOG STARTED: 27.12.2024 11:07:31=====
=====Version=1.112.9 Build=1.112.9 P=False NoDelayELM327Init=False
====Connection profile=Nissan Leaf ZE0/AZE0 (2010-2017)
[Connection type: WIFI]
[WiFi name: <null>]

[Connecting to 192.168.2.65:23]
client = new TcpClient();
BEFORE await client.ConnectAsync(server, port);
TCP connection timeout by Task.Delay(8)

Connection failed.

===== DISCONNECTED AT: 27.12.2024 11:07:35
===== LOG STARTED: 27.12.2024 11:07:36=====
=====Version=1.112.9 Build=1.112.9 P=False NoDelayELM327Init=False
====Connection profile=Nissan Leaf ZE0/AZE0 (2010-2017)
[Connection type: WIFI]
[WiFi name: <null>]

[Connecting to 192.168.2.65:23]
client = new TcpClient();
BEFORE await client.ConnectAsync(server, port);
TCP connection timeout by Task.Delay(8)

Connection failed.

===== DISCONNECTED AT: 27.12.2024 11:07:40
===== LOG STARTED: 27.12.2024 11:07:41=====
=====Version=1.112.9 Build=1.112.9 P=False NoDelayELM327Init=False
====Connection profile=Nissan Leaf ZE0/AZE0 (2010-2017)
[Connection type: WIFI]
[WiFi name: <null>]

[Connecting to 192.168.2.65:23]
client = new TcpClient();
BEFORE await client.ConnectAsync(server, port);
TCP connection timeout by Task.Delay(8)

Connection failed.

===== DISCONNECTED AT: 27.12.2024 11:07:45Connection attempts failed

===== LOG STARTED: 27.12.2024 11:08:07=====
=====Version=1.112.9 Build=1.112.9 P=False NoDelayELM327Init=False
====Connection profile=Nissan Leaf ZE0/AZE0 (2010-2017)
[Connection type: WIFI]
[WiFi name: <null>]

[Connecting to 192.168.2.65:23]
client = new TcpClient();
BEFORE await client.ConnectAsync(server, port);
client.Connected
got stream from client

[Connected to 192.168.2.65:23]
ATZATE0
[ELM RESET]
ELM327 v1.3a
>
Initialize(initMode=Default)

InitializeCustomInitString(initMode=Default)
ATE0OK
>ATE0OK
>STI?
[... 938 lines cut ...]
Stopped

===== DISCONNECTED AT: 06.01.2025 19:43:46
[///**** CONTACT DEVELOPER REPORT:[Car Scanner settings:]
AccelerationItems=5
AccelerationTestSpeedSourcePid=13
AccelerationUseG=True
AdaptiveTimings=1
AddNissanConsult3Pids=True
AllowProfilesBackgroundUpdate=True
AlwaysPingECU=True
AlwaysRecordFuelConsumption=False
ATCommandStateOptimization=True
ATCRAOptimization=False
ATSTIdx=6
AutomaticallySwitchTheme=True
AutomaticallySwitchThemeAvailable=True
AutomaticReoptimization=True
BindSocketToFirstWiFiNetwork=True
BindSocketToWiFiNetwork=True
BluetoothLocationWarningShowed=False
BoostCalculationMethod=Auto
BoostCalculationMethodIdx=0
BrandForDTC=Nissan
BTDeviceID=

timnijland3, 2025-01-06:
[log.txt](https://github.com/user-attachments/files/18323418/log.txt)
@meatpiHQ  I added the Log.

Is it doable to have the following information returned:

1. State of Charge
2. Cable plugged in (charging cable)
3. A toggle to "start/stop" charging?
4. A toggle to "start/stop" the A/C system for preheating the car?

This would really help out my parents with their Home Assistant :)

meatpiHQ, 2025-01-07:
@timnijland3 Please follow the instruction in the documentation to export the logs, we need to find the SoC sensor in the logs make sure to press on the sensor and let it log for few seconds before unplugging WiCAN so that the last reading will be the SoC.

As for the other parameter you requested, we need to find the PIDs for these if they are available. I suggest you search for anyone who has done this or if this is even possible on the Leaf.

https://meatpihq.github.io/wican-fw/car_scanner/logs

dbeulen, 2025-07-31:
I'm also interested in the leaf AZE0.

I have a wican pro installed, but i have not been able to find out how to get the SOC. See my attached log and screenshot for the actual value.

[log 31-07-2025.txt](https://github.com/user-attachments/files/21536176/log.31-07-2025.txt)

[image]

mrnro, 2026-04-13:
Hi @ialex87,
It doesn't match the value displayed in the dashboard perfectly over the entire range "0-100%", but close.
It is spot on from approx 15% to 64%, not tested lower values.
At dashboard value 65% it reports 66% and the offset seems to increase linearly so at 72% the formula reports 74%.
Haven't had time to adjust it.

I found that the SOH from PID 2101 calculation above is wrong but haven't investigated it any further.

jdudekcap, 2026-10-13:
Hello 
Can anybody confirm it is worth to buy this device for Leaf AZE0 ?
BR

maximusvo, 2026-10-13:
> 20\. Hello
>          Can anybody confirm it is worth to buy this device for Leaf AZE0 ?
>          BR

Sadly not. There is no support so far. Despite investing a reasonable amount of time, I was unable to get it working.

meatpiHQ, 2026-10-13:
@maximusvo I've just created a profile for the AZE0, please try it and let me know if it works.

ialex87, 2026-10-26:
hey @meatpiHQ  thanks for adding the profile! I just tested the latest beta firmware with the profile for my nissan leaf AZE0 will gonna test if for a while to make sure the reported data is correct. One thing for sure I can tell that ODO in my case is off by about 5k km.

Thank you for you work on this!