meatpiHQ/wican-fw #289: Possible to connect Wifi(MQTT) and Bluetooth at the same time?

Ipirk88, 2024-12-19:
Hello, 

is it possible to connect Wifi (Send SoC via Wifi/MQTT (with the Automate function) to my SmartHome) and Bluetooth (Phone - CarScanner) at the same time? 

I tried, but i failed - so i wondering if it is possible or not. 

Thanks for helping!
BR

meatpiHQ, 2025-01-02:
@Ipirk88 it might be possible, but few things you need to overcome like the lack of ram to run both mqtt and bluetooth at the same time, you need to optimize the firmware to make it work. Plus car scanner polls the ecu at high rate, some you have to optimize it in a way to share the CAN bus.

FEmser, 2025-05-17:
Related: 
https://github.com/meatpiHQ/wican-fw/issues/413

FEmser, 2025-05-17:
Related
https://github.com/meatpiHQ/wican-fw/issues/179