meatpiHQ/wican-fw #292: Error in logs from wican HA integration

brjhaverkamp, 2024-12-23:
Hello 

I am struggling to get things back running again after the 12V battery in my Opel Ampera-E failed. The car was offline for about a week.
When I replaced the battery, all worked again, except the wican doesn't report the capacity and SOC anymore.
I've been looking through the settings in the wican and HA.
I found the below  in my HA log that I hope is related.
Full story is that I also upgraded HA in that week, so the problem might be on that side as well.

Hope you can advice how to get it up and running again, or find the bug.

Bert

------------------------------
This error originated from a custom integration.

Logger: homeassistant
Source: custom_components/wican/sensor.py:153
integration: WiCAN
First occurred: December 22, 2024 at 11:27:36 AM (710 occurrences)
Last logged: 3:46:24 PM

Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 267, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 478, in _async_refresh
    self.async_update_listeners()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 177, in async_update_listeners
    update_callback()
    ~~~~~~~~~~~~~~~^^
  File "/config/custom_components/wican/sensor.py", line 147, in _handle_coordinator_update
    self.set_state()
    ~~~~~~~~~~~~~~^^
  File "/config/custom_components/wican/sensor.py", line 153, in set_state
    state = self.process_state(self.coordinator.data['status'][self.get_data('key')])
                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^
TypeError: 'bool' object is not subscriptable

meatpiHQ, 2024-12-29:
@brjhaverkamp Try removing the integration and add it again see if that helps.

brjhaverkamp, 2024-12-31:
Thanks,  Probably would have done the trick, but after 3 days it started working again, Not sure what happened. Could be a restart of HA happened.

jwiesel, 2025-02-16:
Hi @brjhaverkamp ,
a similar log-entry has been posted in the repository of the HomeAssistant integration: https://github.com/jay-oswald/ha-wican/issues/18

Maybe both of them are related.