meatpiHQ/wican-fw #294: Vehicle Profile: Toyota Prius 04-09, 10-16 and newer?

black-jmyntrn, 2024-12-25:
Im shocked this is not available and that no other owners have requested this... I'm new here but is there a guide on how to make a Profile?

timnijland3, 2024-12-26:
I can help you if you provide me with the ELM output (there is a guide on how to do this on the Documentation site)

meatpiHQ, 2024-12-29:
@black-jmyntrn There are some basic steps in the Documentation.

https://meatpihq.github.io/wican-fw/config/automate/new_vehicle_profiles

Here's how to export the logs from car scanner

[image]

Boaztheostrich, 2025-01-07:
Hey all, I dropped my logs for gen 2 in another [issue](https://github.com/meatpiHQ/wican-fw/issues/310) but wanted to include them here as well incase it would be helpful to anyone!

[Gen 2 log ecu .txt](https://github.com/user-attachments/files/18327777/Gen.2.log.ecu.txt)
[Prius gen 2 log car scanner.txt](https://github.com/user-attachments/files/18327778/Prius.gen.2.log.car.scanner.txt)

meatpiHQ, 2025-01-09:
Adding this new link here for reference on how to export the logs from car scanner for a specific parameter

https://meatpihq.github.io/wican-fw/car_scanner/logs

Boaztheostrich, 2025-01-09:
Just to clarify will I need to do this for each sensor?

Or will all be captured when I do it for one? Thank you!

meatpiHQ, 2025-01-11:
@Boaztheostrich you need to do it one by one. Lets start with the SoC first.

Boaztheostrich, 2025-01-11:
[Engine coolant temp.txt](https://github.com/user-attachments/files/18386851/Engine.coolant.temp.txt)

I tried to do it for engine coolant temp.

Boaztheostrich, 2025-01-11:
Here is rpm 
[log rpm gen 2 Prius.txt](https://github.com/user-attachments/files/18386885/log.rpm.gen.2.Prius.txt)

Here is speed 

[log speed .txt](https://github.com/user-attachments/files/18386884/log.speed.txt)

Please let me know if this is helpful and working!

if I need to upgrade to pro to get the data you need for distance traveled I can do that.

meatpiHQ, 2025-01-12:
@Boaztheostrich please update to the latest firmware version and try using the standard PIDs. You'll be able to scan and add these parameters.

You might find the SoC is supported under "Hybrid battery pack remaining life"

Boaztheostrich, 2025-01-12:
> @Boaztheostrich please update to the latest firmware version and try using the standard PIDs. You'll be able to scan and add these parameters.
> 
> You might find the SoC is supported under "Hybrid battery pack remaining life"

Where do I find the normal PID’s in the app?

I think I can figure out how to update the scanner today.

meatpiHQ, 2025-01-12:
Follow the instructions in the link below to update. Disable MQTT and submit changes before you update.

You'll find the standard PIDs in the Automate tab.

https://meatpihq.github.io/wican-fw/config/firmware-update

Boaztheostrich, 2025-01-24:
Got the scanner updated to latest version

Boaztheostrich, 2025-01-24:
[log state of charge gen 2 Prius.txt](https://github.com/user-attachments/files/18538459/log.state.of.charge.gen.2.Prius.txt)

I tried to get state of charge it said you need pro mode though.

Boaztheostrich, 2025-01-24:
[image]
[image]
[image]

I also tried the standard PID page you mentioned above are the results of that scan. 

What should my next steps be? Sorry for taking so long to update and get more scans. 

What would it take to get info like total miles traveled off the device to a separate esp32 or similar and display it on a screen? Would love to be able to add more info to my hud. Rpm would also be great.

Boaztheostrich, 2025-03-09:
Wanted to follow up on this, it’s been over a month. Really would like to get this working if at all possible!

black-jmyntrn, 2025-03-14:
hi there, you and I both. what are your can settings as on my end none of them even get me to a completed auto pid screen and....

I'm well versed with things Prius related, one thing i found out is they do not like low cost dongles. for example i have a gen 2, 3 and 4 Prius and on the Gen 3 check PCS error always pops up with this one elm dongle.  So when this Gen 2 got a red triangle after i forget clicking something in the web gui not to mention its as if the can options arent working for the Prius... Ive had to try and focus on just getting the normal PIDs without success.