meatpiHQ/wican-fw #296: Problems with FW 4.00

cr0i, 2024-12-29:
Hello @meatpiHQ ,

today I´ve tried your new FW 4.00 and I found many interesting features (WPA3, Min Max values and more).
But I don´t get it working like before in my configuration.

When I use Custom PIDs, I get no answer on my MQTT Topic.
When I use vehicle Profiles and delete all topics exept one, I get the value on the configured destination.
With two ore more topics in vehicle profiles, the values exept the first are calculated wrong.
On reboot WiCAN often does not connect to my WLAN or crashes completely.

meatpiHQ, 2024-12-29:
@cr0i which profile are you using? It would be great if you can give me some steps to reproduce.

If it's not connecting to wlan please change it back to WPA2.

cr0i, 2024-12-29:
WPA2 or WPA3 both sometimes don´t connect to my WLAN.
Then I have to connect to AP Mode and reboot again, then it´s fine.
I use VW:ID Profile, with Custom Init and pid_init.
Mainly I need SoC and Range, when I delete everything else, then the range is reported with "0".

meatpiHQ, 2024-12-29:
What are the custom init and the pid_init strings do you use?

cr0i, 2024-12-29:
I use the strings, that comes with the Profile VW:ID without changing it.
I already switched back to 3.48, one moment please, then I can give you some screenshots.

cr0i, 2024-12-29:
For the test I´ll use only 2 values:
[image]

SoC is calculated correct, Range is wrong:
[image]

Custom PIDs are not working at all, I deleted them:
[image]

cr0i, 2024-12-30:
The main problem, that Custom PIDs are not working at all should be fixed first.

I used Vehicle Profiles before too. After the Update the Topic were it should publish is grayed out and I have to enable Grouping for HA so it would be writeable. 
That should be fixed too. 

I think it is important that the previously configured Options should work after the update without need to change.

meatpiHQ, 2025-01-02:
@cr0i I'm unable to reproduce the issues you're having with v4. Custom PIDs and Vehicle Specific are both working.

Are you now able to get the device connected to your network?

Can you please try to delete all the custom PIDs and Custom init first and just test the profile ?

cr0i, 2025-01-02:
Network must be a problem in my WLAN, this we can ignore for now.
This I´ve done before:
I deleted all Custom Init and PIDs and all other Options in Automate.
Then I Load the actual Profile from VW:ID
Then I deleted all PID except soc and range.
In the PIDs I gave the MQTT Destination I wanted.
Then only the SOC Value is correct, Range is always 0.

Custom PIDs I dont get to work at all.

So I think we have again problems with the Init-Strings.

cr0i, 2025-01-02:
In Custom PIDs the field Init (PID Init) does not work.
When I put the hole Init string I need in Custom Initialisation, then I get an answer.

meatpiHQ, 2025-01-05:
@cr0i You're correct PID Init was not being sent. Please try the firmware update below and let me know if it works.

Btw in v4 you can backup the device configuration by downloading the full device configuration file.

[wican-fw_obd_v401.zip](https://github.com/user-attachments/files/18310228/wican-fw_obd_v401.zip)

cr0i, 2025-01-05:
Yes, now it works perfect, thank you.
One minor issue:
When I want to expand the values under Custom PIDs, I have to click twice.
Under Vehicle Profiles it´s correct, there I only have to click once.

Grouping must now be enabled, so that the values from vehicle profiles were published, but this is not enabled by default after upgrade.

meatpiHQ, 2025-01-05:
@cr0i Awesome thanks for confirming.

Please try the firmware below.

> When I want to expand the values under Custom PIDs, I have to click twice.
Under Vehicle Profiles it´s correct, there I only have to click once.

Fixed 
> In About the Firmware is V 4.1, but Github V 4.01. Is this correct?

Fixed 
[wican-fw_obd_v402.zip](https://github.com/user-attachments/files/18310664/wican-fw_obd_v402.zip)

> Grouping must now be enabled, so that the values from vehicle profiles were published, but this is not enabled by default after upgrade.

Why do you think it should be enabled by default?

cr0i, 2025-01-05:
I don´t know if I need Grouping and what it exactly does, befor it could not be enabled.

Vehicle Profiles before sended data to MQTT Destination and Topic.
This fields are disabled, when grouping is disabled.
I had to enable grouping, so i could activate sending Profile to this MQTT Topic again.

meatpiHQ, 2025-01-06:
If you enable grouping and set the "Destination Type" to MQTT topic it will send all the parameters in one json to the Destination topic specified under Grouping.

cr0i, 2025-01-06:
In FW 3.48 this was the same, but the Grouping Option was greyed out and disabled.
I activated Vehicle Profiles and gave an MQTT Destination.
Then all values from the vehicle profile where published to this one topic.

So if someone used this option, it should enable grouping and Type MQTT on upgrade to FW4.x.
The topic is already there, but it was grayed out, because grouping was disabled.

Otherwise, users would miss data, because nothing is send.