meatpiHQ/wican-fw #298: Mercedes sprinter van 2500 (2024)

richl007, 2024-12-31:
Can someone please help me with the setup for a Mercedes sprinter van 2500 (2024)?

meatpiHQ, 2025-01-01:
@richl007 Did you update to the latest firmware version? 

What parameters are you looking for?

richl007, 2025-01-01:
yes I'm on v4.00. It might just be me not knowing how to set it up

Ignition state
Coolant Temp 
KMS (milage) 
Parking brake
Fuel / DEF level

meatpiHQ, 2025-01-01:
Here's some quick steps for you.

1- Set protocol to autopid and reboot 
2- In the automate tab, standard PID select protocol 7
3- Click scan to check the supported PIDs 
4- select and add the ones you need

Setup the integration in case you're using HA

https://meatpihq.github.io/wican-fw/home-assistant/integration_setup

meatpiHQ, 2025-01-01:
The ignition was ON?

richl007, 2025-01-01:
Yes, it was.

meatpiHQ, 2025-01-02:
Please try to set the protocol to elm327 and connect using car scanner app, just to check if it's able to correctly read the parameters .

richl007, 2025-01-06:
Yes I was able to connect with the car scanner app.

[image: image.png]

[image: image.png]

richl007, 2025-01-06:
Yes I was able to connect with the car scanner app. 

[image]

richl007, 2025-01-06:
[image]

meatpiHQ, 2025-01-07:
That's great can you please export the logs from car scanner.

[image]

richl007, 2025-01-07:
[log.txt](https://github.com/user-attachments/files/18333044/log.txt)

meatpiHQ, 2025-01-07:
Ok thats interesting, doesn't look like it supports the standard PIDs.

Can you please follow the instructions in the link below to export the logs. Lets with one sensor first the coolant temp for example. 

Make sure to press on the coolant sensor and let it log for few seconds before unplugging WiCAN, this way it will the last Reading in the logs. And also note the value of the sensor during the reading.

https://meatpihq.github.io/wican-fw/car_scanner/logs

richl007, 2025-01-07:
[log (1).txt](https://github.com/user-attachments/files/18333407/log.1.txt)

meatpiHQ, 2025-01-07:
Looks like the coolant temp is standard. You should be able to scan and add it. Can you please try again set the protocol to autopid, reboot, select ecu protocol 7 and scan again.

meatpiHQ, 2025-01-09:
@richl007 Please update to the latest firmware version and then try to upload the "Sprinter.json" profile in the automate page, the user who created this profile had a 2023 Sprinter.

https://github.com/meatpiHQ/wican-fw/releases/tag/v4.03

[Sprinter.json](https://github.com/user-attachments/files/18363957/Sprinter.json)

richl007, 2025-01-09:
I updated to 4.03 and uploaded the sprinter.json. MQTT is now greyed out, is that normal? When I scan now, it just scans forever. Is that normal? 

[image]
[image]

meatpiHQ, 2025-01-10:
You mean "MQTT HA discovery" disabled? Then yes that's normal. 

I see you have loopback ip in your configuration?

[image]

richl007, 2025-01-12:
Working :) Thank you for your help!!!

[image]

meatpiHQ, 2025-01-12:
Ok sounds like there is a bug if you put in a bad MQTT URL.

richl007, 2025-01-16:
I fixed the MQTT settings and I'm getting the sensors (the ones you gave me)  in MQTT/HA now. I can not scan for available PIDs though. It either scans forever, or errors out (see pics) . The engine was running on all tests

meatpiHQ, 2025-01-16:
Smarty Van did a video using WiCAN on the Sprinter he's the one who made the profile, but the scan works for him. Can you please download the device configuration from the system tab? Remove all password and sensitive info and post it here so I can look into it.

https://youtu.be/hx8UjOLD1tw?si=yY4iCw6ZnKdcrFEf

richl007, 2025-01-16:
Here is mt config file 

[config_2025-01-16.json](https://github.com/user-attachments/files/18445264/config_2025-01-16.json)

SmartyVan, 2025-01-20:
@meatpiHQ — I've had another Sprinter owner (2018) NCV3 who can't get the scan to work either. Something interesting going on here...

mcgyver78, 2025-01-22:
sprinter van seams do have some unusual CAN. engine CAN 500kbit
comfort ecu 83,333 kbit 

https://www.manualslib.de/manual/202816/Mercedes-Benz-Sprinter-Bm-906-611.html?page=222

i tried a lot of more or less cheep ELM327 BT Adapter with smartphone app, no one was able to connect. only professional OBD tester (BOSCH etc.) ware able to connect. 
And i have Tactrix Openport 2.0 OBD to USB adapter, with original Mercedes Bens Software ( jailbreak / hacked version), with the same adapter vediamo (professional software) software is also able to connect. 

Car scanner doesn't connect.

meatpiHQ, 2025-01-22:
@mcgyver78 Have you tried manually setting the ecu protocol to **"ISO 15765-4 CAN (29 bit ID, 500 Kbaud)"** in car scanner

https://meatpihq.github.io/wican-fw/car_scanner/usage#setting-up-the-car-scanner-app

mcgyver78, 2025-01-22:
yes, automatic mode and advanced setting to 29/500

mcgyver78, 2025-01-22:
ignition is on

mcgyver78, 2025-01-22:
https://github.com/user-attachments/assets/c5248de1-1422-4b57-8c48-10aa103eb8f5

meatpiHQ, 2025-01-22:
@mcgyver78 I noticed you did not start the engine? Maybe that would make a difference?

mcgyver78, 2025-01-22:
i found some information deep in a printer van forum, i cloud be ISO 27145

billybigpotatoes, 2025-04-09:
I am having a similar issue - i have a UK Sprinter - I have tried all the protocols.  I also see data when I do a scan - but I can not scan PIDs etc.

billybigpotatoes, 2025-04-09:
> [@billybigpotatoes](https://github.com/billybigpotatoes) where do you see the data when you scan?

Sorry I should haver said monitor - 

[image]

billybigpotatoes, 2025-04-09:
Scan results in 

[image]

billybigpotatoes, 2025-04-09:
FYI - been following - https://www.youtube.com/watch?v=hx8UjOLD1tw&t=1341s 

MQTT topic for device coming online is working on MQTT.  But no other events coming through

I uploaded the sprinter.json file from this discussion - it has loaded some PIDs - but I am not getting any messages despite assigning them to MQTT with topics etc.

meatpiHQ, 2025-04-09:
@billybigpotatoes Try to reboot and please do not use the Monitor. Then try again.

billybigpotatoes, 2025-04-09:
Tried that no luck

meatpiHQ, 2025-04-09:
@billybigpotatoes 

- Is the ignition ON? 

- When you use the monitor without the PID scan do you see data all the time?

- Please follow the instructions in the links below to export logs from car scanner app. 

https://meatpihq.github.io/wican-fw/car_scanner/usage

https://meatpihq.github.io/wican-fw/car_scanner/logs

billybigpotatoes, 2025-04-09:
> [@billybigpotatoes](https://github.com/billybigpotatoes)
>
> * Is the ignition ON?
> * When you use the monitor without the PID scan do you see data all the time?
> * Please follow the instructions in the [...]

I will test later and get back to you with results.

richl007, 2025-05-10:
with 4.12 the PID scan now works for me :) 

Thank you

CzarofAK, 2025-07-13:
If got a Sprinter 417 (VS30, 907) Europe version.
I use the Wican USB non pro.

meatpiHQ, 2025-07-13:
@CzarofAK have you tried selecting the car profile in car scanner app? It might be possible that this model use different protocol.

CzarofAK, 2025-07-16:
I tried WiCAN USB and an the blue cheap ELM327 dongle with the torque app, no success. After further reading, it seems to be clear, that the european sprinters use the ISO27145.
Do you plan to add ISO27145 to the WiCAN or WiCAN PRO?

meatpiHQ, 2025-07-17:
I could try to look into ISO27145, it will be great feature but the main issue is that I won't be able to test.

CzarofAK, 2025-07-18:
I am also ready to help with testing or other required measurements.

CzarofAK, 2025-08-20:
@meatpiHQ I don't know if I understood you question correctly. No, i have no "reader"-software in mind. I was hoping to push the data into HomeAssistant...

rosstroha, 2025-09-07:
I have a 2023 sprinter and am able to scan for PIDs. I believe the vehicle has to be turned on with the engine running, however.

CenX-3, 2026-04-03:
@meatpiHQ any update on this?
My European 2025 Sprinter don't show any PIDs

CzarofAK, 2026-09-25:
## Confirmed: EU Sprinter VS30 (Euro VI) works via ISO 27145 on WiCAN-USB (non-Pro), FW 4.21

**Setup**
- Mercedes Sprinter VS30 (W907/910), EU, Euro VI, motorhome chassis
- WiCAN-USB, firmware 4.21, wired to OBD pins 6/14/5/16
- CAN 500K, protocol elm327 over TCP (port 23)

**Key point:** the vehicle does not answer classic OBD-II (`01xx`) on any protocol.
It answers WWH-OBD / ISO 27145 (UDS `22 F4xx`) on 29-bit, 500 kbit/s.

**Working sequence (ELM327 emulation)**
ATZ
ATE0
ATH1
ATSP7
ATCP18
ATSHDB33F1   (functional, 0x18DB33F1)
22F400

**Result**
22F400 => 18DAF158 07 62 F4 00 98 18 A0 13 | 18DAF159 07 62 F4 00 98 18 00 01 | 18DAF15D 07 62 F4 00 98 18 00 01
22F40C => 18DAF158 05 62 F4 0C 0B ED   (763 rpm)
22F485 => 18DAF15D 10 0D 62 F4 85 07 00 1B | 18DAF15D 21 00 1B B4 00 00 00 00   (multi-frame OK)

**Findings**
- Responding ECUs: 0x58 (engine, most DIDs), 0x5D (SCR/AdBlue, only one with F485), 0x59, 0x5A (appears after engine start)
- Physical addressing works: `ATSHDA58F1` -> only 0x58 answers
- `22F4xx` DIDs map 1:1 to SAE J1979 PIDs (same scaling): F405 coolant, F40C rpm, F42F fuel level,
  F442 module voltage, F45C oil temp, F45E fuel rate, F47A DPF pressures, F485 reagent system, F494 NOx inducement
- Multi-frame responses (F47A, F485, F494) are handled correctly by the ELM emulation
- 0x5A returns F40C without the /4 scaling (non-standard), use 0x58 for rpm
- 11-bit (`ATSP6`, `7DF`) returns NO DATA for both `0100` and `22F400`

**Suggestion for firmware:** an "ISO 27145" option for Automate/AutoPID that uses 29-bit `18DB33F1`
and `22F4xx` instead of `01xx` would make EU Euro VI vehicles work out of the box.

Full log attached.

[sprinter_27145.txt](https://github.com/user-attachments/files/32666992/sprinter_27145.txt)

CenX-3, 2026-10-02:
That's good to hear. Then there's hope that I can actually use my WiCAN Pro for my Sprinter (2025) after all. It’s just been gathering dust since I bought it over a year ago.

meatpiHQ, 2026-10-02:
I'm working on it, I will post a firmware here in a couple of days

billybigpotatoes, 2026-10-02:
Amazing - just ordered the pro - may not need it now!

meatpiHQ, 2026-10-02:
@billybigpotatoes The update I'm working on is just for the pro version.

CzarofAK, 2026-10-04:
As I dont have a pro, only the USB, i will try to utilize this as a interface and control it from my HA instance.

@meatpiHQ, if you get that to work, and the other proof it, I will order a Pro (even though for the installation, the usb can be better hidden...

[image]

billybigpotatoes, 2026-10-05:
The Wican Pro arrived today - so I am happy to test - please advise what FW version I should use

meatpiHQ, 2026-10-08:
Here's the new firmware please let me know if it works. Try the new Quick setup this should automatically detect your car protocol.

https://github.com/meatpiHQ/wican-fw/releases/tag/v6.00p_alfa-02

billybigpotatoes, 2026-10-09:
I have upgraded to new firmware - but I can not save changes via setup or manually - I see the following in the logs.  Factory reset also fails 

atus_manager' persist failed: ESP_FAIL
E (935) settings_manager: open /settings/cfg/restart_tracker.tmp failed
E (936) settings_manager: 'restart_tracker' persist failed: ESP_FAIL
E (937) settings_manager: open /settings/cfg/external_storage.tmp failed
E (937) settings_manager: 'external_storage' persist failed: ESP_FAIL
E (938) settings_manager: open /settings/cfg/filesystem.tmp failed
E (938) settings_manager: 'filesystem' persist failed: ESP_FAIL
E (939) settings_manager: open /settings/cfg/http_server_manager.tmp failed
E (940) settings_manager: 'http_server_manager' persist failed: ESP_FAIL
E (952) settings_manager: open /settings/cfg/wifi_manager.tmp failed
E (952) settings_manager: 'wifi_manager' persist failed: ESP_FAIL
W (953) wifi_manager: the access point still uses the factory password; the first settings save has to replace it
I (953) wifi_manager: config applied: mode=3 sta_networks=0 ap_ssid=WiCAN_14c19f458145
E (956) settings_manager: open /settings/cfg/obd_gate.tmp failed
E (956) settings_manager: 'obd_gate' persist failed: ESP_FAIL
E (958) settings_manager: open /settings/cfg/obd_chip.tmp failed
E (959) settings_manager: 'obd_chip' persist failed: ESP_FAIL
E (961) settings_manager: open /settings/cfg/ble_manager.tmp failed
E (962) settings_manager: 'ble_manager' persist failed: ESP_FAIL
E (965) settings_manager: open /settings/cfg/socket_manager.tmp failed
E (965) settings_manager: 'socket_manager' persist failed: ESP_FAIL
E (969) settings_manager: open /settings/cfg/websocket_manager.tmp failed
E (969) settings_manager: 'websocket_manager' persist failed: ESP_FAIL
E (972) settings_manager: open /settings/cfg/log_sinks.tmp failed
E (973) settings_manager: 'log_sinks' persist failed: ESP_FAIL
E (976) settings_manager: open /settings/cfg/bridge_manager.tmp failed
E (976) settings_manager: 'bridge_manager' persist failed: ESP_FAIL
E (978) settings_manager: open /settings/cfg/can_manager.tmp failed
E (978) settings_manager: 'can_manager' persist failed: ESP_FAIL
E (980) settings_manager: open /settings/cfg/j1939.tmp failed
E (980) settings_manager: 'j1939' persist failed: ESP_FAIL
E (982) settings_manager: open /settings/cfg/uds_manager.tmp failed
E (983) settings_manager: 'uds_manager' persist failed: ESP_FAIL
E (985) settings_manager: open /settings/cfg/script_engine.tmp failed
E (985) settings_manager: 'script_engine' persist failed: ESP_FAIL
E (987) settings_manager: open /settings/cfg/led_manager.tmp failed
E (987) settings_manager: 'led_manager' persist failed: ESP_FAIL
E (989) settings_manager: open /settings/cfg/rtc_manager.tmp failed
E (989) settings_manager: 'rtc_manager' persist failed: ESP_FAIL
E (991) settings_manager: open /settings/cfg/imu_manager.tmp failed
E (992) settings_manager: 'imu_manager' persist failed: ESP_FAIL
E (993) settings_manager: open /settings/cfg/battery_monitor.tmp failed
E (993) settings_manager: 'battery_monitor' persist failed: ESP_FAIL
E (995) settings_manager: open /settings/cfg/mdns_manager.tmp failed
E (995) settings_manager: 'mdns_manager' persist failed: ESP_FAIL
E (998) settings_manager: open /settings/cfg/mqtt_manager.tmp failed
E (998) settings_manager: 'mqtt_manager' persist failed: ESP_FAIL
E (1001) settings_manager: open /settings/cfg/mqtt_can.tmp failed
E (1001) settings_manager: 'mqtt_can' persist failed: ESP_FAIL
E (1005) settings_manager: open /settings/cfg/vpn_manager.tmp failed
E (1006) settings_manager: 'vpn_manager' persist failed: ESP_FAIL
E (1007) settings_manager: open /settings/cfg/iperf_manager.tmp failed
E (1008) settings_manager: 'iperf_manager' persist failed: ESP_FAIL
E (1010) settings_manager: open /settings/cfg/sleep_manager.tmp failed
E (1010) settings_manager: 'sleep_manager' persist failed: ESP_FAIL
E (1012) settings_manager: open /settings/cfg/usb_host_manager.tmp failed
E (1012) settings_manager: 'usb_host_manager' persist failed: ESP_FAIL
E (1014) settings_manager: open /settings/cfg/usb_acm_cli.tmp failed
E (1014) settings_manager: 'usb_acm_cli' persist failed: ESP_FAIL
E (1017) settings_manager: open /settings/cfg/espnetlink.tmp failed
E (1018) settings_manager: 'espnetlink' persist failed: ESP_FAIL
E (1020) settings_manager: open /settings/cfg/j2534_server.tmp failed
E (1020) settings_manager: 'j2534_server' persist failed: ESP_FAIL
E (1021) settings_manager: open /settings/cfg/ble_http.tmp failed
E (1021) settings_manager: 'ble_http' persist failed: ESP_FAIL
E (1022) settings_manager: open /settings/cfg/cmdline_manager.tmp failed
E (1022) settings_manager: 'cmdline_manager' persist failed: ESP_FAIL
E (1023) settings_manager: open /settings/cfg/button_manager.tmp failed
[... 135 lines cut ...]
I (28260) wifi:<ba-add>idx:4 (ifx:1, 92:03:ac:65:25:be), tid:1, ssn:83, winSize:64
I (75387) wifi_manager: trying 'Explorer_2.4GHz' on request (channel 6 first)
I (75745) wifi:state: init -> auth (0xb0)
I (75762) wifi:state: auth -> assoc (0x0)
I (75793) wifi:state: assoc -> run (0x10)
I (75936) wifi:connected with Explorer_2.4GHz, aid = 19, channel 6, BW20, bssid = 32:3e:9d:2f:04:0c
I (75936) wifi:security: WPA2-PSK, phy: bgn, rssi: -52, cipher(pairwise:0x3, group:0x3), pmf:0
I (75936) wifi:pm start, type: 0
I (75936) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
[... cut ...]
I (77449) wifi_manager: trial: 'Explorer_2.4GHz' connected, 192.168.8.146, -54 dBm, channel 6, in 2061 ms
E (91183) settings_manager: open /settings/cfg/wifi_manager.tmp failed
E (91183) settings_manager: 'wifi_manager' persist failed: ESP_FAIL
E (100475) settings_manager: open /settings/cfg/wifi_manager.tmp failed
E (100475) settings_manager: 'wifi_manager' persist failed: ESP_FAIL
E (234025) settings_manager: open /settings/cfg/bridge_manager.tmp failed
E (234025) settings_manager: 'bridge_manager' persist failed: ESP_FAIL

meatpiHQ, 2026-10-09:
@billybigpotatoes this should fix the settings issue, the upgrade should still work in your current firmware.

https://github.com/meatpiHQ/wican-fw/releases/tag/v6.00p_alfa-03

billybigpotatoes, 2026-10-10:
> 31\. [@billybigpotatoes](https://github.com/billybigpotatoes) this should fix the settings issue, the upgrade should still work in your current firmware.
>      https://github.com/meatpiHQ/wican-fw/releases/tag/v6.00p_alfa-03

That worked - reading Sprinter data now - for standard codes. is there a map or any custom codes availabe? 

Also - when device is sleeping / van is off - HA stores last report so even tho van is off it is showing RPM at 700.   I could add code to zeo it if reading is over x time old - any other recommendations?

meatpiHQ, 2026-10-10:
@billybigpotatoes Thanks for confirming! Did everything work for you in the quick start? Did you find any issues?

You need to use the latest integration, planning to merge this later

https://github.com/meatpiHQ/ha-wican/tree/meatpi-multi-device

billybigpotatoes, 2026-10-10:
Quickstart worked well - also I tested the home assistant integration and that worked also.

I will do some more methodical testing this weekend.

If there any areas you would like me to focus on?  I will keep an eye open for the merge in future.

[3 more replies]