meatpiHQ/wican-fw #301: Add support for Zeekr 001 Vehicle profile

harkoprinssen, 2025-01-01:
Hi!

I updated my Wican-obd to 4.00 today and tried to get the HA integration to work.
It's working a bit as it recognizes the wican and when changing settings I see it in HA.

I had SoC reading working in 2.98 over mqtt, but cannot get it to work anymore, nor can I get any values to be sent to HA.
MQTT HA discovery setting is disabled whether I enable MQTT or not.

I tried to follow the but cannot get it to read/send any data.
Scanning standard PIDs shows no PIDS.

Attached my logfile where I am 99.9% sure 2248011 is the PID for my SoC.
[log (4).txt](https://github.com/user-attachments/files/18286573/log.4.txt)

Tried using "ATD;ATD0;ATE0;ATH1;ATSP6;ATE0;ATH1;ATM0;ATS0;ATAT1;ATAL;ATST64;ATAR;ATCP18;ATSP6;ATFCSM0;ATSH7DF;0100;ATST64;0100;" for specific initialisation and "ATCP1D;ATSP7;ATCRA1EC6AE80;ATFCSH1DD01635;ATFCSD300000;ATFCSM1" for Init.

[image]

In HA I see this.
I am well aware I should set to auto_pid, however as I cannot get it to work this way I had SoC over mqtt working.
My main concern is the ecu status always stays off.

[image]

I tried making a Zeekr001.json file, but cannot get it to work so any help is greatly appreciated!

harkoprinssen, 2025-01-01:
Update: I cannot get my wican odb to send out any data on the mqtt rx topic either :(

meatpiHQ, 2025-01-01:
> Update: I cannot get my wican odb to send out any data on the mqtt rx topic either :(

There's an enable check box next to the Rx/Tx topic

harkoprinssen, 2025-01-01:
> > Update: I cannot get my wican odb to send out any data on the mqtt rx topic either :(
> 
> There's an enable check box next to the Rx/Tx topic

Ah, I though that was to overwrite the topic :)
Ideally I get the proper way to work so any help on creating a Zeekr 001 vehicle profile would be great.
Once I get SoC to work I should be able to work out the rest and submit

harkoprinssen, 2025-01-01:
> > Update: I cannot get my wican odb to send out any data on the mqtt rx topic either :(
> 
> There's an enable check box next to the Rx/Tx topic

Actually had those enabled but still no data coming in on the rx-topic (anymore). I had it working before.
As stated I'd rather get the proper way to work, according to Car Scanner it should also work for Volvo EX30 and some Smart vehicles

[image]

harkoprinssen, 2025-01-01:
Ok. Ecu status is on all of a sudden. 
still no data though. 

[image]

meatpiHQ, 2025-01-02:
@harkoprinssen I've created a new profile for the zeeker-001. Please try it and let me know if it works.

harkoprinssen, 2025-01-02:
That… is very nice!
It works, had to get the car to wake up and fiddle with some settings a bit but it picked up

How did you manage the formula for the calculation? It differs from the one I used before. Mine came close, Yours is spot-on :)

[image]

meatpiHQ, 2025-01-02:
I used your formula and simplified it :)

harkoprinssen, 2025-01-02:
> I used your formula and simplified it :)

Haha, in that case there will still be some discrepanciea but I will hunt down proper calculations for soc and other values. 

Thanks for your fast help.

harkoprinssen, 2025-01-02:
Trying to wrap my head around understanding which AT commands to use for specific initialisation and PID init.
How do you decided which ones to use?
It seems you rare skipping some from the sequence in the log and using some in a different order.

meatpiHQ, 2025-01-05:
@harkoprinssen 

specific initialization is usually the general initialization could be setting the timeout like command "ATST96" or setting the protocol the car support only one protocol like "ATSP6". If a car support multiple protocol, lets say protocol 6 and 7, then no need to set the protocol in the specific initialization because it might change in the PID init, you might see some profiles with a PID "ATSP6" then the next PID has "ATSP7".