meatpiHQ/wican-fw #304: Use OBDb to add 500+ car profiles

gramss, 2025-01-03:
Just came across this impressive open database about OBD profiles.

https://github.com/OBDb

Would it be possible to merge profiles from here with OBDb?
Interested in ideas about this..

Seems to be [SAEJ1979](https://github.com/OBDb/SAEJ1979) exclusive..?

meatpiHQ, 2025-01-05:
@gramss This would be useful if it was not empty :)

All the repos I've looked at, the json had nothing inside, unless I'm look in the wrong place?

jverkoey, 2025-01-07:
Ahoy both! I'm the founder of the OBDb and happy to answer any questions about it. You're correct that most of the repos are template repos, and that's because the repos also exist as issue trackers for given makes/models. As soon as a car is onboarded to the OBDb a template repo is created so that any extended parameters can be more easily documented there.

Here's some examples of vehicles we've onboarded so far:

- https://github.com/OBDb/Ford-Mustang/blob/main/signalsets/v3/default.json
- https://github.com/OBDb/MG-MG4/blob/main/signalsets/v3/default.json
- https://github.com/OBDb/Porsche-Taycan/blob/main/signalsets/v3/default.json
- https://github.com/OBDb/Hyundai-Tucson/blob/main/signalsets/v3/default.json
- https://github.com/OBDb/Hyundai-IONIQ-6/blob/main/signalsets/v3/default.json

Happy to chat more about this! We're all hanging out in the Clutch Engineering Discord here: https://discord.gg/AdJNJqF5vC

gramss, 2025-01-08:
Hi @jverkoey :) Yeah, it seems I have hit the right repos and did not look through the rest, as I have come from your sidecar app.

meatpiHQ, 2025-01-27:
@jverkoey Do you where apps like Tourque and car scanner get their database from? I'm yet to find a reliable source I can use.

jverkoey, 2025-01-27:
I don't know where they get their databases from, but I suspect it's similar to my approach which is to try to scour the internet for anyone who has managed to find a parameter somewhere already. It's a bit of a treasure hunt 😅

meatpiHQ, 2025-01-28:
Yeah and it seems like they don't like to share as well :). I'm struggling to find information about Ultium EV platform.

Btw I think you can easily write a script to translate WiCAN profiles to your format.

jverkoey, 2025-01-28:
> Btw I think you can easily write a script to translate WiCAN profiles to your format.

Ah cool! Are the WiCAN profiles licensed under the same GPL license as the source code? The OBDb parameter specifications are Creative Commons-licensed as the information itself is more akin to documentation than source code.

meatpiHQ, 2025-01-30:
Yup, the same as the code.

DevKoenv, 2026-10-28:
Update: the OBDb repositories that were mostly empty template repos when this was opened now contain actual signal set data, e.g.:

- [Ford Kuga](https://github.com/OBDb/Ford-Kuga/blob/main/signalsets/v3/default.json)
- [Hyundai Veloster](https://github.com/OBDb/Hyundai-Veloster/blob/main/signalsets/v3/default.json)

Full list of vehicles: https://github.com/orgs/OBDb/repositories?type=all

@meatpiHQ is a WiCAN-profile converter (or direct use of the `signalsets/v3/default.json` files) still something you'd consider? Happy to help test with the Veloster data if useful.