meatpiHQ/wican-fw #305: timing between status and automate topic

micha53, 2025-01-04:
My adapter has the following behaviour:
it is sleeping and I wake it up (@Zoe I open the door)
after some seconds the status topic comes but only about 15 seconds later the automate topic has data.
If it goes offline at first the automate topic becomes invalid {} and later the status topic goes offline.
To use the status topic as indicator for valid automate topic it would be nice when the automate topic still/already is valid when the status topic changes.

meatpiHQ, 2025-01-07:
@micha53 which firmware version are you on?

micha53, 2025-01-07:
still 3.48

micha53, 2025-01-09:
I have watched the following behaviour: If the car drives away from my wifi zone, the automate topic stays with valid inforrmation, seems that retain is activated. 
So also should be the behaviour when the car ist standing with closed doors before going to offline status. But in this situation occurs a topic with empty payload: {} If this can be suppressed, it would be better.

meatpiHQ, 2025-01-09:
@micha53 please try the latest firmware, and let me know if you still have this issue.

micha53, 2025-01-09:
I tried the update, like decribed in the doc....with the wican-fw_obd_v403 file  /directory / .bin
The Website does not come back after 30+ seconds, and does not come back at all as Station.
I still have contact with the device as AP. I see that my station config still is ok, but I also see in the about tab that I am still with version 3.48.
A ping to the station had response times up to more than 3000ms
Repeated the update over the AP, same behaviour: The Website does not come back after 30+ seconds. After waiting some time more I could connect with the AP, but now the station does not respond to the ping anymore.
Need help for the next step!

meatpiHQ, 2025-01-09:
Turn off MQTT reboot and then try to update again.

micha53, 2025-01-10:
Don't know what you mean. I do not find any setting possibility for MQTT reboot .
Can you mark the place you mean?
[image]
[image]
[image]

meatpiHQ, 2025-01-10:
I meant turn off MQTT, sleep mode, and battery alerts then submit changes, the device will reboot after you press the submit button. 

When the device boots again try to update to the latest firmware.

micha53, 2025-01-10:
got the update now.
the behaviour is the same, and worse than with 3.48. With 3.48 after about 15 seconds after going to status online (by opening the door) I got valid automate data.
With 4.03 I only get empty automate topic without data.
Could not yet test while loading the car and when the automate data becomes valid.
[image]

meatpiHQ, 2025-01-11:
@micha53 Please try this firmware, let me know if it works

[wican-fw_obd_v403-3-g71b4d90.zip](https://github.com/user-attachments/files/18386300/wican-fw_obd_v403-3-g71b4d90.zip)

micha53, 2025-01-11:
Today i was looking a bit deeper. As result I got out that since the update I do not get any automate data, only empty {} payload, neither when I open the door, nor when I switch on the ignition. Same behaviour with the new Firmware403-3...
I append my downloaded config file. Maybe something is configured wrong. My setting were still valid after the update.

And there is another issue: 
first I tried to "get latest" car model and select Zoe... again, with the result that I lost all zoe specific config data.
Then I tried to load it manually, with the same result.
I could restore my settings by loading up my downloaded config file.
I remember that I had similar problems with ver 3.48.
[config_2025-01-11(3).json](https://github.com/user-attachments/files/18386360/config_2025-01-11.3.json)

micha53, 2025-01-13:
Last week I could receive CAN/rx data sete, repeating all 12 sets the same data, in accordance with the 12 payload parts. Though I did not get valid automate data.
Today, with the same configuration, I even don't get valid rx-data.
Seems that you need longer time for debugging, so I returned to ver. 3.48. Here my device works quite good.

Still one recommendation, when making an update or reboot, the mqtt status can show these states until finished.

What development tools do you use do write this program? It is C and cannot be the arduino ide.

micha53, 2025-01-25:
Hi,
I would like to ask if I am alone with this problem and if it is still on the agenda. Because ver. 4.0x has a lot of improvements, I am very interested in this new version.

meatpiHQ, 2025-01-25:
@micha53 The empty payload issue is fixed in v4.04.

I'm not sure I understand the other issues you're facing?

For the CAN RX data you should enable the check boxes next to the topic.

micha53, 2025-01-26:
It is a hard thing to get the 4.04, coming from 3.48...
I started this morning, now I am in the afternoon and I have the 4.04 and the winner is....
... not me :-(

I have mqtt enabled again after the update, can see the status and rx topics, but now I do not get the automate topic at all.
And, in the automate section above "Standard PID's" everything but "grouping" is grey and cannot be changed.
But I still have my vehicle specific Zoe_Ph2 model.
So I had a look at the single Payload parts and found that there is a destination and a destination type field for each value. Must I fill in here every topic and what is the default destiation type?
I tried this, but nothing helps, I do not get any automation topic, with 4.03 I still had an empty topic.
I add my configuration file, I assume somewhere is a bit to set the automate settings activated and I hope also the mqtt automate topic.
I think, since ver. 4.00 it is time for an update of the documentation.

[config_2025-01-26xx.json](https://github.com/user-attachments/files/18550460/config_2025-01-26xx.json)