meatpiHQ/wican-fw #306: inaccurate Odometer with fw 4.0 for 2023 Hyundai Ioniq5, all electric vehicle

tung256, 2025-01-06:
MQTT shows "odometer": 61318
actual value of my odometer is 10368 miles

here are my settings:
[image]
[image]
[image]

meatpiHQ, 2025-01-07:
@tung256 Do you get the correct values with the other parameters in the profile?

tung256, 2025-01-08:
@meatpiHQ 
i believe other numbers are fine. Odometer is highly inaccurate.... today i see Odometer value of 1:
[image]

meatpiHQ, 2025-01-08:
@tung256 can you please use car scanner app to log, follow the instructions below to find the odometer logs 

https://meatpihq.github.io/wican-fw/car_scanner/logs

tung256, 2025-01-15:
> @tung256 can you please use car scanner app to log, follow the instructions below to find the odometer logs
> 
> https://meatpihq.github.io/wican-fw/car_scanner/logs

the car is sitting in my driveway and charging now.  in CarScanner app, i dont even see a value for the Odometer. strange
[image]

in MQTT explorer, i see "300"
actual odometer is about 10, 700 miles
[image]

timwelchnz, 2025-01-26:
I have the same issue. 

I'm on FW 4.03 and everything is pulling through fine but the odometer is reading 300 and not moving.

I note that the vehicle profile uses PID 22B0023 whereas the [TorquePro CSV](https://github.com/Esprit1st/Hyundai-Ioniq-5-Torque-Pro-PIDs/blob/main/TorqueIONIQ5AWD77kWh.csv) uses 22B002 on (line 317) - although I tried to add this as a custom PID and it doesn't seem to work either.

meatpiHQ, 2025-02-08:
@timwelchnz After checking the torque CSV, seems there's an issue with the profile for the odometer. We should add a pid_init string for the odometer PID.

meatpiHQ, 2025-02-13:
I've fixed the profile, click on "Get latest" and give it a go.

tung256, 2025-02-20:
@timwelchnz  same. im on fw 4.03.  odometer was removed after i got the latest

[image]

[image]

attilaabert, 2025-02-21:
Same here. With v2.92 TPMS/ODO read was working.
With v4.04 sadly not.

Here a Carscanner log:
>ATSH7C6
OK

>22B002
NO DATA

>22B002
NO DATA

>22B002
NO DATA

>22B002
NO DATA

meatpiHQ, 2025-02-25:
@tung256 
@attilaabert 
@timwelchnz 

Can you please tell me the make, model, and year of your vehicle? seems like there is a slight difference.

I've reverted back to the old profile. @attilaabert let me know if it working now.

attilaabert, 2025-02-25:
@meatpiHQ 

My car is an 2020 Hyundai Kona.

I have downgraded to fw2.92. 
Strange but odometer reading work neither.

I have tried your reverted car profile:
`Hyundai: Ioniq5/Ioniq6/Kona
{
  "SOC_BMS": 62.5,
  "Max_REGEN": 0,
  "Max_Power": 0,
  "Batt_Current": 6553.3,
  "HV_Volts": 369,
  "HV_Power": 2418786,
  "Batt_MaxT": 11,
  "Batt_MinT": 10,
  "Batt_Temp_1": 10,
  "Batt_Temp_2": 10,
  "Batt_Temp_3": 10,
  "Batt_Temp_4": 10,
  "Batt_Temp_5": 0,
  "Batt_InletT": 12,
  "Max_Cell_V": 3.76,
  "Max_Cell_V_No": 30,
  "Min_Cell_V": 3.76,
  "Min_Cell_V_No": 7,
  "Aux_Batt_Volts": 14.6,
  "Cum_Energy_Charged": 17010.5,
  "SOH": 100,
  "SOC_D": 64.5,
  "Min_Cell_Det_No": 0,
  "Max_Cell_Det_No": 0,
  "Min_Cell_Det": 2.2,
  **"odometer": 0**
}

Kia/Hyundai: Niro/Soul/Kona
{
  "SOC_BMS": 62.5,
  "Max_REGEN": 0,
  "Max_Power": 0,
  "Batt_Current": 6552.6,
  "HV_Volts": 369,
  "HV_Power": 2418167.75,
  "Batt_MaxT": 11,
  "Batt_MinT": 10,
  "Batt_Temp_1": 10,
  "Batt_Temp_2": 10,
  "Batt_Temp_3": 10,
  "Batt_Temp_4": 10,
  "Batt_InletT": 12,
  "Max_Cell_V": 3.76,
  "Max_Cell_V_No": 30,
  "Min_Cell_V": 3.76,
  "Min_Cell_V_No": 7,
  "Aux_Batt_Volts": 14.6,
  "SOH": 100,
  "SOC_D": 64.5,
  "Min_Cell_Det_No": 0,
  "Max_Cell_Det_No": 0,
  "Min_Cell_Det": 2.2,
  "GearSelector_Raw": 33,
  "Speed_Vehicle": 0,
  "Car_Ready": 0,
  "Car_ParkBreak": 1
}`

But my old node-red flow works!! :)
I send this through mqtt:
`{"bus":"0","type":"tx","ts":29535,"frame":[{"id":1990,"dlc":8,"rtr":false,"extd":false,"data":[3,34,176,2,0,0,0,0]}]};`
After 100ms that:
`{"bus":"0","type":"tx","ts":42070,"frame":[{"id":1990,"dlc":8,"rtr":false,"extd":false,"data":[48,0,10,170,170,170,170,170]}]};`
Then I can read the odometer:

[image]

I realy dont know what the second message is. I call it a "magix frame" :)

timwelchnz, 2025-02-26:
> Can you please tell me the make, model, and year of your vehicle? seems like there is a slight difference.
> 
> I've reverted back to the old profile. [@attilaabert](https://github.com/attilaabert) let me know if it working now.

@meatpiHQ - Hyundai IONIQ 5 73KWH/EV/4WD/FD 2022

timwelchnz, 2025-03-05:
> [@tung256](https://github.com/tung256) [@attilaabert](https://github.com/attilaabert) [@timwelchnz](https://github.com/timwelchnz)
>
> Can you please tell me the make, model, and year of your vehicle? [...]

I restored the old profile and this does indeed return the odometer but it is again incorrect at 300
`{"SOC_BMS":74.5,"Max_REGEN":0,"Max_Power":0,"Batt_Current":6553.6,"HV_Volts":707.9,"HV_Power":4639293.5,"Batt_MaxT":20,"Batt_MinT":18,"Batt_Temp_1":19,"Batt_Temp_2":19,"Batt_Temp_3":19,"Batt_Temp_4":20,"Batt_Temp_5":20,"Batt_InletT":60,"Max_Cell_V":3.92,"Max_Cell_V_No":175,"Min_Cell_V":3.92,"Min_Cell_V_No":166,"Aux_Batt_Volts":12.9,"Cum_Energy_Charged":6301.6,"SOH":100,"SOC_D":75.5,"Min_Cell_Det_No":0,"Max_Cell_Det_No":3,"Min_Cell_Det":2546,"odometer":300}`