meatpiHQ/wican-fw #313: Unable to get Rear Tire Pressure with the correct TX for Ora Funky Cat

spicydog, 2025-01-11:
I'm working on getting sensors of Ora Funky Cat for Home Assistant with WiCAN + MQTT using Car Scanner as a producer.

I already have several sensors that may share to the community including:
- soc
- soe
- hv_bus_current
- hv_power
- hv_voltage
- coolant_temperature
- tire_pressure_front_left
- tire_pressure_front_right
- mileage
[image]

Currently I'm working on Tire Pressure, I have processed a lot of data and already have them ready for Node-RED.
This is my working Google Sheet: https://docs.google.com/spreadsheets/d/1FYhf-c3pWxY5REcZS4yKfbZzTXARMbKtWPH8PJFeL-s/

This is the TX and RX

# Front Tire Pressure
TX (WORKING)
`{"bus":"0","type":"tx","ts":0,"frame":[{"id":1900,"dlc":8,"rtr":false,"extd":false,"data":[3,34,2,16,170,170,170,170]}]}`
RX (WORKING)
`{"bus":"0","type":"rx","ts":0,"frame":[{"id":1964,"dlc":8,"rtr":false,"extd":false,"data":[16,11,98,2,16,170,74,167]}]}`
// tire_pressure_front_left: b[5]*0.2-0.22
// tire_pressure_front_right: b[7]*0.2-0.22

# Rear Tire Pressure
TX (NOT WORKING)
`{"bus":"0","type":"tx","ts":0,"frame":[{"id":1900,"dlc":8,"rtr":false,"extd":false,"data":[48,0,0,170,170,170,170,170]}]}`
RX (Received from Car Scanner)
`{"bus":"0","type":"rx","ts":0,"frame":[{"id":1964,"dlc":8,"rtr":false,"extd":false,"data":[33,75,172,75,166,74,0,0]}]}`
// tire_pressure_rear_left: b[2]*0.2-0.22
// tire_pressure_rear_right: b[4]*0.2-0.22

I got *Front* Tire Pressure working correctly; however, not for the *Rear* Tire Pressure.

For the *Rear* Tire Pressure, I sent `{"bus":"0","type":"tx","ts":0,"frame":[{"id":1900,"dlc":8,"rtr":false,"extd":false,"data":[48,0,0,170,170,170,170,170]}]}` but have get back. (The RX is one that I received when using Car Scanner.

Not sure how to get the *Rear Tire Pressure TX* working correctly. Not sure if it about bit length or not.

If you need any information to investigate more please let me know.

Thank you in advance. ❤️

meatpiHQ, 2025-01-11:
@spicydog how about we create a profile for the Funky cat? :)

Can you please export the logs for the SoC first, check out the instruction below

https://meatpihq.github.io/wican-fw/car_scanner/logs

spicydog, 2025-01-11:
@meatpiHQ I think someone already did for Funky Cat SoC in this discussion.
https://github.com/meatpiHQ/wican-fw/discussions/69#discussioncomment-7331819

Nevertheless, this is my log data. The current SoC is 72.3%
[2025-01-11 log.txt](https://github.com/user-attachments/files/18386473/2025-01-11.log.txt)

Let me know if you need any further information.

meatpiHQ, 2025-01-12:
@spicydog I've created a profile, please update to the latest firmware version and press on get latest button in the Vehicle Specific PID.

spicydog, 2025-01-17:
Dear @meatpiHQ first of all, is there any to sent TX and receive Rear Tire Pressure as I request in the topic.
I guess it may the range is bytes might be double but I have no idea to to make the TX longer.

Thank you in advance.

meatpiHQ, 2025-01-20:
@spicydog Can you please repeat the car scanner logs for the Tire Pressure? we have to do one by one.

spicydog, 2025-01-20:
@meatpiHQ Here is the result.
The display value are
```
front-left:	33.58psi
front-right:	32.98psi
rear-left:	33.78psi
rear-right:	32.79psi
```

It seems like it response all the same but with 2 lines
```
>2202102
7AC100B620210A94AA6
7AC2149AA4BA54A0000
```

This is approximate decoding exp I have done. (hopefully i might help.
```
msg.rx_type = 'tire_pressure_front_left'
const output = b[5]*0.2-0.22;
```
```
msg.rx_type = 'tire_pressure_front_right'
const output = b[7]*0.2-0.22;
```
```
msg.rx_type = 'tire_pressure_rear_left'
const output = b[2]*0.2-0.22;
```
```
msg.rx_type = 'tire_pressure_rear_right'
const output = b[4]*0.2-0.22;
```
The **sample** (not for the data above) from TX I get from the elm327 mode and MQTT.
```
FL 34.18
FR 33.58
RL 34.57
RR 33.38
{"bus":"0","type":"rx","ts":0,"frame":[{"id":1964,"dlc":8,"rtr":false,"extd":false,"data":[16,11,98,2,16,172,75,169]}]}
{"bus":"0","type":"rx","ts":0,"frame":[{"id":1964,"dlc":8,"rtr":false,"extd":false,"data":[33,75,174,76,168,75,0,0]}]}
```

The logs in the attachments.
[front-left-33.58psi-log.txt](https://github.com/user-attachments/files/18480016/front-left-33.58psi-log.txt)
[front-right-32.98psi-log.txt](https://github.com/user-attachments/files/18480018/front-right-32.98psi-log.txt)
[rear-left-33.78psi-log.txt](https://github.com/user-attachments/files/18480017/rear-left-33.78psi-log.txt)
[rear-right-32.79psi-log.txt](https://github.com/user-attachments/files/18480019/rear-right-32.79psi-log.txt)

meatpiHQ, 2025-01-21:
@spicydog I've update the profile with these new parameters, please check.

spicydog, 2025-01-21:
@meatpiHQ Is there a way to make it work with MQTT. I am trying to write TX json but it only work with front tyre.

Furthermore, do you have any document to guild me how to write for the profile. Currently I'm using MQTT and finding the way to convert it to the profile.

Thank you in advance.

meatpiHQ, 2025-01-21:
I'm sorry, I'm not sure I understand. Why not use the profile with MQTT? 

The reason it only works for the front and not the rear is because this is a multi response PID. That means you will get more than one response, but you'll need to send a flow control frame after you receive the first frame. And the rear pressure values are on the 2nd frame.

This is done automatically by the profile and autopid. 

I haven't fully documented how to create profile yet, but I'm hoping I can finish it soon .

spicydog, 2025-01-21:
Well, I am not sure what does work and what does not for WiCAN configuration, there are so many combination with unknown result because this is my first time too.
Initially I followed some guides on the internet, which might be for the old version, and I get it work when I set to elm327 + MQTT with debug disabled where I can send JSON to TX and sub for RX via MQTT.

I think I did once but not sure that with profile I have to set to AutoPID and with AutoPID, my previous working method will not work. So I continue with elm327. However, I will retest again.

Nevertheless, today the car is not with me so I cannot test, so I will do it again tomorrow evening and will let you know the results.

Lastly, looking forward for the document so I might help you build the profile as well.

Thank you :)

meatpiHQ, 2025-01-23:
@spicydog sorry about that, it was not pushed to main. It is available now, please check.

spicydog, 2025-01-23:
@meatpiHQ everything's is working! I'm learning to build custom_pid, I think next I will try myself and will try send the push request for you.

[image]

Anyway, is there anyway I and test the vehicle_profiles.json, the upload feature is not working in the web console. When I upload it removes everything. Not sure if 3.48 is the latest version or could you reproduce?

spicydog, 2025-01-23:
Last one, is it possible to make Custom PIDs as schema, like export and import with json text?
This would be helpful to with working with it. Or anyway to make a backup with Custom PIDs?

[image]

meatpiHQ, 2025-01-24:
@spicydog 

> Anyway, is there anyway I and test the vehicle_profiles.json, the upload feature is not working in the web console. When I upload it removes everything. Not sure if 3.48 is the latest version or could you reproduce?

in the latest firmware you can upload the profile directly.

> Last one, is it possible to make Custom PIDs as schema, like export and import with json text?
> This would be helpful to with working with it. Or anyway to make a backup with Custom PIDs?
> 

Yes, you can download all the device settings from the "System" Tab.

If you want to learn more about ELM327 AT commands you should checkout the datasheet

[ELM327DS.pdf](https://github.com/user-attachments/files/18536578/ELM327DS.pdf)

spicydog, 2025-01-28:
@meatpiHQ Thank you for the information.
This is what I have work so far on CusomPIDs.

Name | Unit | Init | PID | Expression
-- | -- | -- | -- | --
hv_battery_soc | % | ATSH78A;ATFCSM1;ATCRA7CA; | 22D0621 | [B4:B5]/10
hv_battery_soe | kWh | ATSH78A;ATFCSM1;ATCRA7CA; | 22D0621 | ((B6*256)+B7)/10
  |   |   |   |  
hv_bus_current | A | ATSH78A;ATFCSM1;ATCRA7CA; | 22D061 | -((B4*256)+B5-8000)/10
hv_bus_power | kW | ATSH78A;ATFCSM1;ATCRA7CA; | 22D061 | -((([B4:B5])*0.034257)-274.16)
  |   |   |   |  
coolant_temperature | °C | ATSH78A;ATFCSH78A;ATFCSM1;ATCRA7CA; | 22D0311 | ((B6*256+B7)-400)/10
  |   |   |   |  
tire_pressure_front_left | psi | ATSH76C;ATFCSH76C;ATCRA7AC; | 2202102 | B5*0.2-0.22
tire_pressure_front_right | psi | ATSH76C;ATFCSH76C;ATCRA7AC; | 2202102 | B7*0.2-0.22
tire_pressure_rear_left | psi | ATSH76C;ATFCSH76C;ATCRA7AC; | 2202102 | B10*0.2-0.22
tire_pressure_rear_right | psi | ATSH76C;ATFCSH76C;ATCRA7AC; | 2202102 | B12*0.2-0.22
  |   |   |   |  
tire_temperature_front_left | °C | ATSH76C;ATFCSH76C;ATCRA7AC; | 2202102 | B6-50
tire_temperature_front_right | °C | ATSH76C;ATFCSH76C;ATCRA7AC; | 2202102 | B9-50
tire_temperature_rear_left | °C | ATSH76C;ATFCSH76C;ATCRA7AC; | 2202102 | B11-50
tire_temperature_rear_right | °C | ATSH76C;ATFCSH76C;ATCRA7AC; | 2202102 | B13-50
  |   |   |   |  
odometer | km | ATSH78B;ATFCSH78B;ATCRA7CB; | 22D0041 | [B5:B6]

meatpiHQ, 2025-01-28:
@spicydog thank you. I'll add these to the profile.

closed 2025-02-15 by spicydog