meatpiHQ/wican-fw #316: New "car" request: Citroen Ami

drmakris, 2025-01-13:
I am trying to create a profile for the Citroen Ami to get parameters to Home Assistant. To be honest I don't know what I am doing but I am willing to learn. Based on parameters from https://github.com/stwoowts/Citroen-Ami-Opel-Rocks-e-OBD I came up with this:
[ami.json](https://github.com/user-attachments/files/18400935/ami.json)
I am obviously doing something wrong since although I loaded the profile in WiCAN the parameters appear as unavailable in Home Assistant. I am not sure if the AT commands are correct (copied from another Stelantis car) and/or the PIDs were supposed to be written like this. Any help will be much appreciated.

drmakris, 2025-01-15:
Thank you @meatpiHQ for your answer.

It seems that the car is indeed just broadcasting and not responding to OBD commands since I couldn't set any protocol in car scanner to connect to the ECU. WiCAN connected fine through WiFi. No ECU connection, no sensors to log, no logs unfortunately in car scanner. Is there a way to just sniff on the CanBus? There was a reference to a "terminal" mode but I couldn't find it.

What "works" is the "Monitor" tab in WiCAN's Web UI which produced the following while the car was charging:

[Monitor Settings.pdf](https://github.com/user-attachments/files/18430134/Monitor.Settings.pdf)

No way to save these logs either. Do you see a way to get these values out?

meatpiHQ, 2025-01-15:
Did you set the ECU protocol manually in car scanner?

https://meatpihq.github.io/wican-fw/car_scanner/usage#_5-set-the-ecu-protocol

drmakris, 2025-01-15:
Yes, tried the suggested ones. Perhaps I missed some? Wasn't patient enough. Should I try again?

To elaborate on my previous message it seams that the values in "Monitor" are correct, I could verify the VIN number from the 712h 711h 710h IDs and also seems that the values are least significant bit first (16bit).

meatpiHQ, 2025-01-15:
I think you should try again. It's very unlikely that OBD protocol is not supported. Did you set protocol to elm327? Is MQTT enabled?

Please post a screenshot of your device configuration webpage.

drmakris, 2025-01-16:
That's interesting. In this video the protocol is referenced as CAN_KWP2000 - (ISO 11898-1). So after all it does have a protocol. I don't see it in car scanner.

https://youtu.be/9fTK4xGRa7Y?si=XKDXvQgOGChEPgE4&t=384

drmakris, 2025-01-16:
Nope, does not work. Need ISO 14230 with the high bitrate extension of ISO 11898.

meatpiHQ, 2025-01-20:
It's really unusual that OBD protocol is not supported... unless this is really a tin on wheels :)

I think the best way is to use CAN filters, until I add support for CAN broadcast in AutoPID.

https://meatpihq.github.io/wican-fw/mqtt/usage#_4-can-to-json-interpreter-filtering

Keep the CAN baudrate at 500K.
Try this filter, it will publish the SoC to MQTT every 5 seconds

**CAN ID:** 1408
**Name:** SoC
**PID:** -1
**Index:** 0
**Start Bit:** 0
**Bit Length:** 1
**Expression:** B6
**Cycle:** 5000

drmakris, 2025-01-24:
Since I want a little more flexibility (I intent to create a full dashboard for the car) I proceeded to make a script that connects on one side to WiCAN through Websockets and on the other to an MQTT Server. I managed to get some values out and feed them to Home Assistant. I proudly present:

[image]

The only issue I have is that I don't know how WiCAN will behave. Could you please, @meatpiHQ, provide some insight. Is the WiCAN module stressed by this? Why does it have to reboot after enabling the "Monitor" tab in the Web UI?

ghadafri, 2025-05-20:
Did anyone find the SOH or battery capacity Pids of the citroën ami?

drmakris, 2025-06-14:
@ghadafri this is the info I managed to pull into home assistant:

[image]

drmakris, 2025-06-20:
Added some more info if anyone is interested:

[image]

meatpiHQ, 2025-07-17:
@baraoresmungao you don't have to use MQTT you can use WiCAN custom integration. If you want to use MQTT you have to add the sensors manually. 

https://meatpihq.github.io/wican-fw/home-assistant/integration_setup

baraoresmungao, 2025-07-18:
Got it, I created all by hand and it works.
@ghadafri where did you get the range, I don’t see it on the parameters from @drmakris!