meatpiHQ/wican-fw #318: Honda Prologue - New Vehicle

Ncgodwod, 2025-01-17:
Could use some help. This domain type of totally new to me but I do a lot of bash and yaml…

[log.txt](https://github.com/user-attachments/files/18447480/log.txt)

meatpiHQ, 2025-01-20:
@Ncgodwod Could you please export the car scanner logs again following the instructions in the link. Lets start with the SoC first.

https://meatpihq.github.io/wican-fw/car_scanner/logs

Ncgodwod, 2025-01-20:
[Odometer.txt](https://github.com/user-attachments/files/18481060/Odometer.txt) = 498 miles or so
[Remaining distance.txt](https://github.com/user-attachments/files/18481061/Remaining.distance.txt) = 89 miles or so
[SOC log.txt](https://github.com/user-attachments/files/18481062/SOC.log.txt) = 36%
[Battery remaining percentage.txt](https://github.com/user-attachments/files/18481059/Battery.remaining.percentage.txt) = 36%

Here ya go. Followed the directions to a tee.

Ncgodwod, 2025-01-20:
[Charging DC current Amps (31-30).txt](https://github.com/user-attachments/files/18481349/Charging.DC.current.Amps.31-30.txt)
[Bat 1 SOC ( 41.18_).txt](https://github.com/user-attachments/files/18481351/Bat.1.SOC.41.18_.txt)
[Dc Charge KW (9-9.12 KW).txt](https://github.com/user-attachments/files/18481350/Dc.Charge.KW.9-9.12.KW.txt)

meatpiHQ, 2025-01-22:
@Ncgodwod Did you select the sensor in Car scanner before unplugging the adapter? I see a lot of different PIDs in log before the last disconnection.

[image]

Ncgodwod, 2025-01-22:
Yes.I sat on the sensor for multiple minutes to make sure that was the last one out per the documentation.Sent from my iPhone

image.png (view on web)

Ncgodwod, 2025-01-23:
Im guessing that these should all look more like the odometer log?

Ncgodwod, 2025-01-27:
[Dc Charge KW (9-9.12 KW).txt](https://github.com/user-attachments/files/18554226/Dc.Charge.KW.9-9.12.KW.txt)
[Bat 1 SOC ( 63_).txt](https://github.com/user-attachments/files/18554227/Bat.1.SOC.63_.txt)

Yeah, I can confirm these both followed the procedures 100%.

meatpiHQ, 2025-01-27:
@Ncgodwod the KW log looks good, but the SoC is still bad. Maybe it's not anything you did wrong it's just car scanner is requesting some extra data.

Try to find SoC PiDs for "Ultium EV vehicles" seem car scanner is using this profile.