meatpiHQ/wican-fw #319: Monitor Data and Home Assistant

justynbell, 2025-01-17:
I'm having trouble tying everything together, and maybe someone can help me:

All I'm really interested in is my car's Odometer reading, and getting mileage into Home Assistant so I can set alerts for regular service maintenance tasks.

I had everything working with MQTT, but I upgraded my firmware to 4.03 (from much older version) and also see that there's great progress on the Home Assistant Integration (huge shout out for that, by the way, that's great work), and would like to use the integration instead of MQTT.

I can get my odometer reading by looking at the Monitor page:

[image]

I know that 0xDFE5 is 57317, which is my car's mileage.  However, there's no PID column on that page.  I'm not a CAN expert, and tried to do some research, but I'm not sure where the PID is, or its relation to the "ID" on the "Monitor" page.  I was wondering if I can get the Custom PID section of the Automate page to send me my Odometer data that I can see from the Monitor page.

But it also sounds like the Monitor page listens/sees unsolicited CAN data, where the Automate page is for "requests" to the ECU for data.

Basically, I have my odometer reading from the Monitor page, it worked with MQTT, and can I use the Automate page to send data via the HA integration?  Or is this not possible, or am I misunderstanding everything?

I also followed the instructions for getting the Car Scanner app and played with it a little bit last night, but my car (a Nissan Sentra 2019 with the 1.8) is not in the database of the app, and I was not able to see odometer readings at all (however, I'm new to the app and only played with it for ~30 minutes or so).  At some point (hopefully soon), I'll pull the logs and contribute my car to the list of supported vehicles, but just getting mileage for now would be huge for me.

meatpiHQ, 2025-01-20:
@justynbell If all you need is the Odometer data I suggest you try the standard PID. Checkout this video and see if the odometer is available over standard PID. If not we can try a different method.

https://youtu.be/hx8UjOLD1tw?t=785

justynbell, 2025-01-21:
I did check out the standard PID options and unfortunately, the Odometer doesn't pop up for my car.

Since all I really want/need is odometer info, I'm able to get it by sending MQTT data, parsed from the CAN "monitor" output, and it works.  I was just hoping to get it into the pretty WiCAN HA integration.

I don't know much about CAN data or the protocol at all.  So you can send requests to your ECU for some data, but some data is just "broadcasted" all the time; and that's the data that shows up on the monitor page?

Would the ability to send that "monitor" data via http to the HA integration sort of break the paradigm you have set up?  Or is it just a feature request or a PR away? ;)

meatpiHQ, 2025-01-22:
> Would the ability to send that "monitor" data via http to the HA integration sort of break the paradigm you have set up? Or is it just a feature request or a PR away? ;)

I prefer not, I want to keep the processing on the device as much as possible. Adding this feature to the profile will allow users to share profiles and get things working out of the box.

There are better ways than the monitor. 

Here is a better solution for now.

Add this CAN filter and it will send your odometer value to mqtt.

[image]

justynbell, 2025-01-28:
> I prefer not, I want to keep the processing on the device as much as possible. Adding this feature to the profile will allow users to share profiles and get things working out of the box.

I agree with that, I'm not saying send all that crazy data to HA.  I'm saying once you set up a filter, and you're sending that data to HA via MQTT at some interval (5 seconds in your example), can you send it via HTTP to the HA integration instead/as well?

Basically in my ideal world, I'd like to solely use the HA integration and ditch MQTT: it would be much better/easier for automations, etc.

jay-oswald, 2025-03-12:
@justynbell Try the new version of HA integration, should just work. Even on the older verions I've not been using MQTT for months, only via HA integration