meatpiHQ/wican-fw #322: "Problems" with FW 4.03

Heavendenied81, 2025-01-18:
I updated today to FW 4.03 and want to say that the new "look and feel" is great.
But i struggled a little bit with two things:

1. After updating from FW 3.48 to FW4.03 the Dongle did not connect to my WiFi. After connected to the wican AP i recognized that WPA3 was enabled insstead of WPA2 which would be correct and used before. After changing back to WPA2 it worked. Maybe system should check what was configured before and use that?
2. Using multiple entrys in Automate->Vehicle specific area is working. But if I change one of them the entry "send_to" disappears and if I store and reboot I can see that in fact the entry was deleted. Could you check this?

Thanks in adavance and hopefully this helps improving this great project!

justynbell, 2025-01-21:
> After updating from FW 3.48 to FW4.03 the Dongle did not connect to my WiFi. After connected to the wican AP i recognized that WPA3 was enabled insstead of WPA2 which would be correct and used [...]

FWIW, that threw me off for a little bit too.  After the upgrade I had to go into the device via it's AP mode, and change that as well.

meatpiHQ, 2025-01-22:
@Heavendenied81

> Using multiple entrys in Automate->Vehicle specific area is working. But if I change one of them the entry "send_to" disappears and if I store and reboot I can see that in fact the entry was deleted. Could you check this?

Can you please post a screenshot of this issue? I'm unable to reproduce, I assume you mean the custom PIDs because that's the only one with "Send_to" field.

Heavendenied81, 2025-01-22:
Thanks for your response.
Here are two screenshots.
1.
Filling in everthing and click on "store"

[image]

2.
After automatic reboot:

[image]

meatpiHQ, 2025-01-26:
@Heavendenied81 i was finally able to reproduce this issue. The reason is that you have duplicate PIDs.

meatpiHQ, 2025-01-26:
This should fix it.

[wican-fw_obd_v404-1-g0921185.zip](https://github.com/user-attachments/files/18548794/wican-fw_obd_v404-1-g0921185.zip)

commit 09211854 (2025-01-26): duplicate pids topic fix #322