meatpiHQ/wican-fw #323: Need some 12V Battery values from VW:ID

cr0i, 2025-01-18:
Hello @meatpiHQ ,

because I had problems with my 12V battery, I want some more values from this.
In Carscanner I can see 12V Battery current or 12V Battery total discharge, that would be interesting.
The init-Strings I can extract me from the Carscanner Logfile, but I don´t find anything about an expression.
Both values are from this PID:

>222AF74
77A101A622AF71F68FF
77A21FFDEAA0C472C2D
77A22262300006CA800
77A230068B8644431AA

12V Battery total discharge was 2680,8 Ah at this moment, Battery current something around -0,7 A.

Can you help with this?

meatpiHQ, 2025-01-25:
@cr0i did you figure this out? If not and you still need it please reopen the issue. For the expression you could either find a reference or try and guess it by recording multiple values.

cr0i, 2025-01-26:
I didn´t found a reference and the value changes very fast in carscanner.

[log 27.txt](https://github.com/user-attachments/files/18548686/log.27.txt)

In this log the 12V current was beetween -0,6A and -0,8A.
Can I export the values logged by live data from carscanner too?
Perhaps this could help to guess it better?

I found a reference for 12V voltage, wich is in the same PID, but this is wrong, I got 66,04V:
([B4:B5])/1024+4.26

cr0i, 2025-01-26:
This should be the corresponding CSV for log27.txt above:

[Log 27.csv](https://github.com/user-attachments/files/18548767/Log.27.csv)

meatpiHQ, 2025-01-26:
@cr0i For the battery voltage try: ([B5:B6])/1024+4.26

The current is harder to find, but I know that the value is in B14, B13 and B12

cr0i, 2025-01-26:
In the meatime I found other values, that could be interesting for my problem occured, like chargedtotal, dischargedtotal and SoC (12V) too.
But in the moment I analyse this with hv-batt_energycontent, that I can request without problems.
If the 12V battery is empty, it will be charged from the hv-batt, so I can see this indirectly.

So it is not so important, that we find this values, even it would be nice. ;-)

cr0i, 2025-01-26:
Hello @meatpiHQ ,

12V Battery Voltage is now correct.
In this PID 222AF7 there are 4 values "hidden", that are interesting:
12V Voltage
12V Current
12V Total Discharged
12V Total Charged

Here is another example with corresponding export of the values carscanner reported at this time:

[log 31.txt](https://github.com/user-attachments/files/18551943/log.31.txt)

[Log 31.csv](https://github.com/user-attachments/files/18551953/Log.31.csv)

I found some other PIDs, that are working for VW:ID.
Perhaps some of the calculating here helps in finding, how the 12V values are calculated?

[vehicle_profiles (7).json](https://github.com/user-attachments/files/18551964/vehicle_profiles.7.json)

cr0i, 2025-01-26:
Hello @meatpiHQ ,

and here the next Logs, there are some other values, like Temperature and SoH of 12V Battery too in this PID.
I keeped all of them in the CSV, perhaps it helps separating all the values and some could be interesting too.

This log is in the moment, when charging the 12V Battery occured, so we have higher current and a rising of Total charged.

[log 32.txt](https://github.com/user-attachments/files/18552057/log.32.txt)

[Log 32.csv](https://github.com/user-attachments/files/18552063/Log.32.csv)

typxxi, 2025-03-16:
looks like you are dealing with the old issue from the days when VW introduced the start stop automatic for ICE cars.

There is a battery monitor keeping track of the amount of Ah or Wh which are going into a battery and what goes out.
The usual issues in the past had been a drift when the cars finally says that it needs a new 12V battery which can be wrong. 

You simply need to pull the cars plugs from the battery, then charge it again and then reconnect the fully charged battery to the car. 
Finally and quite  often still forgotten by the mechanics you have to use odb tools to change the serials number of the battery. Yes, of cause everyone without a clue will now shout that it is a stupid lead acid battery and the serial does not matter, but that is wrong.

The "new battery serial" is a trigger which causes a reset of the shunt and meter which means it tells the car that the battery is now full and also what full means cause the 12V battery size is also part of the whole calculation where some mechanics do not adjust the values. If they replaced a 60 Ah battery with a 55 or 45 Ah battery in the past there was no big deal: bigger better, smaller would only mean that it could run empty earlier.

Anyhow: from the introduction onwards the car was calculation the current capacity based on the full capacity when the battery had been reconnected and gotten a new serial as the trigger. Fully charged did not improve the general situation.

From that moment on the car would constantly calculate the current state of the Wh stored. If you got a 60 Ah battery and it was fully charged, then you had 720 Wh capacity. You start the engine and the smart shunt will measure how much energy had been taken out of the battery for example 18 Wh and therefore 702 Wh left.
Driving into the city the charger would slowly charge maybe by 7 Wh and then when you parked the battery energy counter was at 709 Wh. Next start to drive to the hospital and 15 Wh might be gone and 704 Wh left.
1 km down to the parking of the hospital you had recharged 6 Wh and had 700Wh left.

And this was an important thing back in the days that the start stop automatic would never stop the engine from running at a red light when there was not enough juice left in the battery to start that car safely again.

This is the heritage of the energy or colombmeter in the VW cars and the reason, why the serial of the battery had to be changed once that had been fully charged outside of the car. 
Smart shops did another trick and simply used the current value like abcd12380 and simply counted for each reset the last digit 1 up.
Hence everyone in the shop later would instantly knew about how often the battery had been recharged cause a car they had recharged would have at the last digit a counter. No matter what they put a 1 in as the last digit once the battery had been recharged for the first time. 
And this story belongs to all the ID cars cause even e-Up and e-Golf and especially ID first suffered from an empty 12V battery which was not that empty nor that dead - if you knew how to solve it properly. Most mechanics of the old world still believed in the Multimeter and its figures like 12,8V and then meant the car has to start. It has not if the internal capacity counter is below 300 Wh for example. It would not run again untill you charge it up again and reset the counter.