meatpiHQ/wican-fw #326: HTTP PROXY

gerhardvilsecker, 2025-01-22:
I would like to use wican/mqtt for sending car data to homeasistant while driving (currently i am using torque). Since your mqtt stack is currently not encrypted I would like to use a wireguard vpn connection (ha to car mobile) to secure mqtt.
Question: is it possible to configure a http proxy in wican? If yes it should be possible to redirect wican mqtt traffic to vpn Tunnel of car mobile using local installed webproxy like every proxy...

jomud9, 2025-01-27:
If you are already have WG, you could already allow HA integration to access to WiCon. That's what I am using. 

Although it is a bit of work as you need to set your WG connection as a Site-2-Site.

gerhardvilsecker, 2025-01-27:
Hello jomud9, thanks for your answer. It is good to know, that someone has managed to get vpn connection to car for wican :-) up and running. However I dont believe it is possible to setup WG as site2site on a mobile phone (android). If you have achieved that I would be very happy to get your setup. Thanks in advance

gerhardvilsecker, 2025-01-29:
I meanwhile bought a lte mobile router supporting Wireguard VPN. ==>my Wican is up and running using home assistant integration and VPN site2site tunnel. Works like a charm :-)

closed 2025-01-29 by gerhardvilsecker

meatpiHQ, 2025-01-30:
@gerhardvilsecker which LTE route did you buy?

Just FYI , I plan to support wiregurd on WiCAN PRO and ESPNetlink.

I'm also working on getting the ESPNetlink to work as an wifi to LTE gateway.

gerhardvilsecker, 2025-01-30:
@meatpiHQ : I bought a  GL-XE300 called "Puli" from a company called GL.iNet.. What is the current time schedule for wican pro? I would be very interested to become an "early adapter". Could we move this discussion to a more "private" platform? thanks Gerhard