meatpiHQ/wican-fw #330: Filter wrong data possible?

cr0i, 2025-01-25:
Hello @meatpiHQ ,

I´m trying to add some more values to the vehicle Profile from VW:ID.
With one of them I have a problem, that I could not solve.
I added PID 222613 as int_temp (cabinet temperature sensor). The expression for this is:
`([B4:B5])/5-40`
As Min value I took -35, as max value 95.
But sometimes I get wrong data from the car, then the temperature is reportet wrong with 0 °C.
I analysed this and it´s really, that the car presents B4=0 und B5=200, even this is wrong.
Is there a possibility to filter out this one value, so it´s ignored?
As workaround I put 0.01 as min value, but this is a little bit wrong, because the temparature could be negative too.

meatpiHQ, 2025-01-25:
@cr0i it's possible you could have the wrong pid_init string. Please upload the car scanner logs and I'll have a look.

cr0i, 2025-01-26:
Hello @meatpiHQ ,

even carscaner shows the wrong value of 0°C.
When I start climate, the value is correct.
When I stop Climate and wait a little bit, then the wrong value appeares.

I think we must ignore this one value, if this is possible.

on ext_temp this problem does not exist, it happens only on int_temp (cabinet temparature)

gerhardvilsecker, 2025-01-27:
I can confirm this problem. I have torque running in my Cupra Born (= VW ID.3). I see the same problem. I am also using carscanner init_string for pid 222613. In my environment its even more coincidential: most of the time it works but very seldom I get "0" degrees.  I could not finde any "rule" for misfunciton so far...

Derdirkf, 2025-05-22:
Hi @cr0i 
I have a similar problem:
My Renault Kangoo reports sometimes a SOC of 0% - herr I also would like a filter.
Now I read that you set a minimum - this would fix my problrm. Where can I setup this?

cr0i, 2025-06-06:
> Hi [@cr0i](https://github.com/cr0i) I have a similar problem: My Renault Kangoo reports sometimes a SOC of 0% - herr I also would like a filter. Now I read that you set a minimum - this would fix my problrm. Where can I setup this?

What Firmware are you running? In v4 the min max fields were added.