meatpiHQ/wican-fw #331: Disconnects every hour for about 3 minutes.

richl007, 2025-01-26:
I'm on firmware 4.03 and sending the data to HA through MQTT. When driving on a long trip the WiCan seems to stop sending data every hour for about 3 minutes. The ODB Status goes offline and then comes back online by itself. Any idea why?  

Mercedes sprinter van 2024

meatpiHQ, 2025-02-25:
@richl007 are you able to get the logs so we can get to the bottom of this?

https://github.com/meatpiHQ/wican-fw/issues/337#issuecomment-2665774155

richl007, 2025-02-25:
Yes, if you give me detailed instructions.

meatpiHQ, 2025-02-25:
Please follow the instructions in the comment. Let me know if anything is unclear.

https://github.com/meatpiHQ/wican-fw/issues/337#issuecomment-2665774155