meatpiHQ/wican-fw #335: new vehicle: Fiat 500e (2020 version)

Ham60elu, 2025-01-29:
Model: Fiat 500e (2020 Version)

Hello,
I am using the Home Assistant integration with firmware 4.04 on my WiCAN-OBD-Adapter.
The integration works, as I can see the diagnostic entities as well as one sensor: Status Voltage.
But no other sensors are shown, especially not the SOC from the battery.

On the WiCAN-OBD-Adapter in the Vehicle Specific section I have selected the Fiat 600e entry, because there is no Fiat 500e entry,
and 
in the Standard PIDs section I have tried to enable the standard PIDs with all the ECU Protocols, but with no result.

As mentioned on https://meatpihq.github.io/wican-fw/home-assistant/integration_setup
"Select your vehicle model from the dropdown list. If your vehicle model is not yet supported, please open an issue on GitHub, and we will walk you through adding support."

Any help is appreciated.
Thank you.

meatpiHQ, 2025-01-29:
@Ham60elu please follow the instructions in the link below to get the SoC car scanner logs.

https://meatpihq.github.io/wican-fw/car_scanner/logs

cr0i, 2025-01-29:
Try these values (They are from Custom PIDs):
Custom Initialisation: ATSP7;ATCP18;
Init: ATSHDA44F1;
PID: 22A010
Expression: (B5/0.45)-6.4

The Expression could be too:
(B5/(0.9*255)*100)-5

Ham60elu, 2025-01-31:
> Try these values (They are from Custom PIDs, you could change the corresponding fields in vehicle profile): Custom Initialisation: ATSP7;ATCP18; Init: ATSHDA44F1; PID: 22A010 Expression: (B5/0.45)-6.4
> 
> The Expression could be too: (B5/(0.9*255)*100)-5
> 
> P.S: It seems like the SoC is not updating, when loading.

Thank you very much! 
In my case the PID for the SoC was 22A0102 with the Expression (B5/(0.9*255)*100)-5 .
Is there a way to find the right expression? How do I get it?
I'd like to import the values for the odometer and the aux battery.

meatpiHQ, 2025-02-01:
@Ham60elu the best way is to collect as much data points as possible and check the change in values then try and guess.

If you export the logs from car scanner and tell me the SoC value I can try help guess the correct expression.

Ham60elu, 2025-05-15:
@cr0i Is there a way to also get the "estimated range" value for the Fiat 500e out of the wican?
Tried nearly all values out of the car scanner log, but no luck.
Maybe you have the necessary values or can point me in the right direction ??

NCO3, 2025-06-13:
Hi @Ham60elu 
Did you get your 500e to work with WiCAN?
I just got the 500e and ordered the WiCAN.
If you could share your config, I would greatly appreciate this!
Thanks and greetings from Hamburg

Ham60elu, 2025-06-15:
Hi @NCO3,
I could only get the SOC to work, so far. For the config see the above post,
and the SoC is not updating, when loading :-(

NCO3, 2025-06-15:
OK, thank you.
And don’t worry
Please allow one more question though.
Will your vehicle be recognized as a known car when you reconnected it?

cr0i, 2025-06-15:
> [@cr0i](https://github.com/cr0i) Is there a way to also get the "estimated range" value for the Fiat 500e out of the wican? Tried nearly all values out of the car scanner log, but no luck. Maybe you have the necessary values or can point me in the right direction ??

Sorrry for the late reply,

I only helped someone else to get values from his Fiat 500e car, I don´t have one.
Can you see the estimated range in carscanner?
Then we can try to get it in WiCAN too.
For the problem, not updating SoC while charging, you can try to open your app, then suddenly the SoC was updated.
We use a openWB Wallbox, this is calculating the SoC manually while charging, so it was not such a big problem.

thecem, 2025-09-05:
Could this help? It is from OVMS.

```txt
Can bus: CAN1 C-CAN(500kbps); CAN2 B-CAN(50kbps)

MsgId        bus     Msg                  Signal    variabl                               factor/offset  status  
0xC10A040    can1   MSG09_BPCM                   ms_v_bat_soc                SOC                          ok
0xC10A040    can1   MSG09_BPCM                   ms_v_bat_range_est          EStRANGE                     ok
0xC10A040    can1   MSG09_BPCM                   ms_v_bat_energy_used_total  EStRANGE                    test
0xA18A000    can1   STATUS_B_CAN  19/1           ms_v_door_fl                DriveDoorSts                 ok
0xA18A000    can1   STATUS_B_CAN  53/1           ms_v_env_handbrake          HandBrakeSts                 ok
0xA18A000    can1   STATUS_B_CAN  24/8           ms_v_env_temp               ExternalTemp      5  -40  wrong value
0xA18A040    can1   STATUS_C_EVCU 28/3           ms_v_charge_state           ChargingSystemSts            test
                                                 ms_v_charge_inprogress                                   test
                                                 ms_v_door_chargeport                                     test
0xA28A000    can1   VEHICLE_SPEED_ODOMETER 52/20 ms_v_pos_odometer           TotalOdometer             wrong value     
***********************************************************************************************

***********************************************************************************************
0xC41401f    can2   TCU_REQ       0x20           CommandStartCharge          TCU_ChargeNow_Req
0xC41401f    can2   TCU_REQ       0x40           CommandStopCharge           TCU_ChargeNow_Req

0xC41401f    can2   TCU_REQ       0x8            CommandLock                 TCU_DoorLockCmd
0xC41401f    can2   TCU_REQ       0x10           CommandUnLock               TCU_DoorLockCmd
                                                 ms_v_env_locked
0xC41401f    can2   TCU_REQ       0x2                                        TCU_HornLights_Req
0xC41401f    can2   TCU_REQ       0x0                                        TCU_HornLights_Req

0xE194031    can2   TCU_REQ       0x10      5/3  CommandActivateValet        TCU_PreCondNow_Req      PreConditionNow         
0xE194031    can2   TCU_REQ       0x20      5/3  CommandDeactivateValet      TCU_PreCondNow_Req      `StopPreconditionNowStopPreconditionNow`
```

I managed to get SoC? with 22A0102 with the Expression (B5/(0.9*255)*100)-5.

eiswind, 2026-05-13:
@faniskou is the soc updating while charging at your side?

mf76130, 2026-10-05:
hi,
Just bought a used Fiat 500e Cabrio from 2021 yesterday.
Are there any tips on how to get it working with the Wican?
I tried several hours with AI, but no real outcome but getting blocked.
Minimum i want is SOH and SOC