meatpiHQ/wican-fw #340: "FROZEN MeatpiHQ DONGLE" no longer reporting - HOW TO EASILY RESET that without pulling it out ?

typxxi, 2025-02-02:
I have bought a usual MeatpiQ OBD Dongle arund April 2024 and started working with that dongle on my e-GOLF in August 2024. The dongle looks like this - I know there have been new ones released in autumn 2024, but mine is older from easter 2024 maybe.

[image]

It worked finally from September onwards. It is not my car even though I call it this way cause it is my parents car. So I am not using that on a daily basis, rather on ocasions when they asked me to pick up something for them. 
But in August / September during developement how to get the SOC, ODOMETER and so on into HA I had the car a lot more and from my impression it looked stable, I mean the HA commuication and the dongle itself. 

BUT: Chrismas came and went by, New years eve too and after 2 weeks I called my parents asking if there is something wrong cause they seem to not drive the car anymore cause the SOC was the same, odometer was the same. 

At first I had thought they would not have reached the WLAN or have not followed the procedure which meant that I had faced some transmission errors back then in the beginning, more precise: I did not see new  data and so I had developed a procedure like when you have arrived turn the ignotion off and then on again, hit the break that the car is ready to drive again which means the full communication must be available. Then count from 1 to 10 and turn the ignition off.

This way we got the data when the car is arriving or leaving without occassions where data were missing. Each drive into and out of the garage had been recorded. 
But early January I had recogniced that the car did no longer move, my parents were fit and never told me that the car was broken. So they said, no we are driving around like usual, nothing has changed. 

BUT: the car did not report anything

With a few trials and errors I came behind that the dongle must have been frozen cause new data only arrived in the second I had pulled the Meatpi Dongle out and pushed it in again. HA showed instantly proper odometer, SOC and range.

[image]

2  weeks later = yesterday I had a similiar feeling, HA data for the GOLF were not changing anymore. 
I needed to pull the dongle out around midnigth 1. / 2. of Februar.

"My" firmware is from August 2024 or so, I checked my github thread: firmware 2.98 
Has there been any fix in a newer stable Firmware regarding to such a freezing ?

I call it freezing cause no new data arrived in HA. 
I was not there, had no physical access to the car nor meatpi dongle  so I could not check and connect to the meatpi dongle. 

The only thing I can tell for sure that HA has seen the Dongle in the WLAN alive, but that means only that a ping to the dongle would arrive. 

#1 is there any smarter way to reset the dongle without pulling out the dongle and pushing it in again ?
I have not found any kind of reset button in the hurry I was in yeasterday when I was at the car. 

#2 Is this freezing a firmware issue ?

#3 If I would upgrade to a newer version can I break my working HA integration ?
What steps would I have to do before a firmware upgrade to make sure that it will be working again after wards ?
I have done the upgrade from 2.x something to 3.x experimental last August and it had worked for me somehow but I have not the best memories with this procedure and went back to the latest stable firmware in July / August 2024. 
Back then there had been 2 firmwares, both unstable - maybe 3.08 and 3.2x or whatever, but I still can remember that there were 2 or so.

I am asking this cause I do not have so much time anymore like in July / August when it took 3 weeks to get it running. 

#4 assuming that even a new firmware would not fix it is there a way for me to add a small reset switch to the current dongle with 2 cables ? I mean such tiny tactile switches cause my parents are too old to unplug and plug the dongle in that position. Press a button might work for them. 

#5 I have remote access to my parents smarthome system ( HA running on Pi 4) and on a Windows laptop which is monitoring the big LFP battery which is Windows only. So I have access via VNC to this Laptop. Can I connect that laptop to the cars meatpi dongle to the setup or was that a function only available from my phone when I am inside the car ?

Thanks a lot. Sorry, these are just my memories from the installation

ccremer224, 2025-02-02:
Unfortunately I have no answer for you, but you mentioned you read out the odometer. 
Can you post the PID for the e-Golf odometer?

typxxi, 2025-02-02:
> Unfortunately I have no answer for you, but you mentioned you read out the odometer. Can you post the PID for the e-Golf odometer?

everything I had done and can remember is explained here
https://github.com/meatpiHQ/wican-fw/issues/168#issue-2453508045

Make sure to open and load it fully, then press strg + f and search for odometer.
If you only find 3 spots then you have not loaded it fully, otherwise you would have 47 hits in this whole topic.
I have developed it step by step so you should find all the details and pitfalls I had encoutered back in late summer of last year.

JabeBRD, 2025-02-07:
Until you find better solution, i have seen this obd-adapter that has switch, sold in eBay and AliExpress, what it does, i dont know.

[image]

typxxi, 2025-02-17:
Found one for 3€ or so here
https://aliexpress.com/item/1005005557901975.html

JabeBRD, 2025-02-17:
I have OBD extension cord like this (without switch) in my vehicles, wican is placed inside bottom of left A-pilar (fusebox in some models), i have no connection issues, even wican is ziptied to metal bar.
I wanted to hide my dongle, also it got hits from legs when hanging in obd port.

typxxi, 2025-02-19:
Same issue here and now I need the RESET Switch cause it is not the fault of the Home Assistant mqtt addon.

MeatPi dongle seems to drop the connection and does not get it back or does no longer run the mqtt task. 

I will go the same way regarding the new position cause most cars have space there except the luxurious audi, bentley and so on which tend to have control units there next to your left leg deep down.

And when it rains and the drain tubes from the roof are full of leafs then the water ingress starts and slowly the whole noise insulation under the drivers feet soaks full of water till the Control unit on the left also gets under water. Damage around 5000€  - with some luck 500€. 
Item has been shipped.

After a while the 2 most right LEDs are dead, no matter what i try like ignition on or off, nothing is received by mqtt explorer until I unplug the dongle and put it back into the car socket.

[image]

Werner1000, 2025-02-21:
Have had similar problems: meatPi ver. 3.48 bought in 8/2024 was working absolutely fine, booking in to my wifi and mqtt-publishing was functional. About 2-3 months lateron most often meatpi was frozen when coming home. Blue light permanently on, green and yellow off, no access point visible, not booked in to my strongly available wifi. Luckily I had implemented this OBD cable with switch - switch off and on and everything was OK until next leaving home.

My fix: update to ver. 4.04. OTA-Update bricked meatpi, so I flashed via USB, settings lost. With same settings now everything is fine again. Connecting to wifi and mqtt-publishing works since a few weeks.

meatpiHQ, 2025-02-25:
@typxxi did you try @Werner1000 suggestion ?

typxxi, 2026-06-13:
not yet, as the whole system was running and the switch on the cable worked. 

I have seen now the HA integration and I wanted to make sure first about the adventure and risks involved to upgrade as I had done the developelment in so many hours that I did not wanna risk bricking the dongle and start from scratch.

Now i have heard and seen [in this video](https://youtu.be/3JW3J4_Ukrs) that the most recent firmware(s) is more or less based on a car selection like VW and then e-GOLF which then delivers a set of data to be transmitted which looked good.

Yes, the car is mentionened  in the compliant list but the biggest issues with the meatpi dongle is the halfbaked infos.

I will not loose 100 hours of work and then have to go back. 
Every car owner wants to know what to expect or at least if all the basics will be support or how to increase the data elements.
range , odometer, SoC for example are key but what does compliant mean for e-Up and e-GOLF 

I am suspicious cause there is no asterix and no explanation if this means support for the first generation or only for the facelift like egolf from 2017 onwards and e-Up from 2020 onwards. 

Hence I have waited so long as I still can remember the efforts involved 2 years ago. Now it sounds too good to be true and I am carefull to not loose important functions for a short gain which might not work. 

You need a proper and better register or repository or people who are willing to help for certain vehicles and maintain what is available and how. I stopped recommanding  meatpi after it worked cause you can hardly explain that way as "normal" or efficient how I had finally worked it out. 

I hope you have a simple, clean path how to upgrade as I am now on may 2.98 or 3.01