meatpiHQ/wican-fw #341: New vehicle profile for Tesla Model Y

bogdanionut, 2025-02-03:
Hi Ali,
Can you help create a new vehicle profile for Tesla Model Y? I'm mainly interested in:
- Nominal full pack when full
- Nominal full pack when new
- Energy buffer
- SOC
- Cell temp avg
- Odometer
I have the DBC file for Model 3 and with Savvycan I did get some parameters, but did not find a lot of documentation on how to setup the specific filters and custom PIDs.
Thanks,
Bogdan

jomud9, 2025-02-03:
Just curious - Did you try teslamate? I wish my car have the same. :-(

bogdanionut, 2025-02-03:
I have also teslamate running for the last 3 years, but those are mainly estimates based on Tesla API(which might become paying soon) and I'm missing the temperature from those...:D