meatpiHQ/wican-fw #344: New Vehicle Profile - Ford Transit - Tested and Ready

anthonysecco, 2025-02-05:
Hi @meatpiHQ ,

I've developed a profile to pull some additional data for the Ford Transit.  This has been tested on my 2021 Ford Transit Ecoboost North America.  This can also serve as a template for those wanting to build profiles for Fords other of models.  I've used the [Vehicle Profile Editor](https://meatpihq.github.io/vehicle_profile_editor/) to put together the following JSON.  I'm happy to contribute in how best I can, let me know.

`{
  "car_model": "2021-Ford-Transit-North-America",
  "init": "ATE0;ATL0;ATH0;ATR1;ATST32;ATSP6;ATCAF1;",
  "pids": [
    {
      "pid": "221E1C",
      "pid_init": "ATSH7E0",
      "parameters": [
        {
          "name": "Ford-PCM-Transmission-Temp",
          "class": "temperature",
          "expression": "(([B4:B5])*45)/720",
          "unit": "C",
          "min": "-40",
          "max": "200"
        }
      ],
      "isExpanded": true
    },
    {
      "pid": "2204661",
      "pid_init": "ATSH7E0",
      "parameters": [
        {
          "name": "Ford-PCM-Desired-Boost",
          "class": "pressure",
          "expression": "((256*B4)+B5)/128",
          "unit": "kPa",
          "min": "0",
          "max": "1000"
        }
      ],
      "isExpanded": true
    },
    {
      "pid": "22054B1",
      "pid_init": "ATSH7E0",
      "parameters": [
        {
          "name": "Ford-PCM-Oil-Life",
          "class": "percentage",
          "expression": "B4",
          "unit": "%",
          "min": "0",
          "max": "100"
        }
      ],
      "isExpanded": true
    },
    {
      "pid": "221E121",
      "pid_init": "ATSH7E0",
      "parameters": [
        {
          "name": "Ford-PCM-Gear",
          "class": "",
          "expression": "B4",
          "unit": "",
          "min": "0",
          "max": "100"
        }
      ],
      "isExpanded": true
    },
    {
      "pid": "2204621",
      "pid_init": "ATSH7E0",
      "parameters": [
        {
          "name": "Ford-PCM-Wastegate",
          "class": "percentage",
          "expression": "B4/128*100",
          "unit": "%",
          "min": "0",
          "max": "100"
        }
      ],
      "isExpanded": true
    },
    {
      "pid": "220404",
      "pid_init": "ATSH7E0",
      "parameters": [
        {
          "name": "Ford-PCM-Knock-Count",
          "class": "",
          "expression": "B4",
          "unit": "",
          "min": "0",
          "max": "255"
        }
      ],
      "isExpanded": true
    },
    {
      "pid": "22F49D1",
      "pid_init": "ATSH7E0",
      "parameters": [
        {
          "name": "Ford-PCM-Fuel-Rate",
          "class": "volume_flow_rate",
          "expression": "([B4:B5])*2/100",
          "unit": "g/s",
          "min": "0",
          "max": "255"
        }
      ],
      "isExpanded": true
    },
    {
      "pid": "2203E8",
      "pid_init": "ATSH7E0",
      "parameters": [
        {
          "name": "Ford-PCM-Learned-Octane-Ratio",
          "class": "percentage",
          "expression": "([B4:B5])/16384",
          "unit": "%",
          "min": "-100",
          "max": "100"
        }
      ],
      "isExpanded": true
    }
  ]
}`

If I can find other useful PIDs and be able to initialize other modules, I can expand this profile.  Right now, I can only initialize the PCM.  

I'm hoping, in the future, to enable PIDs from other modules like the Body Control Module and All-Wheel Drive module, however, I don't believe my current WiCAN hardware supports the needed instruction set.  See [here ](https://github.com/meatpiHQ/wican-fw/discussions/334)for details on that.

meatpiHQ, 2025-02-07:
@anthonysecco Great work! Please create a PR with this profile.

closed 2025-02-07 by anthonysecco

anthonysecco, 2025-02-07:
@meatpiHQ  - Sure thing, please add me as a collaborator and I'll be able to create the PR.

meatpiHQ, 2025-02-08:
@anthonysecco you can fork the repo and add the profile to the folder, then create a PR.

askpatrickw, 2025-04-13:
Just re-postig the original JSON in a formatted readable format.
I'm going to be reviewing this, some notes from [Ford Transit USA Forum Community](https://www.fordtransitusaforum.com/threads/meatpi-wican-on-ford-transit.102499/) and my own discovery and I'll send a PR for a Vehicle Profile when I'm further along as I don't see one yet for the Transit and the Transit is not listed as supported yet.

```json
{
  "car_model": "2021-Ford-Transit-North-America",
  "init": "ATE0;ATL0;ATH0;ATR1;ATST32;ATSP6;ATCAF1;",
  "pids": [
    {
      "pid": "221E1C",
      "pid_init": "ATSH7E0",
      "parameters": [
        {
          "name": "Ford-PCM-Transmission-Temp",
          "class": "temperature",
          "expression": "(([B4:B5])*45)/720",
          "unit": "C",
          "min": "-40",
          "max": "200"
        }
      ],
      "isExpanded": true
    },
    {
      "pid": "2204661",
      "pid_init": "ATSH7E0",
      "parameters": [
        {
          "name": "Ford-PCM-Desired-Boost",
          "class": "pressure",
          "expression": "((256*B4)+B5)/128",
          "unit": "kPa",
          "min": "0",
          "max": "1000"
        }
      ],
      "isExpanded": true
    },
    {
      "pid": "22054B1",
      "pid_init": "ATSH7E0",
      "parameters": [
        {
          "name": "Ford-PCM-Oil-Life",
          "class": "percentage",
          "expression": "B4",
          "unit": "%",
          "min": "0",
          "max": "100"
        }
      ],
      "isExpanded": true
    },
    {
      "pid": "221E121",
      "pid_init": "ATSH7E0",
      "parameters": [
        {
          "name": "Ford-PCM-Gear",
          "class": "",
          "expression": "B4",
          "unit": "",
          "min": "0",
          "max": "100"
        }
      ],
      "isExpanded": true
    },
    {
      "pid": "2204621",
      "pid_init": "ATSH7E0",
      "parameters": [
        {
          "name": "Ford-PCM-Wastegate",
          "class": "percentage",
          "expression": "B4/128*100",
          "unit": "%",
          "min": "0",
          "max": "100"
        }
      ],
      "isExpanded": true
    },
    {
      "pid": "220404",
      "pid_init": "ATSH7E0",
      "parameters": [
        {
          "name": "Ford-PCM-Knock-Count",
          "class": "",
          "expression": "B4",
          "unit": "",
          "min": "0",
          "max": "255"
        }
      ],
      "isExpanded": true
    },
    {
      "pid": "22F49D1",
      "pid_init": "ATSH7E0",
      "parameters": [
        {
          "name": "Ford-PCM-Fuel-Rate",
          "class": "volume_flow_rate",
          "expression": "([B4:B5])*2/100",
          "unit": "g/s",
          "min": "0",
          "max": "255"
        }
      ],
      "isExpanded": true
    },
    {
      "pid": "2203E8",
      "pid_init": "ATSH7E0",
      "parameters": [
        {
          "name": "Ford-PCM-Learned-Octane-Ratio",
          "class": "percentage",
          "expression": "([B4:B5])/16384",
          "unit": "%",
          "min": "-100",
          "max": "100"
        }
      ],
      "isExpanded": true
    }
  ]
}
```