meatpiHQ/wican-fw #345: New Car Profile: Mercedes G580

noobey, 2025-02-08:
Would love any help in getting this built please!

My HA setup is looking pretty bare in it's current setup. 😄 

[image]

Thanks in advance!

meatpiHQ, 2025-02-19:
Hey @noobey can you please export the logs for the SoC from car scanner and I'll help you create a profile for your car.

https://meatpihq.github.io/wican-fw/car_scanner/logs

noobey, 2025-02-20:
> Hey [@noobey](https://github.com/noobey) can you please export the logs for the SoC from car scanner and I'll help you create a profile for your car.
> 
> https://meatpihq.github.io/wican-fw/car_scanner/logs

Hi @meatpiHQ appreciate the reply and offer!
I'm pretty sure the issue here is me, so apologise, but none of the options in the guide you've linked to grab the logs correlate to what I see in my Wican menu.
For example on Step 3, I dont see an All Sensors button anywhere inside any of the menu items on the left ribbon.

If you could just let me know where I'm going wrong I'd appreciate it, cheers!

[image]