meatpiHQ/wican-fw #347: Revive possibly bricked WiCan

boecks, 2025-02-11:
**Describe the Issue**

After changing the Wi-Fi settings on my WiCan, including the AP channel, I lost access to it. To recover it, I attempted a firmware flash via USB following the instructions in the Wiki. However, I mistakenly flashed the USB firmware onto my OBD WiCan.

Since then, the device no longer shows any lights when plugged in via USB or OBD. However, I can still enter flash mode (orange light). Despite successfully flashing the correct OBD firmware afterward, the device remains unresponsive.
Steps to Reproduce

1. Change Wi-Fi settings, including AP channel, causing loss of access.
2. Flash the USB firmware onto an OBD WiCan.
3. Observe that no lights appear when powered via USB or OBD.
4. Enter flash mode (orange light still works).
5. Flash the correct OBD firmware.
6. Device remains unresponsive.

**Expected Behavior**

After flashing the correct OBD firmware, the WiCan should function normally again.

**Actual Behavior**

The device does not power up or show any LEDs, except when entering flash mode.

**Device Details**

WiCan Model: WiCan OBD (previously said Hardware Version 3.1 if I remeber correct)

Firmware Used:
 -  Incorrectly Flashed: wican-fw_usb_v404u
 -  Attempted Recovery: wican-fw_obd_v404

Power Sources Tried: USB, OBD
Current Status: Can enter flash mode (orange light), but otherwise unresponsive

**Possible Solutions Tried**

1. Flashing the correct OBD firmware
2. Power cycling via USB and OBD

**Additional Context**

Fortunately, I have a second WiCan that is still operational, but I would like to restore this one. Is there a way to fully recover it after flashing the wrong firmware?

gerhardvilsecker, 2025-02-16:
I had a similar problem when I started to use wican. At least with my hardware (dont know if there are different version on the market) it was possible to flash device with obd version after having flashed device with wrong usb version. 
Did you ever try to use "erase flash" in esp flash tool? This was Ali's hint for me. After erasing (which was successful) I realized that flashing  was not successful at all. I checked this by reading the esp chip. All I got was a lot of "FFFF" in a row. After successful flash device worked immediately...

meatpiHQ, 2025-02-18:
@boecks as @gerhardvilsecker Have you tried to erase the flash?

I've added an extra note to erase the flash in the [documentation ](https://meatpihq.github.io/wican-fw/config/firmware-update#_2-usb-flash) 

> WiCan Model: WiCan OBD (previously said Hardware Version 3.1 if I remeber correct)

Are you sure it v3.1 ? when did you buy it?

boecks, 2025-02-18:
> Are you sure it v3.1 ? when did you buy it?

Sadly I didn't memorize it well before flashing the latest FW. Though I could swaer it said v3.1 (or more general, something with "v" and a Version number)

Bought it from Mouser in June 2024.

I still have to try to reflash the bricked one...