meatpiHQ/wican-fw #353: RESET MEATPI DONGLE

typxxi, 2025-02-17:
Is there any way to reset the dongle by pressing a button or connecting 2 points on the pcb ?

I have an issue with a frozen dongle that is no longer reporting data to HA. If pull the meatpi dongle out then the data will instantly arrive. Days or weeks later the dongle suddenly stops.
THis is not an mqtt server issue cause I have restarted the mqtt  addon running under HomeAssistant multiple times with no success. I have also rebooted HA by a full start and again no data from the car.

Only way to awake the dongle is to pull it out and push it in. Works for me, but not for my elderly parents who can not reach the dongle where it sits.

**Is there anyway to reset hardware wise (PCB or a button on the dongle it self) ?**

I have the usual white dongle, not USB or other connector.  

[image]

cr0i, 2025-02-19:
When you use USB Flash method from here:
https://meatpihq.github.io/wican-fw/config/firmware-update
You can ERASE and then write an empty config again.

tung256, 2025-03-08:
would be nice if the wican can auto reboot itself at midnight or something.
for now, this is my solution with on/off switch. when my wican freezes, weekly,  i can just flip to Off then back to On instead of pulling the wican out and plug back in.

[image]