meatpiHQ/wican-fw #356: Use Home-Assistant API ?

vincegre, 2025-02-20:
Hi

Would it be possible to add the feature where Wican Pro would call the API of Home Assistant to send datas ? It would need the Home-Assistant URL and a long time token as parameters.

Thanks

Vincèn

kquinsland, 2025-04-10:
Use MQTT for this?

vincegre, 2025-04-10:
> Use MQTT for this?

nope sorry it's non sense as it would imply to expose an MQTT broker on Internet while HA API is always avalaible if you have a Nabu Casa subscription or if you have setup internet access at your HA with a proxy ;)

goldbattle, 2025-04-13:
Why not just use the official HACS package to pipe it all in?
https://github.com/jay-oswald/ha-wican

vincegre, 2025-04-13:
> Why not just use the official HACS package to pipe it all in? https://github.com/jay-oswald/ha-wican

because it's useless ! It's designed for WiCAN to be in your LAN but please explain me how you can get Wican device in your LAN when it's installed in the car......
I don't understand why it's not designed in a way that Wican calls the HA API to push datas which allows Wican to work everywhere so long it has internet access !

vincegre, 2025-04-16:
> Most WiCAN users use it on EVs, and you’ll notice that all the current vehicle profiles are for EVs. It’s primarily used for monitoring and managing charging.

It's a so restrictive use of device :( Hope some more useful communication ways will be implemented in near future !