meatpiHQ/wican-fw #364: Gets stuck scanning for standard PIDs on Nissan Ariya

mt104, 2025-02-28:
Scanning for standard PIDs gets stuck on my Nissan Ariya. Photos of my settings attached. With the same WiCAN I can talk to the car using the iOS Car Scanner app no problem. Am I doing something wrong?

[image]
[image]

mt104, 2025-04-24:
Could I have an update on this please @meatpiHQ ?

meatpiHQ, 2025-04-24:
@mt104 just to confirm the car is ON when you are scanning right?

meatpiHQ, 2025-04-24:
@mt104 can you please post your device configuration.  

System-> Download Configuration 

Make sure to remove any sensitive information like ssid and passwords

mt104, 2025-04-26:
Yes, the car was on. Config attached, with passwords redacted.

[config_2025-04-26.json](https://github.com/user-attachments/files/19921825/config_2025-04-26.json)

meatpiHQ, 2025-04-26:
@mt104 are you using the WiCAN or WiCAN PRO?

mt104, 2025-04-26:
It's a WICAN-OBD-C3 from Mouser.

meatpiHQ, 2025-04-26:
@mt104 I see you've selected the Leaf profile. Is that working?

Can you please try to disable MQTT and try the scanning again?

mt104, 2025-04-26:
I disabled MQTT the tried to scan the standard PIDs again. Same behaviour... the scan button changed to greyed out "Scanning..." and after 3 or 4 minutes there's still no change.

meatpiHQ, 2025-04-26:
Does show any errors like scan failed or it just dies?

mt104, 2025-04-26:
No errors, it just sits there saying it's scanning. The device itself doesn't lock up though, as I can click through to other pages fine.

meatpiHQ, 2025-04-26:
I think its crashing when you click the scan button. If the car is close by, please check the blue LED if it blinks when you press the scan button.

mt104, 2025-04-26:
The blue LED does not blink, but I see an orange LED flash 6 times.

meatpiHQ, 2025-04-26:
@mt104 I'm still unable to reproduce. Can you please connect using car scanner APP and export the logs

https://meatpihq.github.io/wican-fw/car_scanner/usage

mt104, 2025-04-26:
Logs from Car Scanner attached. The car was off during this, but it still fetched data ok. I can do another with the car on later if necessary.

[log.txt](https://github.com/user-attachments/files/19922278/log.txt)

meatpiHQ, 2025-04-26:
@mt104 the car scanner log looks pretty standard. The only way to get to the bottom of this is to hookup the usb cable and look at the debug logs. Is this something you're comfortable doing ? I'll need to send you an new firmware with the debug logs enabled, and I will walk you through it.

The other option is that I can help you setup custom PIDs?

mt104, 2025-04-26:
I'm happy with either approach, and comfortable with the USB + debug firmware approach. But... my original goal was to use a set of custom PIDs, learning how to set them up, so that I can choose what to monitor. I'll be feeding data into Home Assistant via MQTT. Ideally I'm looking for traction battery SOC, SOH, voltage and temperature. I can see all of those using Car Scanner whether the car is on or off.

meatpiHQ, 2025-04-26:
I'll help you create a profile for SoH and SoC .. etc after we figure this out.

Here's the debug firmware. This firmware will reset your configuration I suggest to download so you can restore. After the update and you restore your config, please disable **sleep mode**, **mqtt** and **"Vehicle Specific"**. And post the log file when you're done.

[wican-fw_obd_v404-52-g3550918-dirty.zip](https://github.com/user-attachments/files/19922596/wican-fw_obd_v404-52-g3550918-dirty.zip)

1- Plug in a Micro USB cable into WiCAN, and it will enumerate as USB to serial device
2- Use [PuTTY](https://www.putty.org/) to log the serial output.

[image]

[image]

mt104, 2025-04-26:
Ok, great, thanks. Serial log for a failed autopid scan attached.

[202504261457 Nissan Ariya failed autopid scan.txt](https://github.com/user-attachments/files/19922810/202504261457.Nissan.Ariya.failed.autopid.scan.txt)

meatpiHQ, 2025-04-26:
After you press the scan button let it scan for like 30 sec or more before grabbing the logs.

mt104, 2025-04-26:
Here you go...

[202504261534 Nissan Ariya failed autopid scan.txt](https://github.com/user-attachments/files/19922986/202504261534.Nissan.Ariya.failed.autopid.scan.txt)

meatpiHQ, 2025-04-26:
There is something wrong with the logs, both files are exactly 10KB. It looks like logs are cut short, I don't see the page refresh.

mt104, 2025-04-26:
Hmmm... I'm on a Mac, using CoolTerm. It appears that the default history is 10kB. I can increase that though. How large roughly are you expecting the log to be?

meatpiHQ, 2025-04-26:
my logs file was about 1 MB, maybe make it 10MB max to be safe.

mt104, 2025-04-26:
This one with the capture buffer set to 100MB, it comes out at 49kB.

[202504261608 Nissan Ariya failed autopid scan.txt](https://github.com/user-attachments/files/19923165/202504261608.Nissan.Ariya.failed.autopid.scan.txt)

meatpiHQ, 2025-04-26:
@mt104 Alright! thanks for that! I was able to simulate and fix the error.

[wican-fw_obd_v404-55-g14021d7-dirty.zip](https://github.com/user-attachments/files/19923720/wican-fw_obd_v404-55-g14021d7-dirty.zip)

mt104, 2025-04-26:
Superb, thank-you. With that firmware it takes 9 seconds to discover a set of standard PIDs.

meatpiHQ, 2025-04-26:
Awesome! Now lets create the profile. Please follow the instruction in the link to export the logs, make sure to click on the SoC sensor in car scanner so we can find the correct PID in the log.

https://meatpihq.github.io/wican-fw/car_scanner/logs

mt104, 2025-04-26:
I'm not near the car now, so can't pull the adapter out physically. I am connected via a VPN though, so can connect the Car Scanner app ok from afar.

Is there a way I can capture what you need without pulling the adapter out? If not, I'll be back near the car tomorrow.

Thank-you for all your help today.

mt104, 2025-04-26:
Would rebooting the adapter via a web browser do the trick, causing Car Scanner to disconnect?

meatpiHQ, 2025-04-26:
if you can reach wican through the vpn from your phone then you can connect car scanner you don't have to pull it out. If you cannot do it today that's alright we can do it later. Just post the logs here when you can.

mt104, 2025-04-26:
Car Scanner labels this one "[BMS] TRACTION BATTERY SOC" and the current value is 97.84%

[log.txt](https://github.com/user-attachments/files/19924013/log.txt)

mt104, 2025-05-04:
Now using that profile, I still only see one topic via MQTT...

[image]

meatpiHQ, 2025-05-04:
@mt104 have changed the "Destination Type" to MQTT topic?

mt104, 2025-05-04:
I hadn't, but have now. I also upgraded to the latest firmware. Still only seeing the status topic.

[image]

meatpiHQ, 2025-05-04:
@mt104 what if you try this in your browser?

http://your_ip/autopid_data

mt104, 2025-05-04:
It said there were none, I forget the exact wording, but then I realised Standard PIDs was turned off. So I turned on Standard PIDs and tried a scan, which helped me realise that protocol was still set to Elm32 from the data capture I did last week.

Now working, thank-you for your help.

[image]