meatpiHQ/wican-fw #368: Audi Q5 Type 8r Setup New Hevicule

luiscaloto, 2025-03-08:
Hello everybody!

I´m trying to setup a new car but I have some questions....

I have been followed documentation about to add new car with no success! I´ve got a question about CAN ID???? Where can i find this number...???

Thanks to all!!!!!

luiscaloto, 2025-03-09:
Hello again!!! I had been follow an old tutorial!!!! I upgrade to 4.04 version and now I scan for PIDs and I configure almost all!!

I can´t configure any standrad PID. Any help??

meatpiHQ, 2025-03-10:
@luiscaloto 

> I upgrade to 4.04 version and now I scan for PIDs and I configure almost all!!

So the scan works?

> I can´t configure any standrad PID. Any help??

Can you please provide more information?

luiscaloto, 2025-03-10:
hello!!!! 

1) Yes, of course. Scan works well but there is some PIDs, that don´t show there (ODOMETER is one of then)
2) So, I try to setup ODOMETER, vía standard PID section, but no luck!

I think my initializacion code isn´t working...

meatpiHQ, 2025-03-10:
@luiscaloto have you used car scanner app? Can you read the odometer in car scanner?

luiscaloto, 2025-03-12:
Good morning!!! 

I try today... no, I can´t! Car scanner doesn't show it either!!

So, I understand, that I can´t read it from canbus?? The only parameters related to distance, that I can read are "Distance travelled while inducement system active in current 10k block (0-10000km)"   I don´t know what is it related to...!!

Thanks again for your help!