meatpiHQ/wican-fw #371: New Vehicle - Xpeng G6

stevelea, 2025-03-17:
I am try to add a custom vehicle an XPENG G6. I have it connected with Car Scanner and can read values fine, so collected some logs and used chatgpt to help me create a json file. However i try to ypload the json file in the web interface but does not appear in the pluudown. Here is the json i have , any clues ? chatgpt suggests using scp to transfer but dont see how to enable ssh. thx

[xpengg6.json](https://github.com/user-attachments/files/19279019/xpengg6.json)

gerhardvilsecker, 2025-03-19:
Already tried to use an existing car profile and change initcommands and pids  to commands and pids according to your car?

pepelillo, 2025-04-03:
Can I ask you to elaborate the process a little more? mostly how to use ChatGPT, what was the input? the Carscanner logs? and the prompt?
Thanks

stevelea, 2025-04-09:
Yes i used car scanner logs to feed chatgpt and it created the json. I tried getting an existing profile and transcribed into that with no luck. I started with this prompt "i am using car scanner to connect to an xpeng g6 and want to create a profile for the car in wican as described here https://meatpihq.github.io/wican-fw/config/automate/new_vehicle_profiles" and worked with that uploading car scanner logs

frankyman88, 2025-04-15:
Hey Steve, 
Maybe those PIDs it has grabbed from the carscanner logs aren't correctly formatted for the MeatPi, and there are only 5 listed in your JSON which just say "vehicle parameter".
Since the MeatPi has "SaavyCan" support, could you try that to pull out the supported PIDs? https://meatpihq.github.io/wican-fw/savvycan/usage
If you do, can you share them here please as I am trying to find them to use with my VGATE iCar Pro wifi OBD adapter too 😄

stevelea, 2025-04-30:
OK so got it workibg at least for SOC, will work on other PID's now, think i understand the process.I did a pull request but here is the info in the meantime. 
  "profileName": "XPeng G6",
  "car_model": "XPeng G6",
  "init": "ATZ;ATE0;ATH1;ATSP6;ATS0;ATM0;ATAT1;ATSH7E0;ATFCSH704;ATFCSD300000;ATFCSM1;",
  "pids": [
    {
      "pid": "22031E",
      "pid_init": "",
      "parameters": [
        {
          "name": "Battery SOC",
          "pid_init": "ATSH7E0",
          "expression": "((B4*256)+B5)/10",
          "unit": "%",
          "class": "",
          "period": "5000",
          "min": "0",
          "max": "100"

frankyman88, 2025-07-25:
Hey @stevelea and team,
I finally got a MeatPi OBD and have been experimenting. I think i have figured out how to get the odometer value decoded 😄 
Can someone test the following and see if it works for them too? (Matches with the BMS Odometer value in CarScanner)
`      "pid": "2201011",
      "pid_init": "ATFCSM1;ATCRA784",
      "parameters": {
        "ODOMETER": "[B4:B6]"`
If it does work, I'm not sure how to actually get the vehicle profile updated, I'm just using custom currently xD

frankyman88, 2025-09-05:
Hey @stevelea , I see you did another PR recently but the update doesn't work for me :( 
I've been working on a bunch of extra PIDs for awhile now, getting frustrated as they would work sporadically but sometimes return weird numbers when the car was off etc. I figured out that there is another PID for SOC that uses the 784 header instead of the 7E0 header for the one that we have used previously. I've attached my customised vehicle_profiles.json which you can upload and test to see if these all work for you too :) I'm using Max Battery Pack Temp PID >2211071 for battery temp, I see yours is different, which does it correspond to?

[vehicle_profiles.json](https://github.com/user-attachments/files/22170310/vehicle_profiles.json)

frankyman88, 2025-09-09:
> Just sharing on this issue as it is still opened. I started decoding some other parameters (thanks ChatGPT that simplify the docoding process). I don't know if somebody is working on this part too (we can split the job to go faster)n and if you have any other json to share with some new parameters. The one I tested this evening that seems to be working:
> 
> ```
> {
>       "pid": "220104",
>       "pid_init": "ATFCSM0;ATAR;ATSH704;ATFCSM1;ATCRA784;",
>       "parameters": {
>         "TRIP_DISTANCE": "(B4*256+B5)/640"
>       }
> ```
> 
> To do this:
> 
>     * open a single parameter in a Car Scanner
> 
>     * extract from the car scanner log the parameter read (is the one that you will find all long your log)
> 
>     * send the log with the parameter name to chat gpt (adding the wican pro context, and what you would like as answer)
> 
>     * enjoy the decode :P
> 
> 
> If nobody is working on parameter decoding, I can share here and propose a new PR when I've an interesting bunch of base parameters.

Hey mmornati,

Good find!
I tested your PID and it shows 0km but I'm currently parked and charging which makes sense. Will post back when I'm able to confirm that it tracks my next drive. If you check my attached vehicle_profiles.json in the comment above yours, you can upload it on the Automate page by pressing the Browse button and then select XPENG G6 from the list (I deleted all the others for simplicity haha). I've so far got SOC, SOH, Odometer, Battery temp (max), HV battery voltage and current :) I also found that using a different SOC PID to the current one, I am able to just use the single init command at the top and no PID specific inits are needed; hopefully it will work for you too! I've attached an updated version with your trip distance added

[vehicle_profiles.json](https://github.com/user-attachments/files/22225126/vehicle_profiles.json)

stevelea, 2025-09-09:
> Just sharing on this issue as it is still opened. I started decoding some other parameters (thanks ChatGPT that simplify the docoding process). I don't know if somebody is working on this part too [...]

I tried also and get 0 for TRIP_DISTANCE

stevelea, 2025-09-09:
> Hey [@stevelea](https://github.com/stevelea) , I see you did another PR recently but the update doesn't work for me :( I've been working on a bunch of extra PIDs for awhile now, getting frustrated [...]

Sorry have been away, i was also getting all sorts of sometimes no response etc, i thought i solved it by addinig the full init to the pid init. However i have now tried you json with no pid inits and it works, looks like your SOH is better than mine also. I suggest we replace the current file with your one and continue to add there, good work !   Cant actually the batt_tempp one was, i think the BMS version. Also you can try this one CELL_V_DELTA_MAX , 2211061, [B4:B5] / 216 .. Cell Voltage Diff.

frankyman88, 2025-09-09:
> I tried also and get 0 for TRIP_DISTANCE

I will leave it out of the PR for now until I have tested some more

> Also you can try this one CELL_V_DELTA_MAX , 2211061, [B4:B5] / 216 .. Cell Voltage Diff.

Car Scanner does this cheeky thing where it combines the states of multiple PIDs, so the "Delta PID" is actually polling max voltage (2211051) minus min voltage (2211061) alternating in the logs. I had the same problem with "HV EV battery power" reported figure, couldn't get it out but figured it combines the HV voltage and HV current. You can then just set up a home assistant template to combine them once you get the individual values out :)

stevelea, 2025-09-09:
> [@stevelea](https://github.com/stevelea) try this version which has the battery pack temp and voltage min/maxes added. I also remembered i changed odometer from [B4:B6] to [B5:B6] as I was getting some wacky numbers in the middle of the night. Change it back when we get over 65,000km on the clock i guess 😆
> 
> [@mmornati](https://github.com/mmornati) I figured out you were missing a 1 on the PID for trip (should be 2201041 not 220104), which I notice car scanner logs often cuts off the last digit on the first few polls logged for some reason 🤷 will report back whether it works or not soon
> 
> [vehicle_profiles.json](https://github.com/user-attachments/files/22231227/vehicle_profiles.json)

Yep all working, trip still 0 though  

{"SOC":16.3,"SOH":100,"HV_A":0,"HV_V":532.7,"ODOMETER":24714,"HV_T_MAX":27,"HV_T_MIN":24,"HV_C_V_MIN":35.47,"HV_C_V_MAX":35.64,"TRIP_DISTANCE":0}

frankyman88, 2025-09-09:
> Yep all working, trip still 0 though
>
> {"SOC":16.3,"SOH":100,"HV_A":0,"HV_V":532.7,"ODOMETER":24714,"HV_T_MAX":27,"HV_T_MIN":24,"HV_C_V_MIN":35.47,"HV_C_V_MAX":35.64,"TRIP_DISTANCE":0}

Ahh the HV C V min and max should be \1000 not 100 😅

stevelea, 2025-09-09:
> Ahh the HV C V min and max should be \1000 not 100 😅

Perfect
 {"SOC":16.2,"SOH":100,"HV_A":0,"HV_V":532.4,"ODOMETER":24714,"HV_T_MAX":27,"HV_T_MIN":24,"HV_C_V_MIN":3.55,"HV_C_V_MAX":3.56,"TRIP_DISTANCE":0}

meatpiHQ, 2025-09-10:
@mmornati can you post you car scanner logs for the odometer?

jcugat, 2025-10-20:
Hey @stevelea @frankyman88 @mmornati and the rest, I managed to get quite a few more PIDs and uploaded everything here: https://github.com/meatpiHQ/wican-fw/pull/568

I'm not convinced about all the different init commands, seems a bit messy, but maybe you want to try it and see if you can improve it a bit. My OBD2 knowledge is very limited 😅

mmornati, 2025-10-21:
Cool, going to make some tests. Thanks

wambs, 2026-03-02:
There is an easier way to create your PID.
I developed a new GUI that allows you to create PIDs to groups.
Its makes it easier to create and organize your PIIDs.
See https://github.com/meatpiHQ/wican-fw/discussions/673