meatpiHQ/wican-fw #373: New profile MG Marvel and some Qs

pepelillo, 2025-04-01:
Hello. I'm using a Vlink obd adapter to get obd2 data from my MG marvel R top use ABRP and eventually Ezs app and CarScanner, also get a custom PID file for Torque.
I find  Wican device quite interesting and want to substitute the Vlink device with a Wican device so I can get mqtt integration, so I have some Qs.

Can I get the same from the Wican as from the Vlink device?
How can I prepare a new car profile for my MG Marvel with the info I have now, custom PID for torque, CarScanner logs etc.?

AFAIK  ABRP only accepts BLE connections, is it possible to get connected qith WiCAN?

Thanks all

meatpiHQ, 2025-04-02:
@pepelillo 

>Can I get the same from the Wican as from the Vlink device?

What are the parameters are you getting from blink now?

>How can I prepare a new car profile for my MG Marvel with the info I have now, custom PID for torque, CarScanner logs etc.?

You can post the logs here and I can help you create the profile.

>AFAIK ABRP only accepts BLE connections, is it possible to get connected qith WiCAN?

I never tested with abrp. At the moment you cannot use WiCAN in station mode and enable BLE at the same time.

pepelillo, 2025-04-03:
Thanks, this is the CarScanner log.

[log.txt](https://github.com/user-attachments/files/19588520/log.txt)

meatpiHQ, 2025-04-09:
@pepelillo I've created a profile for the MG marvel, please give it a go and let me know.

pepelillo, 2025-04-10:
Thanks a lot for the help, I'm yet considering the purchase because need to asure that I can get what I need, ABRP and wifi simultaneously, So I have to test using 2 obd2 devices at the same time with a Y cable.

meatpiHQ, 2025-04-10:
I'm working on this feature, but it will be only available on the PRO version, I'm not sure it can be reliable implemented on the original version. You can check back again in a week or so.

pepelillo, 2025-04-10:
Great, I'll check it in a week. Thanks a lot