meatpiHQ/wican-fw #374: New vehicle profile - Ram Promaster 2500

lynea08, 2025-04-10:
Just received my new wican pro but there isn’t a vehicle profile for my Ram Promaster 2500 van. 

Thanks for help on next steps as the installation guide just said to open a new issue.

meatpiHQ, 2025-04-10:
@lynea08 Please try to scan for PIDs first in the Automate tab. You might not need a profile.

lynea08, 2025-04-22:
I’ve got the generic profile selected and the readings of the Wican itself are coming thru but none of the van readings. 

[image]

lynea08, 2025-04-25:
For anyone who looks at this issue - the answer is to use Standard PIDs and then ECU Protocol 7-ISO 15765-4 (CAN 29 bit/500k).

closed 2025-04-25 by lynea08

lkazanov, 2026-01-30:
> Just received my new wican pro but there isn’t a vehicle profile for my Ram Promaster 2500 van.
> 
> Thanks for help on next steps as the installation guide just said to open a new issue.

What year is your Promaster? I have a 2026 coming in.