meatpiHQ/wican-fw #375: [Question] - Where is the documentation for `Custom PIDs` ?

kquinsland, 2025-04-10:
For $reasons, my car uses a non-standard(?) PID for odometer readings (it predates the 2019 requirement for `0x01A6` ...) 

Fortunately, a super quick google tells me that `0x4F2` is the PID. The first 3 (of 8) bytes are important and if you concat them together, you get the mileage in kilometers.

```
0x02 0x47 0x0f 0x00 0x00 0x00 0x00 0x00
```

Becomes `02470f`. Convert to base 10: `149263` and then from km to mi: `92747.7283` and if you account for rounding, that's what my odometer displays.

So what do i need for `init`, `Expression`, min/max value?
This seems to be broadcast so I don't (?) need a period? 

Destination should be `Default` to get this new parameter showing up in web and Home Assistant, no?

----

There are a few other signals that I'm curious about but don't know if they're even made available over the ODB2 port (this vehicle has 3 busses...) so I may not have anything else.

Given that, is it worth trying to add a new vehicle profile for just this?  For what it's worth, it seems that Ford uses `0x4f2` for odometer on a _ton_ of their vehicles. I found that PID on a thread discussing an older generation of my car... 

I'm not sure if it would be preferable to have a `Ford: Generic` profile or if you'd prefer profiles for known good.

meatpiHQ, 2025-04-13:
@kquinsland please post the logs from car scanner, I will create a car profile since this is a vehicle specific PID.

> So what do i need for init, Expression, min/max value?

The init is the ELM327 AT commands to initialize the bus and other options like setting the protocol `ATSP6 `and setting timeout `ATST`, the expression is how you calculate the value.

kquinsland, 2025-04-14:
> the expression is how you calculate the value.

Right, but where is the syntax defined?  Which operation(s)/function(s) are supported? I just need the FW equivalent of `int(msg[0:2], 10)` 

> please post the logs from car scanner, I will create a car profile since this is a vehicle specific PID.

Attached. The password the same as the `sha1` of the `log.txt` within the zip: `199c92f900ebaa71caa7c64d055ead3938f7f89d`

[log.zip](https://github.com/user-attachments/files/19728047/log.zip)

meatpiHQ, 2025-04-14:
Here you go 
https://meatpihq.github.io/wican-fw/config/automate/usage#expression-format-for-automate

kquinsland, 2026-02-02:
Heh, took _a while_ to find the time to get back to this **but** I managed to get the odometer carved out:

[image]

And with the webhook stuff, it just showed up in Home Assistant <3.