meatpiHQ/wican-fw #376: New Vehicle Audi A3 1.6 2016

slofii, 2025-04-10:
Hello, 

got my WiCan Pro today. 
The Guide told me to open a new issue if there's no profile, so here I am. :)

meatpiHQ, 2025-04-13:
@slofii You might not need to create a profile for this vehicle unless there is a manufacturer specific parameters. You should try to use the automate to scan for the available PIDs.

slofii, 2025-04-16:
That works. Thanks. :)

slofii, 2025-05-15:
Hope its okay to use my old issue:
How can I get the odometer from my vehicle? It didnt appear in the available PIDs.

meatpiHQ, 2025-05-15:
@slofii can you see the odometer reading on car scanner app?

slofii, 2025-05-16:
Yes, there i can see it.

slofii, 2025-05-21:
[log.txt](https://github.com/user-attachments/files/20370224/log.txt)

thanks a lot! :)

meatpiHQ, 2025-05-22:
@slofii what was the value of the Odometer at the time of reading?

slofii, 2025-05-22:
The value was 1014190 km.
 
[log 2.txt](https://github.com/user-attachments/files/20394466/log.2.txt)
Now I did it exactly like in the tutorial, but Im afraid the odometer is missing again. 

Here are two pictures of the car scanner. Maybe this helps.
[image]
[image]

meatpiHQ, 2025-05-22:
@slofii Done, please try and let me know if it works.

slofii, 2025-05-22:
Yes, it works! Thanks a lot. :)