meatpiHQ/wican-fw #377: New Vehicle - BMW X1 2014

droopanu, 2025-04-11:
Hello,

I tried to read your "New Vehicle" instructions but got lost.
Is there anything I can provide to help create a profile for this vehicle?

Cheers,
Liviu

meatpiHQ, 2025-04-13:
hey @droopanu a profile for this vehicle might not be needed unless there's a manufacturer specific parameters. You should try to use the automate to scan for the available PIDs first.

droopanu, 2025-04-14:
Thank you very much! I completely missed that option :)
The only thing not working now is the odometer. I guess I should familiarize myself with the Car Scanner logs :)

meatpiHQ, 2025-04-14:
If you're able to see the odometer in car scanner then we can create a profile for your vehicle. You can post the logs here and I can help you create a profile. Here's how to export the logs.

https://meatpihq.github.io/wican-fw/car_scanner/logs

droopanu, 2025-04-14:
After changing a few profiles on car scanner I managed to find the odometer. I can't find the value in the log. Hope I managed to capture it 😁 

[image]

[car scanner_odometer.txt](https://github.com/user-attachments/files/19728872/car.scanner_odometer.txt)

[car scanner_noodometer.txt](https://github.com/user-attachments/files/19728877/car.scanner_noodometer.txt)

meatpiHQ, 2025-04-15:
@droopanu can you please re-do the logs, you need to press on the odometer, this way car scanner will only request the odometer, this way we can isolate it. 

https://meatpihq.github.io/wican-fw/car_scanner/logs

droopanu, 2025-04-16:
gotcha!
hope this will be better:

[carscanner_odomoeter_95504.txt](https://github.com/user-attachments/files/19770269/carscanner_odomoeter_95504.txt)