meatpiHQ/wican-fw #378: [FeatureRequest] - "cache" last read PID values for Home Assistant

kquinsland, 2025-04-12:
I am using the wican pro with home assistant. To save (a bit) of vehicle battery, I have configured the home assistant integration to poll the device every 300 seconds (5 min).

The reasoning is simple: the device only has WiFi so Home Assistant can only ever interrogate the device when it is in the garage and there's little value in asking the dongle for updates 120x (or more!) an hour if the vehicle is in the garage... especially when (nearly?) all CAN bus traffic stops the second I take the key out of the ignition switch.

This means that there is a _very_ narrow window of time in which the vehicle is

a) powered/running / has active CAN bus traffic
b) within range of the WiFi (read: in or very near to the garage)

The odds of `a)` and `b)` being true at the _precise exact same time_ that Home Assistant _also_ interrogates the device is virtually zero.

So this is the ask: 

For each PID/Value that's monitored, _cache_ the last known value until overridden (default/simplest configuration) or until successfully "read".

Here is the current situation that I have:

```mermaid
timeline
    title Current sequence of events
    t=0   : Driving home, nearly there but no WiFi coverage for another 120 seconds
    t=1   : Home Assistant polls device, times out due to not on network, next poll at t=501
    t=121 : I pull into the garage and select "park"
    t=125 : I power off the car, remove the key and exit the garage
    t=126 : All PID values are "cleared" on the WiCan-Pro
    t=130 : Device successfully authenticates to the WiFi AP
    t=501 : Home Assistant polls device, finds it, records "not available" for fuel tank level and other PID values
```

Here is how I'd like things to work:

```mermaid
timeline
    title Desired sequence of events
    t=0   : Driving home, nearly there but no WiFi coverage for another 120 seconds
    t=1   : Home Assistant polls device, times out due to not on network, next poll at t=501
    t=121 : I pull into the garage and select "park"
    t=125 : I power off the car, remove the key and exit the garage
    t=130 : Device successfully authenticates to the WiFi AP
    t=501 : Home Assistant polls device, finds it, records the last observed values for fuel tank level and other PID values
```

TO the best of my knowledge, there is no _trivial_ way to "backdate" a sensor value in home assistant. E.G: at t=501, inquire about fuel level and learn that at t=121, the measurement values was 47%, therefore insert into home assistant state db that `sensor.car_fuel_level = 47`... 300 seconds ago.

So it may be worth adding a "last observed" sensor for each measurement that can be optionally disabled in home assistant. I can imagine several scenarios where it doesn't really matter _when_ the measurement was taken... so long as it's accurate. I can imagine just a few scenarios where knowing how old a measurement is may effect it's perceived accuracy.

meatpiHQ, 2025-04-15:
@kquinsland This will be fixed in the next release. The issue here is that WiCAN won't poll bus if it's not connected to WiFi.

meatpiHQ, 2025-04-15:
it will behave as you described in your proposed solution.