meatpiHQ/wican-fw #381: Losing Config/Reverting to factoy default.

CAHutchins, 2025-04-14:
Often, after saving changes, my WiCan Pro is loosing configuration and reverting to factory default.
I'm generally saving changes to the 'Automate' page while attempting to configure PIDs.
The device reboots after saving changes and never returns to the WiFi, so I have to re-connect to the AP and I find the WiFi configuration data is gone.
It's happened to me a half dozen times now which is frustrating.
I did search to see if this is a known bug but didn't see anything.
FWIW, my WiFi is WPA2 and the WiFi password is 63 characters long.

CAHutchins, 2025-04-14:
I switched to a different SSID that has a shorter password and that seems too have solved the problem.

meatpiHQ, 2025-04-15:
@CAHutchins Thanks for reporting this, the max ssid length is 32 char.

hessijames79, 2025-04-15:
Same problem here. SSID length is 7.

CAHutchins, 2025-04-15:
> Same problem here. SSID length is 7.
How long is your password?
My SSID was only 9 characters but my password is 63 characters.
I switched to a different SSID with a smaller 51 character **password** and the problem stopped.
I have a separate WiFi vLAN for iOT devices because many of those don't like super long passwords and can't use WPA3

Also, I just realized I misspelled 'factory'. Oops.

hessijames79, 2025-04-15:
> How long is your password? My SSID is only 9 characters but my _password_ is 63 characters. I switched to a different SSID with a smaller 51 character **password** and the problem stopped. I have a separate WiFi vLAN for iOT devices because many of those don't like super long passwords and can't use WPA3

63 characters which works fine with all other devices in the same network. It's a dedicated IoT network as well. Robot vacuums, a bunch of ESPs wich custom firmware or Tasmota, kitchen appliances, a wallbox, a nuki bridge etc. Changing the password won't work in my case.

CAHutchins, 2025-04-15:
Yeah, I'm guessing there's a bug with passwords that long. 
Maybe an overflow causing a configuration data corruption resulting in a config reset to default.
It allowed me to set the WiFi password and connected fine but then when I made changes to the config it would lose the WiFi config again.

meatpiHQ, 2025-06-02:
@FEmser Can you confirm that you're running the latest firmware  version?

FEmser, 2025-06-03:
@meatpiHQ running this right now - I will update and report back after testing later this week.
Firmware ver:	4.11
Hardware ver:	WiCAN-OBD-PRO

FEmser, 2025-06-21:
@meatpiHQ Sorry it took longer to test.
I was testing with the 4.22 Beta and could not reproduce it.
Just now I tried it with the 4.23 and it also seems to work just fine.

I had an assumtion that loosing the connection had something to do with enabling mqtt, but this works fine too now.

Thanks :)

Derdirkf, 2026-09-05:
@meatpiHQ I have the same problem. WiCan OBD - now on 4.21
Short Wifi name with a "-" and short WPA2 with "* + #"

Looses the configuration when setting up the different PIDs to mqtt wallbox
additional there ist a Problem that I can save only after one PID change, when I change die mqtt wallbox of two PIDs the save-button is grey