meatpiHQ/wican-fw #382: Serious issues with wifi

mejmo, 2025-04-15:
I just bought MeatPi WiCAN and tried to get it running, but it has some serious issues, mainly related to the WIFI.

1) When I see the AP available, I click once, it disconnects, I click a second time, it disconnects, and then I get the IP address
2) The web server is available in 30% of tries. If it stops working, I need to either reconnect from the AP or unplug the device. 
3) WiCAN cannot connect to any AP in 95% of cases, once it got connected to my Android hotspot. No chance to connect it to Raspberry, or any other AP.

CAN monitor works, and I guess everything else, but anything related to WIFI is very, very unstable. The question is, what can I do? I guess this is not how it should work. I will connect to the USB a check the logs and put it here.

meatpiHQ, 2025-04-15:
@mejmo do you have the WiCAN PRO or original WiCAN-OBD and what's the firmware version?

mejmo, 2025-04-15:
@meatpiHQ The original one. Firmware version is the newest 4.04. The device does it from day one, I have not flashed anything yet. I will get the logs - 115200, serial? Do I need a driver in Windows to connect with Putty?

meatpiHQ, 2025-04-15:
How do you power the device?

martinformi, 2025-04-15:
> How do you power the device? 

Only through OBD2.

martinformi, 2025-04-15:
> How do you power the device? 

Only from OBD2. I tried two cars, the same.

meatpiHQ, 2025-04-15:
Try to change to WPA2 instead of WPA3.

meatpiHQ, 2025-04-15:
Do you have it AP mode or ap+station?

martinformi, 2025-04-15:
Ap+station mode. My AP in raspberry is using wpa2 (i tried to switch to wpa3 as well). I connected USB cable to my RPi, when I open /dev/ttyACM0 for serial communication with minicom I am not getting any data :/

meatpiHQ, 2025-04-15:
> Ap+station mode. My AP in raspberry is using wpa2 (i tried to switch to wpa3 as well). I connected USB cable to my RPi, when I open /dev/ttyACM0 for serial communication with minicom I am not getting any data :/

Yes logs are disabled by default.

Can you please download the device configuration from the system tab and post it here. Don't forget to remove any sensitive information like password...

meatpiHQ, 2025-04-15:
Btw are you using MDNS or direct IP to access the webserver?

martinformi, 2025-04-15:
Direct. 192.168.80.1 . Now when I switched back to AP, I think it got a bit more stable. It might be it cannot connect to the RPi AP and it blocks other processing on esp? Webserver is accessible more often. So I guess I need to go that route. I will create vustom firmware with possibility of changing the ssid.

meatpiHQ, 2025-04-15:
I'll send you a custom firmware later today. 

Try to turn off your 4G network.

martinformi, 2025-04-15:
> I'll send you a custom firmware later today. 
> 
> Try to turn off your 4G network.

I am afraid this is not intereference. Because I am getting 1ms pings without absolutely no packet loss. However ports 80 and 3333 are open, but connection is reset instantenously. Connection to other port returns connection refused, so networking stack works beautiful. This must be application layer. I will be waiting for fw, thank you for your time! I will post the config too.

mejmo, 2025-04-15:
```
{
  "config": {
    "wifi_mode": "AP",
    "ap_ch": "6",
    "sta_ssid": "gogo",
    "sta_pass": "XXXXXXXXX",
    "sta_security": "wpa2",
    "can_datarate": "500K",
    "can_mode": "normal",
    "port_type": "tcp",
    "port": "3333",
    "ap_pass": "XXXXXXXXX",
    "protocol": "elm327",
    "ble_pass": "123456",
    "ble_status": "disable",
    "sleep_status": "disable",
    "sleep_volt": "13.1",
    "sleep_time": "16",
    "batt_alert": "disable",
    "batt_alert_ssid": "MeatPi",
    "batt_alert_pass": "TomatoSauce",
    "batt_alert_volt": "11.0",
    "batt_alert_protocol": "mqtt",
    "batt_alert_url": "mqtt://mqtt.eclipseprojects.io",
    "batt_alert_port": "1883",
    "batt_alert_topic": "CAR1/voltage",
    "batt_alert_time": "1",
    "batt_mqtt_user": "meatpi",
    "batt_mqtt_pass": "meatpi",
    "mqtt_en": "disable",
    "mqtt_url": "mqtt://127.0.0.1",
    "mqtt_port": "1883",
    "mqtt_user": "meatpi",
    "mqtt_pass": "meatpi",
    "mqtt_tx_topic": "wican/dc06753c294d/can/tx",
    "mqtt_tx_en": "disable",
    "mqtt_rx_topic": "wican/dc06753c294d/can/rx",
    "mqtt_rx_en": "disable",
    "mqtt_status_topic": "wican/dc06753c294d/can/status",
    "mqtt_elm327_log": "disable"
  },
  "auto_pid_car_data": {
    "cars": [
      {
        "car_model": "",
        "init": "",
        "pids": []
      }
    ]
  },
  "auto_pid": {
    "initialisation": "",
    "grouping": "disable",
    "group_dest_type": "Default",
    "destination": "",
    "cycle": "5000",
    "car_specific": "disable",
    "ha_discovery": "disable",
    "car_model": "",
    "pids": [],
    "std_pids": [],
    "standard_pids": "disable",
    "ecu_protocol": "6"
  }
}
```

meatpiHQ, 2025-04-15:
Here you go. No need for drivers to use putty and baudrate does not matter.

[wican-fw_obd_v404-36-g36380b6.zip](https://github.com/user-attachments/files/19760023/wican-fw_obd_v404-36-g36380b6.zip)

mejmo, 2025-04-15:
The problem is how (or when) hostapd requests the re-association. The device gets connected each time hostapd is restarted. However, restarting the device does not re-establish the Wifi connection - hostapd just does not care. Lowering the inactivity timeout in `hostapd.conf` did the trick. Tomorrow I will try with the actual car.
Question - it is not possible to power the device with just USB (only if powered via USB power adapter). When I plug it into PC, the device keeps on restarting. When it is connected to OBD2 it is working - I can open the serial port. However, when not inserted into OBD, all the LEDs are blinking very lightly, and the device does not boot up. Other than that, I was able to use the HA integration now.

martinformi, 2025-04-16:
After the setting of mqtt, my device got bricked. No AP anymore, no connection to my RPi AP. How can I flash it with the USB?

meatpiHQ, 2025-04-16:
Yes USB. What was the last setting that you've changed before it got bricked?

https://meatpihq.github.io/wican-fw/config/firmware-update#_2-usb-flash

meatpiHQ, 2025-04-16:
Was it bricked or did you enable "Auto AP Disable"?

This option will turn off the device AP if it's connected to wifi network.

mejmo, 2025-04-30:
It got bricked when I enabled MQTT. 
It states I need to erase it at first... does it mean I need also the bootloader and partition table bin?

mejmo, 2025-04-30:
It is back online. Sorry for a bit more questions, but I would really like to get this working with HA. I would prefer that MeatPi pushes the data to MQTT instead of polling of data (that is how the integration does it - I am not a fan of this approach). In the MQTT approach - how often is data pushed to MQTT? I would need just a couple of parameters, but I prefer they are more frequent. Thank you for your time.