meatpiHQ/wican-fw #385: Where to find firmware files for Wican Pro ?

vincegre, 2025-04-16:
Hi

As title says, where can be found the firmware files for Wican Pro ? or still no updates since product has been shipped ?

Thanks

Vincèn

meatpiHQ, 2025-04-16:
@vincegre I haven't released any updates yet. It will be released very soon.

meatpiHQ, 2025-04-16:
@vincegre what do you mean publish the code? Its in this repo, just switch the branch to wican-pro.

vincegre, 2025-04-16:
> [@vincegre](https://github.com/vincegre) what do you mean publish the code? Its in this repo, just switch the branch to wican-pro.

euh sorry but I don't see any branch in the repo :( Do you have a direct link please ?

Thanks

meatpiHQ, 2025-04-16:
Here you go

https://github.com/meatpiHQ/wican-fw/tree/wican-pro

vincegre, 2025-04-17:
> Here you go
> 
> https://github.com/meatpiHQ/wican-fw/tree/wican-pro

Thanks ;)

closed 2025-04-17 by vincegre