meatpiHQ/wican-fw #387: New Vehicle - Honda Prologue 2024

wrkn4alivn, 2025-04-17:
Hello, I just received my WiCAN PRO! I would like to add a profile for one of my vehicles: 2024 Honda Prologue EV. In AutoPID mode, I can only get odometer (in km instead of miles). For other sensors like charge, range, etc. I think I will need a new vehicle profile? 

(After connecting via Car Scanner, there are dozens of custom PIDs available for this vehicle.)

meatpiHQ, 2025-04-17:
@wrkn4alivn please export the logs, lets start with SoC first.

Follow the instructions in the link to get the logs for the SoC,

https://meatpihq.github.io/wican-fw/car_scanner/logs

wrkn4alivn, 2025-04-18:
logs attached

[log_odometer.txt](https://github.com/user-attachments/files/19814440/log_odometer.txt)
[log_soc.txt](https://github.com/user-attachments/files/19814441/log_soc.txt)

wrkn4alivn, 2025-05-04:
Hello, friendly bump

meatpiHQ, 2025-05-05:
@wrkn4alivn sorry for the delay. What were the SoC and odometer values at the time of the log?

wrkn4alivn, 2025-05-05:
SoC was 62%, and odometer was 5156 mi.

meatpiHQ, 2025-05-05:
@wrkn4alivn I've created a profile for the Prologue, if the SoC value is incorrect please make another log and note the value.

For the Odometer no need to put that in the profile you can simply scan for the Standard PIDs and you will find it in the list, make sure to select protocol 7.