meatpiHQ/wican-fw #393: 2023 Chevrolet Bolt - Custom PID Codes / Profile

points-unknown, 2025-04-30:
I am just getting into my wican pro and I can't figure out how to bulk add additional PIDs. There is a pretty complete PID mapping available for this vehicle (as there is for so many) and I want to add several of them to the vehicle specific output. 

When I select "Chevrolet: Bolt" from the vehicle model list it just vies me 2 PIDs. At least , that's what I see. When I try to upload a custom "Vehicle Profile" json file, it... reloads and does... nothing. I am sure I have a syntax issue or something, but it's hard for me to tell, since I am not following the documentation very well - or rather, it appears there might be some different syntax depending on ... something.

I kinda expected it to be like Torque - where I could just upload the CSV and then it would load all the PIDs in that list. But then I went down the rabbit hole of making it fit this JSON schema.

In any case, below is my custom vehicle PID file. It's just converted to the provided json schema using a python script. I would love help either troubleshooting this json so it 'just works' or to have the Bolt profile updated with these PIDs. Happy to do whatever I can to help.

```
{
    "cars": [
        {
            "car_model": "Chevrolet: Bolt EV",
            "init": "ATSP6;ATST96;",
            "pids": [
                {
                    "pid": "2245F9",
                    "pid_init": "ATSH7E4",
                    "parameters": {
                        "name": "Bat_Cap_Raw",
                        "expression": "((A*256)+B)/100",
                        "class": "battery",
                        "unit": "Ah",
                        "min": "100",
                        "max": "250"
                    }
                },
                {
                    "pid": "2245F9",
                    "pid_init": "ATSH7E4",
                    "parameters": {
                        "name": "Bat_Cap_Est",
                        "expression": "((A*256)+B)*0.0032",
                        "class": "battery",
                        "unit": "kWh",
                        "min": "30",
                        "max": "70"
                    }
                },
                {
                    "pid": "228334",
                    "pid_init": "ATSH7E4",
                    "parameters": {
                        "name": "Batt_%_DIC",
                        "expression": "A*100/255",
                        "class": "battery",
                        "unit": "%",
                        "min": "0",
                        "max": "100"
                    }
                },
                {
                    "pid": "2243AF",
                    "pid_init": "ATSH7E4",
                    "parameters": {
                        "name": "SoC_Raw_HD",
                        "expression": "((((A*256)+B)*100)/65535)",
                        "class": "battery",
                        "unit": "%",
                        "min": "0",
                        "max": "105"
                    }
                },
                {
                    "pid": "2240D4",
                    "pid_init": "ATSH7E4",
                    "parameters": {
                        "name": "Batt_Amps_HD",
                        "expression": "((Signed(A)*256)+B)/20",
                        "class": "battery",
[... 250 lines cut ...]
                        "class": "battery",
                        "unit": "?",
                        "min": "0",
                        "max": "10"
                    }
                },
                {
                    "pid": "2224AB",
                    "pid_init": "ATSH7E4",
                    "parameters": {
                        "name": "Chrg_Inactive",
                        "expression": "((A*256)+B)*2",
                        "class": "battery",
                        "unit": "sec",
                        "min": "0",
                        "max": "131072"
                    }
                }
            ]
        }
    ]
}

```
Any help is much appreciated.

meatpiHQ, 2025-05-04:
@points-unknown The profile schema has changed a bit, we recently standardized the parameters, check [params.json](https://github.com/meatpiHQ/wican-fw/blob/main/.vehicle_profiles/params.json)

```
{
  "car_model": "Chevrolet: Bolt",
  "init": "ATSP6;ATST96;",
  "pids": [
    {
      "pid": "2283341",
      "pid_init": "ATSH7E4",
      "parameters": {
        "SOC": "(B4*100)/255"
      }
    },
    {
      "pid": "2241A31",
      "pid_init": "ATSH7E4",
      "parameters": {
        "BATT_CAPACITY": "[B4:B5]/10"
      }
    }
  ]
}
```
---

To transform the expression from Torque to WiCAN format you can check the table blow

[image]

points-unknown, 2025-05-15:
I am not sure I am understanding what to do here correctly. Using the example above, my desire is to add additional PIDs to the vehicle specific profile. So, for example I would make a new bolt_advanced.json file with the following:
`
{
  "car_model": "Chevrolet: Bolt",
  "init": "ATSP6;ATST96;",
  "pids": [
    {
      "pid": "2283341",
      "pid_init": "ATSH7E4",
      "parameters": {
        "SOC": "(B4*100)/255"
      }
    },
    {
      "pid": "2241A31",
      "pid_init": "ATSH7E4",
      "parameters": {
        "BATT_CAPACITY": "[B4:B5]/10"
      }
    },
    {
      "pid": "2243AF",
      "pid_init": "ATSH7E4",
      "parameters": {
        "SoC_Raw_HD": "((((B5*256)+B6)*100)/65535)"
      }
    },
    {
      "pid": "224368",
      "pid_init": "ATSH7E4",
      "parameters": {
        "CAC_Volts": "B5*2"
      }
    }
  ]
}
`
This takes the base json schema from bolt.json and adds two additional PIDs, from the CSV that works in torque, but with converted bytes for the A and B designations in torque going to B5 and B6, per the above. 

Then on the WiCan  web app, i go to >> Automate >> vehicle specific >> vehicle profiles and choose my my new bolt_advanced.json file. When I do this, the page reloads, takes me back to status. And when I go back to automate I get... nothing new.

Is this expected behavior? Am I doing something wrong? Help appreciated, as always. Thanks.

meatpiHQ, 2025-05-19:
@points-unknown The profile upload format is still based on the old schema. I'll have to merge this profile in the repo to generate the profile properly. If you want to test you can add these as a custom PIDs. Or you can submit a PR and I'll merge it then you can test.

windseye, 2025-05-23:
I have the same problem with my 2023 Chevrolet Bolt EUV. SOC and Battery Capacity are unavailable, even with fresh reinstallation of Homeassistant with the vehicle on

michieldevries, 2025-05-24:
> I have the same problem with my 2023 Chevrolet Bolt EUV. SOC and Battery Capacity are unavailable, even with fresh reinstallation of Homeassistant with the vehicle on

This worked for me, select Vehicle Model: Opel Ampera-E

Specific Initialisation: | ATSP6;ATST96;ATSH7E0;ATSH7E1;ATSH7E4;

Name: | SOC
PID: | 2283341
PID Init: |  
Expression: | (B4*100)/255
Unit: | %
Class: | battery
Min Value: |  
Max Value: |  
Period(ms): |  
Destination Type: | Default
Destination: |  

I would like to see also more PID's...

windseye, 2025-05-24:
Thank you michieldevries.The settings that you gave me expose SOC and HV_CAPACITY in Homeassistant.I also learned that enabling MQTT makes them unavailable in HA.windseyeSent from my Galaxy

michieldevries, 2025-05-25:
> [@windseye](https://github.com/windseye) can you please provide car scanner logs?
> 
> https://meatpihq.github.io/wican-fw/car_scanner/logs

Tody, I tried to log via Autoscanner. Does this help? I have an Opel Ampera-e (which is technically identical to the Chevrolet Bolt)

[log 2.txt](https://github.com/user-attachments/files/20430343/log.2.txt)

points-unknown, 2025-08-20:
I believe the padding byte (1 at the end) is required. But I haven't dug into it or tested both ways. 

I haven't had a chance to dig in more, but I'm glad what's been done so far it's helpful!