meatpiHQ/wican-fw #395: BYD Shark Vehicle Profile

nashy008, 2025-05-03:
Hi, 

I am keen to add the BYD Shark and any specific PIDs to my WiCAN. 

The documentation said I should open a issue and someone would be able to step me through how to add the details.

Cheers

meatpiHQ, 2025-05-04:
@nashy008 Please export the car scanner logs, don't forget to take note of the SoC when making the logs.

https://meatpihq.github.io/wican-fw/car_scanner/logs

bigfeetneedbigboots, 2025-06-01:
Hi, I was able to connect t my WiCAN PRO to my BYD Shark 6, connect to the WiCAN PRO via my network, see the SoC in the "All sensors" page, monitor the reading for a while, disconnect the WiCAN PRO then export the logs. The SoC was 39% when I removed the WiCAN PRO. I can't seem to see this sensor in the logs though. Seeking to get the SoC sensor reporting into MQTT so I can use it from there.

[log 3.txt](https://github.com/user-attachments/files/20537403/log.3.txt)

bigfeetneedbigboots, 2025-06-03:
Hi @meatpiHQ, thanks for your reply and support.

I updated the firmware on my WiCAN OBD PRO from v4.11p to v4.22p-beta.

Following the firmware update, I went to the "Automate" sidebar menu and set "Vehicle Specific" to "Enable", "Vehicle Model" to "BYD: Dolphin", and left the other items as default. A screenshot is attached.
[image]

I can see a bunch of data being sent by the WiCAN to my MQTT broker in the wican/48ca43xxxxxx/can/rx topic but it doesn't make much sense to me. The WiCAN OBD PRO is publishing about 50 messages per second to the MQTT broker! It would be nice to get the volume of data down as I only want the SOC every 5 seconds or so.
[image]

How do I test if the SOC is being read correctly? How will this data object be represented in MQTT (so I can filter it out)?

bigfeetneedbigboots, 2025-06-03:
OK, so I worked out that if I edit the vehicle specific SOC profile (not sure what it's called) and set the "Destination Type" to "MQTT_Topic" and the "Destination" to a unique MQTT topic address, I can get the SOC separated out from the rest in the .../rx topic. WIN!

I note the data being sent to the .../rx topic includes about 7 different "ID" numbers i.e. 301, 660, 847, 985, 1076, 2023 and 2031. How do I work out what these are? Can you give me a hint on how I decode the "data" for each? 

What do I put in the MQTT CAN Filter if I only was only wanting SOC to be published (and not everything else going to the .../rx topic?
[image]

meatpiHQ, 2025-06-03:
@bigfeetneedbigboots there is a check box next to the MQTT topic in the settings tab. Disable that and you won't receive these CAN frames, but you'll still receive the autopid data.

bigfeetneedbigboots, 2025-06-03:
> [@bigfeetneedbigboots](https://github.com/bigfeetneedbigboots) just confirm. You've unticked both boxes and submitted the changes?

Yes. I submitted the change after trying each combination i.e. one ticked, the other ticked, both ticked and neither ticked. Messages published to .../rx regardless.

meatpiHQ, 2025-06-05:
@bigfeetneedbigboots First thing you can scan for Standard PIDs, that will give you more sensors. For car specific parameters you can use car scanner to export the logs for the specific sensor (Link below). I can help you add more paramters, if you want to do it yourself you need to look into ELM327 protocol

To get the 12V battery voltage you can send `{"cmd":"get_vbatt"}`  to topic: `wican/_device_id_/cmd`

Btw here's the latest release: https://github.com/meatpiHQ/wican-fw/releases/tag/v4.23p

https://meatpihq.github.io/wican-fw/car_scanner/logs

bigfeetneedbigboots, 2026-03-05:
@meatpiHQ I've done some more work with the WiCAN PRO and my BYD Shark recently to identify more PID codes etc. Please see a vehicle_profiles.json file content below which shares the few codes I have managed to extract and validate as (probably) correct. 

Unfortunately the vehicle doesn't allow the WiCAN PRO to read these PID codes during charging so that is a current limitation. 

SOC is to 1 decimal place and SOC_D is to 0 decimal places.

I hope this is helpful to you and others. If anyone else is able to work out what the other PID codes are and do, please share so we can develop a full BYD Shark map. It would be really nice to get the PID codes for SOC during charging, charging power and charging state.

`{
  "cars": [
    {
      "car_model": "BYD: Shark",
      "init": "ATST96;ATSP6;",
      "pids": [
        {
          "pid": "2200051",
          "pid_init": "ATSH7E7;",
          "parameters": [
            {
              "name": "SOC_D",
              "expression": "B4",
              "unit": "%",
              "class": "battery",
              "min": "0",
              "max": "100"
            }
          ]
        },
        {
          "pid": "221FFD",
          "pid_init": "ATSH7E7;",
          "parameters": [
            {
              "name": "SOC",
              "expression": "((B5*256)+B4)/100",
              "unit": "%",
              "class": "battery",
              "min": "0",
              "max": "100"
            }
          ]
        },
        {
          "pid": "220008",
          "pid_init": "ATSH7E7;",
          "parameters": [
            {
              "name": "HV_V",
              "expression": "([B4:B5])/100",
              "unit": "V",
              "class": "voltage",
              "min": "0",
              "max": "1000"
            }
          ]
        },
        {
          "pid": "220032",
          "pid_init": "ATSH7E7;",
          "parameters": [
            {
              "name": "BATT_TEMP",
              "expression": "B4-40",
              "unit": "Â°C",
              "class": "temperature",
              "min": "-40",
              "max": "120"
            }
          ]
        },
        {
          "pid": "22002D1",
          "pid_init": "ATSH7E7;",
          "parameters": [
            {
              "name": "HV_C_V_MAX",
              "expression": "((B5*256)+B4)/1000",
              "unit": "V",
              "class": "voltage",
              "min": "0",
              "max": "10"
            }
          ]
        }
      ]
    }
  ]
}`