meatpiHQ/wican-fw #396: Auto PID

awmr2023, 2025-05-03:
Hi I have finally set up my Wican on my Peugeot boxer when I try to scan the PIDs I just get set to AutoPID and reboot to scan, I don't know if I'm being stupid but there is not autopid in the menu currently now stuck unable to set up

meatpiHQ, 2025-05-03:
@awmr2023 are you using WiCAN or WiCAN PRO?

awmr2023, 2025-05-03:
Wican

meatpiHQ, 2025-05-03:
@awmr2023 First change the protocol to autopid and click the submit button. The device will reboot, then try the scan again.

If the scan fails or does not return anything let me know. 

[image]

awmr2023, 2025-05-03:
Thanks you that's done the trick, been pulling my hair out try to find it lol need to change my glasses