meatpiHQ/wican-fw #397: WiCan & Engine Managment Light

NobleNetAdmin, 2025-05-03:
Not sure what am doing wrong. Tried my WiCan on 3 cars now.

2007 Ford Mondeo would not ready anything.
2014 Citroen DS3 scans successfully on "6) ISO 15765-4 CAN (11 bit ID, 500 Kbaud)" then dash start flashing up warnings for various things. This was few weeks ago and the codes cleared by my garage and been fine since without the WiCan pluged in.
Today tried it in me new 2014 Jag XF and similar to the DS3 scans on "6) ISO 15765-4 CAN (11 bit ID, 500 Kbaud)" then the engine managment light came on and various errors that did not have before pluggin in the WiCan.

Any help on what am doing wrong would be a great help.

Thanks Steve

meatpiHQ, 2025-05-03:
@Steve66253 I've never seen this before! The scan is successful and you still get check engine lights? Did you add any PIDs? Have you tried using car scanner app?

Steve66253, 2025-05-03:
Yes scan is successful and lets me add PIDs but see the info in Home Assistant even but car shows fault. I'd did with the DS3 and gave me a list of errors. At that time was not sure if was phantom errors or not so took car to garage they could not see anything and cleared the codes. That was few months ago and been fine since. Today thought would try again with the WiCan and band once start adding PIDs and errors start appering. Only adding the 5 standard PIDs below.

05-EngineCoolantTemp
0C-EngineRPM
0D-VehicleSpeed
21-DistanceMILOn
2F-FuelTankLevel

meatpiHQ, 2025-05-03:
Just to confirm you have WiCAN? Not the PRO right?

Steve66253, 2025-05-03:
Yes WiCAN

meatpiHQ, 2025-05-03:
Could open up the device? And take a photo of the front and back of the board?

Steve66253, 2025-05-03:
[image]
[image]

meatpiHQ, 2025-05-04:
@Steve66253 the PCB looks fine. what's the firmware version? 

Please post the device configuration from the System tab. Make sure to remove any sensitive information like passwords and ssids.

> WiCan and band once start adding PIDs and errors start appering. 

You add all 5 PIDs and pressed on the submit button, then the lights came up?

I'm trying to reproduce it here.

Steve66253, 2025-05-06:
Yes when hit the submit button fine until that point

{
  "config": {
    "wifi_mode": "APStation",
    "ap_ch": "6",
    "sta_ssid": "*************",
    "sta_pass": "************",
    "sta_security": "wpa2",
    "can_datarate": "500K",
    "can_mode": "normal",
    "port_type": "tcp",
    "port": "3333",
    "ap_pass": "@meatpi#",
    "protocol": "elm327",
    "ble_pass": "123456",
    "ble_status": "disable",
    "sleep_status": "enable",
    "sleep_volt": "13.1",
    "sleep_time": "15",
    "batt_alert": "disable",
    "batt_alert_ssid": "MeatPi",
    "batt_alert_pass": "TomatoSauce",
    "batt_alert_volt": "11.0",
    "batt_alert_protocol": "mqtt",
    "batt_alert_url": "mqtt://mqtt.eclipseprojects.io",
    "batt_alert_port": "1883",
    "batt_alert_topic": "CAR1/voltage",
    "batt_alert_time": "1",
    "batt_mqtt_user": "******",
    "batt_mqtt_pass": "******",
    "mqtt_en": "disable",
    "mqtt_url": "mqtt://*******",
    "mqtt_port": "1883",
    "mqtt_user": "*****",
    "mqtt_pass": "*****",
    "mqtt_tx_topic": "wican/f0f5bdfa40f9/can/tx",
    "mqtt_tx_en": "disable",
    "mqtt_rx_topic": "wican/f0f5bdfa40f9/can/rx",
    "mqtt_rx_en": "disable",
    "mqtt_status_topic": "wican/f0f5bdfa40f9/can/status",
    "mqtt_elm327_log": "disable"
  },
  "auto_pid_car_data": {
    "cars": [
      {
        "car_model": "AAA: Generic",
        "init": "ATSP6;",
        "pids": []
      }
    ]
  },
  "auto_pid": {
    "initialisation": "",
    "grouping": "disable",
    "group_dest_type": "Default",
    "destination": "",
    "cycle": "5000",
    "car_specific": "disable",
    "ha_discovery": "disable",
    "car_model": "AAA: Generic",
    "pids": [],
    "std_pids": [
      {
        "Name": "05-EngineCoolantTemp",
        "ReceiveHeader": "",
        "Period": "1000",
        "Type": "Default",
        "Send_to": ""
      },
      {
        "Name": "0C-EngineRPM",
        "ReceiveHeader": "",
        "Period": "1000",
        "Type": "Default",
        "Send_to": ""
      },
      {
        "Name": "0D-VehicleSpeed",
        "ReceiveHeader": "",
        "Period": "1000",
        "Type": "Default",
        "Send_to": ""
      },
      {
        "Name": "21-DistanceMILOn",
        "ReceiveHeader": "",
        "Period": "1000",
        "Type": "Default",
        "Send_to": ""
      },
      {
        "Name": "2F-FuelTankLevel",
        "ReceiveHeader": "",
        "Period": "1000",
        "Type": "Default",
        "Send_to": ""
      }
    ],
    "standard_pids": "enable",
    "ecu_protocol": "6"
  }
}

Steve66253, 2025-05-11:
@meatpiHQ  Hi wondering if had made any progress on getting to the bottom of this?

meatpiHQ, 2025-05-12:
@Steve66253 I still cannot figure out what is special about this car. Have you tried using car scanner app? Please check if triggers the lights. If not then there must be some vehicles specific command being sent to the ECU.

Steve66253, 2025-05-16:
@meatpiHQ Not just this car done it on 2 Citreon and Jaguar both 2014. I've tried the Car Scanner app and seems to been okwas no warnings log is below.

===== LOG STARTED: 16.05.2025 17:20:23=====
=====Version=1.118.0 Build=401180 P=False NoDelayELM327Init=False
====Connection profile=Jaguar OBD-II / EOBD
[Connection type: WIFI]

[WiFi name: <unknown ssid>]

[Connecting to 192.168.*.*:3333]
TCPConnectionV1
client = new TcpClient();
BEFORE await client.ConnectAsync(server, port);
Binding...
Start binding WiFi network
Wifi networks found:2
client.Connected
got stream from client

[Connected to 192.168.*.*:3333]
ATZ
ATE0

[ELM RESET]
ELM327 v1.3a

>
Initialize(initMode=Default)

InitializeDefaultInitString(ProtocolNumber=6,initMode=Default,useATST96=False)
ATE0
OK

>ATE0
OK

>STI
?

>
[STSupported=False(ByResponse)]VTI
?

>
[VTSupported=False(ByResponse)]
[ELM RESET]
ATD
OK

>ATD0
OK

>ATE0
OK

>ATH1
OK

>ATSP6
OK

>ATE0
OK

>ATH1
OK

>ATM0
OK

[... 6638 lines cut ...]
ZeroConsumptionWhenZero=False

Override_roles=
ATST=64
P=False
FPGC=2
CC=0

OS version=14
Platform=Android
Idiom=Phone
DeviceModel=SM-S906B
DeviceManufacturer=samsung

OrigCulture=en-GB
OrigUICulture=en-GB
LocationPermission=False
StoragePermission=False
BluetoothPermission=True
NearbyDevicesPermission=Granted
Version=1.118.0/401180

PrVersion=1.118.5
#Gplay
[***/// END OF CONTACT DEVELOPER REPORT]

meatpiHQ, 2025-05-20:
@Steve66253 That's weird, since WiCAN is doing exactly what car scanner is doing, and it works for all of the cars I've seen so far.

Please try add a custom PID `0100` with expression `B0+1`, this is a PID that gets all supports PIDs and it's the first thing that is sent to the ECU.

Steve66253, 2025-05-29:
@meatpiHQ seem to gotten somewhere I've managed to drive car and not trough up errors as long as use the Vehicle Specific AAA option try to add Standard PIDs has issues. So wondered if was possible to have profile for Jaguar XF (X250) Diesel add please so can get few more sensors that showed on Car Scanner.

[image]

Steve66253, 2025-06-05:
@meatpiHQ sorry any progress with this notice all using the AAA profile returns odometer at 300 something million km so custom profile would much appreciated thanks Steve

Steve66253, 2025-06-06:
@meatpiHQ 

[config_2025-06-06.json](https://github.com/user-attachments/files/20631975/config_2025-06-06.json)

Steve66253, 2025-06-17:
@meatpiHQ the engine lights issue does not happen if you Vehicle Profiles, been using the generic one with success apart for the incorrect mileage and would also like some more pids as above from CarScanner. Thanks Steve

Steve66253, 2025-07-31:
@meatpiHQ log is already further up on this thread. Yes millage and if possible MPG and range please

[image]

meatpiHQ, 2025-07-31:
@Steve66253 please read the instructions in link carefully. You need to click on the specific sensor you want in car scanner let it log for few seconds and unplug WiCAN. And if course note the value you see in car scanner