meatpiHQ/wican-fw #398: Wican causing vehicle errors

awmr2023, 2025-05-04:
So... after getting the wican set up on my 2014 Peugeot Boxer and following the great tutorial from Smarty van I have now got the van integrated to my HA. Unfortunately as soon as I start to drive my dashboard lights up like a Christmas tree. Specifically hill start and ESC/ASR. These resolve them selves restart the vehicle but come back shortly after maybe 60sec of driving. If I remove the wican and restart the vehicle then the errors resolve itself. It's disappointing. I have gone on to the wican and removed the PIDs then readdrd them. On saving and reboot I am getting the same engine lights coming on again resolving on an engine restart. Any ideas how to resolve this issue? I have attached screenshots of my wican set up
[image]
[image]
![image](https://github.com/user-attachments/assets/adb81c1d-
[image]c79b-41ce-aa57-fb3de19af03d)

meatpiHQ, 2025-05-04:
@awmr2023 let's try to figure out this issue, can you please try to change protocol to elm327 and use "car scanner" while driving to see if it triggers the dash board lights?

Car scanner will basically reading the data from the ECU similar to what WiCAN is doing.

awmr2023, 2025-05-04:
Okay I will try this tomorrow when we are traveling home. I assume I need to use BLE with my phone? Which app do I need to use there are a couple of them labels as car scanner on the App Store

meatpiHQ, 2025-05-05:
@awmr2023 no need for BLE you can use it in station or AP mode

https://meatpihq.github.io/wican-fw/car_scanner/usage

awmr2023, 2025-05-05:
- Thanks for this info I have been unsuccessful in getting it to connect in car scanner

Below is a c+p of the log I can't seem to upload on my phone 

===== DISCONNECTED AT: 05.05.2025 14:44:08Connection attempts failed

===== LOG STARTED: 05.05.2025 14:45:09=====
=====Version=1.118.0 Build=1.118.0 P=False NoDelayELM327Init=False
====Connection profile=Peugeot Fiat: Ducato; Peugeot: Boxer/Manager; Citroen: Relay/Jumper; RAM: ProMaster; Opel/Vauxhall: Movano 2.2 HDI
[Connection type: WIFI]

[WiFi name: <null>]

[Connecting to 	 192.168.1.41:3333]
client = new TcpClient();
BEFORE await client.ConnectAsync(server, port);
TCP connection timeout by Task.Delay(8)

Connection failed.

===== DISCONNECTED AT: 05.05.2025 14:45:13
===== LOG STARTED: 05.05.2025 14:45:14=====
=====Version=1.118.0 Build=1.118.0 P=False NoDelayELM327Init=False
====Connection profile=Peugeot Fiat: Ducato; Peugeot: Boxer/Manager; Citroen: Relay/Jumper; RAM: ProMaster; Opel/Vauxhall: Movano 2.2 HDI
[Connection type: WIFI]

[WiFi name: <null>]

[Connecting to 	 192.168.1.41:3333]
client = new TcpClient();
BEFORE await client.ConnectAsync(server, port);
TCP connection timeout by Task.Delay(8)

Connection failed.

===== DISCONNECTED AT: 05.05.2025 14:45:18
===== LOG STARTED: 05.05.2025 14:45:19=====
=====Version=1.118.0 Build=1.118.0 P=False NoDelayELM327Init=False
====Connection profile=Peugeot Fiat: Ducato; Peugeot: Boxer/Manager; Citroen: Relay/Jumper; RAM: ProMaster; Opel/Vauxhall: Movano 2.2 HDI
[Connection type: WIFI]

[WiFi name: <null>]

[Connecting to 	 192.168.1.41:3333]
client = new TcpClient();
BEFORE await client.ConnectAsync(server, port);
TCP connection timeout by Task.Delay(8)

Connection failed.

===== DISCONNECTED AT: 05.05.2025 14:45:23Connection attempts failed

===== LOG STARTED: 05.05.2025 14:47:05=====
=====Version=1.118.0 Build=1.118.0 P=False NoDelayELM327Init=False
====Connection profile=Peugeot Fiat: Ducato; Peugeot: Boxer/Manager; Citroen: Relay/Jumper; RAM: ProMaster; Opel/Vauxhall: Movano 2.2 HDI
[Connection type: WIFI]

[WiFi name: <null>]

[Connecting to 	 192.168.80.1:3333]
client = new TcpClient();
BEFORE await client.ConnectAsync(server, port);
TCP connection timeout by Task.Delay(8)

Connection failed.

===== DISCONNECTED AT: 05.05.2025 14:47:09
===== LOG STARTED: 05.05.2025 14:47:10=====
=====Version=1.118.0 Build=1.118.0 P=False NoDelayELM327Init=False
====Connection profile=Peugeot Fiat: Ducato; Peugeot: Boxer/Manager; Citroen: Relay/Jumper; RAM: ProMaster; Opel/Vauxhall: Movano 2.2 HDI
[... 247 lines cut ...]
WiFiPort=3333
WiFiServer=	 192.168.80.1
ZeroConsumptionPIDId=-1
ZeroConsumptionWhenZero=False

Override_roles=
ATST=64
P=False
FPGC=0
CC=0

OS version=18.4.1
Platform=iOS
Idiom=Phone
DeviceModel=iPhone14,5
DeviceManufacturer=Apple

OrigCulture=en-GB
OrigUICulture=en-GB
RunningOnMac=False
Version=1.118.0/1.118.0

PrVersion=1.118.4
#iOS
[***/// END OF CONTACT DEVELOPER REPORT]

awmr2023, 2025-05-05:
Still not connecting sorry 

￾
===== LOG STARTED: 05.05.2025 10:58:17=====
=====Version=1.118.0 Build=1.118.0 P=False NoDelayELM327Init=False
====Connection profile=Peugeot Fiat: Ducato; Peugeot: Boxer/Manager; Citroen: Relay/Jumper; RAM: ProMaster; Opel/Vauxhall: Movano 2.2 HDI
[Connection type: WIFI]

[WiFi name: <null>]

[Connecting to 192.168.0.10:35000]
client = new TcpClient();
BEFORE await client.ConnectAsync(server, port);
TCP connection timeout by Task.Delay(8)

Connection failed.

===== DISCONNECTED AT: 05.05.2025 10:58:21
===== LOG STARTED: 05.05.2025 10:58:22=====
=====Version=1.118.0 Build=1.118.0 P=False NoDelayELM327Init=False
====Connection profile=Peugeot Fiat: Ducato; Peugeot: Boxer/Manager; Citroen: Relay/Jumper; RAM: ProMaster; Opel/Vauxhall: Movano 2.2 HDI
[Connection type: WIFI]

[WiFi name: <null>]

[Connecting to 192.168.0.10:35000]
client = new TcpClient();
BEFORE await client.ConnectAsync(server, port);
TCP connection timeout by Task.Delay(8)

Connection failed.

===== DISCONNECTED AT: 05.05.2025 10:58:26
===== LOG STARTED: 05.05.2025 10:58:27=====
=====Version=1.118.0 Build=1.118.0 P=False NoDelayELM327Init=False
====Connection profile=Peugeot Fiat: Ducato; Peugeot: Boxer/Manager; Citroen: Relay/Jumper; RAM: ProMaster; Opel/Vauxhall: Movano 2.2 HDI
[Connection type: WIFI]

[WiFi name: <null>]

[Connecting to 192.168.0.10:35000]
client = new TcpClient();
BEFORE await client.ConnectAsync(server, port);
TCP connection timeout by Task.Delay(8)

Connection failed.

===== DISCONNECTED AT: 05.05.2025 10:58:31Connection attempts failed

===== LOG STARTED: 05.05.2025 10:58:36=====
=====Version=1.118.0 Build=1.118.0 P=False NoDelayELM327Init=False
====Connection profile=Peugeot Fiat: Ducato; Peugeot: Boxer/Manager; Citroen: Relay/Jumper; RAM: ProMaster; Opel/Vauxhall: Movano 2.2 HDI
[Connection type: WIFI]

[WiFi name: <null>]

[Connecting to 192.168.0.10:35000]
client = new TcpClient();
BEFORE await client.ConnectAsync(server, port);
TCP connection timeout by Task.Delay(8)

Connection failed.

===== DISCONNECTED AT: 05.05.2025 10:58:40
===== LOG STARTED: 05.05.2025 10:58:41=====
=====Version=1.118.0 Build=1.118.0 P=False NoDelayELM327Init=False
====Connection profile=Peugeot Fiat: Ducato; Peugeot: Boxer/Manager; Citroen: Relay/Jumper; RAM: ProMaster; Opel/Vauxhall: Movano 2.2 HDI
[Connection type: WIFI]

[WiFi name: <null>]
[... 1323 lines cut ...]
WiFiPort=3333
WiFiServer=	 192.168.80.1
ZeroConsumptionPIDId=-1
ZeroConsumptionWhenZero=False

Override_roles=
ATST=64
P=False
FPGC=0
CC=0

OS version=18.4.1
Platform=iOS
Idiom=Phone
DeviceModel=iPhone14,5
DeviceManufacturer=Apple

OrigCulture=en-GB
OrigUICulture=en-GB
RunningOnMac=False
Version=1.118.0/1.118.0

PrVersion=1.118.4
#iOS
[***/// END OF CONTACT DEVELOPER REPORT]

meatpiHQ, 2025-05-06:
@awmr2023 please try to connect over BLE.

awmr2023, 2025-05-07:
So what a ball ache this is becoming. I activated BLE and nothing dosent show up on my iPhone or the android tablet! I have noticed some times when I navigate to the ip I don't get a full menu? See image below, I have also noticed that it when I save any changes that also triggers the same vehicle error

awmr2023, 2025-05-09:
Thanks for that finally got BLE to connect had to use a different device but its work what info would you like from Car scanner

wdarney, 2025-05-22:
If the CAN bitrate is set improperly, it will interfere with canbus network. Maybe your car doesn't support 500k