meatpiHQ/wican-fw #402: Fiat Ducato 250 (2009)

danderik88, 2025-05-06:
Hi! I’m trying to get data from my 2009 Fiat Ducato 250 camper (2.3 MJT).
I followed the SmartyVan guide and tried using CarScanner, but I can’t connect to the ECU or read any data.
I’m using a WiCAN device and connecting to the OBD port, but I’m not sure if I’m using the correct settings (CAN speed, protocol, etc.).

Is there a known working configuration or example for this vehicle?
Any help would be appreciated—I’m quite new to CAN and diagnostics. Thanks in advance!

danderik88, 2025-05-07:
Hi. Using auto PIDs I wasn’t able to connect to any of the standard PIDs. In the ‘Monitor’ section and also in SavvyCAN I can see CAN messages at 50 kbps, and I noticed some changes when performing actions (RPM, speed, door open/close). In the CarScanner Pro app, I can connect to the ELM but not to the ECM. I also tested it on a 2020!Fiat Daily and everything works fine there.
Thanks and best regards.
[image]

meatpiHQ, 2025-05-07:
@danderik88 btw for the CAN messages on 50K you can setup a filter to calculate the values and send them over MQTT.

danderik88, 2025-05-07:
Thank you very much! That was actually my next step, but I have no experience with this and I couldn’t find a good guide on the topic. Do you know of any tutorial or documentation you could recommend? Thanks again and best regards.

danderik88, 2025-05-07:
Hi. Thank you! I will try this way. Best regards

danderik88, 2026-09-24:
I found some time for this project and I managed to integrate the CAN body in Home assistant. Those interested can follow the guide dedicated for the x250 Ducato model in the repo below.

https://github.com/danderik88/fiat-ducato-x250-wican

Tidle999, 2026-09-24:
Nice. Great work. That’s what I was looking for. I managed to get fuel and odo working but did not find the other values. Thanks for your work. Really appreciate it and excited to test it.

danderik88, 2026-09-24:
I’m glad you appreciate! The AI helped a lot but anyway it was a puzzle for a couple of months, starting from zero skills. With the wican pro in theory you should also have access to turbo pressure etc...

Tidle999, 2026-09-24:
It would be interesting to read out the current diesel flow rate to accurately calculate instantaneous fuel consumption. Right now, I am using the odometer mileage and the fuel consumed from the tank to calculate it, but this leads to deviations of +/- 3 liters, especially during shorter trips. For longer trips, it is relatively accurate.

danderik88, 2026-09-24:
Which wican device do you have? I added some sensors: average consumption, average speed, travel duration, travel distance, instantaneous consumption.

Tidle999, 2026-09-24:
I have the pro. The values are all provided on the 50kbps bus? Or do you calculate them by yourself?

danderik88, 2026-09-26:
Everything comes from the 50k body CAN, nothing is calculated. The WiCAN just listens (silent mode) to frames the car already broadcasts for the dashboard, including the trip computer and range values. The only thing I derive is "engine running", which is RPM > 300. Boost, load and DTCs aren't on that bus: they're K-line only.

I've only tested the non-Pro. The Pro should see the same frames at 50K, but I haven't tried it. If you do, let me know how it goes!